raspberrypi/rpi-eeprom #521: FAT read error with minimal-sized image

vmmon, 2024-01-16:
### Describe the bug

Booting from a minimized image fails with `FAT read failed @ 168` .

We are reducing the size of the image to boot so that the contents perfectly fits into the FAT partition.  Or in other words: the file-system is completely full. 

There seems to be an invalid end-of-disk check in the FAT driver:
- booting works if the image is a bit larger
- Linux tools like `mount` and `fsck.vfat` are happy with the filesystem

### Steps to reproduce the behaviour

A test image is available at: https://indigo.alpico.io/minimal.img

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

Nothing special here.

- BOOT_UART=1
- HTTP_HOST=...

### Bootloader logs

RPi: BOOTLOADER release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 9dfaa81a boardrev b03115 stc 421581
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x1 total-size: 16 Gbit 3200
DDR 3200 0 0 16 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f7
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: HTTP (07) order f
GENET: RESET_PHY 0
CTL 1140 PHY ID 600d 84a2
MII_CONTROL 1140
MII APD 0001 SCTL3 0001 IDDQ 0000

NETWORK: e4:5f:01:dc:ac:cf wait for link TFTP: 0.0.0.0
LINK STATUS: speed: 1000 full duplex
Link ready
GENET START: 64 16 32
GENET: UMAC_START 0xe45f01dc 0xaccf0000
RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
DHCP src: 50:e6:36:7d:73:35 192.168.178.1
YI_ADDR 192.168.178.45
SI_ADDR 192.168.178.1
OPTIONS:-
	op: 53 len:   1 DHCP recv OFFER (2) expect OFFER
	op: 54 len:   4 192.168.178.1
	op: 51 len:   4 
	op: 58 len:   4 
	op: 59 len:   4 
	op:  1 len:   4 255.255.255.0
	op:  3 len:   4 192.168.178.1
	op:  6 len:   4 192.168.178.1
	op: 15 len:   9 
NET 192.168.178.45 255.255.255.0 gw 192.168.178.1 tftp 0.0.0.0
DHCP src: 50:e6:36:7d:73:35 192.168.178.1
YI_ADDR 192.168.178.45
SI_ADDR 192.168.178.1
OPTIONS:-
	op: 53 len:   1 DHCP recv ACK (5) expect ACK
	op: 54 len:   4 192.168.178.1
	op: 51 len:   4 
	op: 58 len:   4 
	op: 59 len:   4 
	op:  1 len:   4 255.255.255.0
	op:  3 len:   4 192.168.178.1
	op:  6 len:   4 192.168.178.1
	op: 15 len:   9 
	op: 26 len:   2 
	op: 28 len:   4 192.168.178.255
	op: 42 len:   4 
	op: 158 len:   5 
NET 192.168.178.45 255.255.255.0 gw 192.168.178.1 tftp 0.0.0.0
RX: 2 IP: 0 IPV4: 2 MAC: 2 UDP: 2 UDP RECV: 2 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
Loading boot.img ...
HTTP: GET request for http://indigo.alpico.io:80/net_install/boot.sig
SIG boot.sig e5a1ec658a562539c838c6cebe50aa956c7d0bcbc31483a4fabd4119e6dd253e 1705395360
HTTP: GET request for http://indigo.alpico.io:80/net_install/boot.img
HTTP: got 0% at 1 kB/s
HTTP: got 87% at 993 kB/s
HTTP: got 100% at 1140 kB/s
Verifying
RSA verify
rsa-verify pass (0x0)
MBR: 0x00000008,    4552 type: 0x0c
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 12 lba: 8 oem: ' alpico ' volume: '            '
rsc 5 fat-sectors 2 c-count 568 c-size 8
root dir cluster 1 sectors 1 entries 16
FAT12 clusters 568
Read config.txt bytes       16 hnd 0x11
FAT read failed @ 168
Read start4.elf failed
Firmware not found
ERROR: 4
Boot mode: RESTART (0f) order 0
Restart 0 max -1

timg236, 2024-01-16:
Please can you attach dump of the FAT partition to that this can be investigated.

vmmon, 2024-01-16:
Here is the partition: [minimal.fat.gz](https://github.com/raspberrypi/rpi-eeprom/files/13948815/minimal.fat.gz)

It works if there are at least two clusters free...

timg236, 2024-01-17:
Reproduced bug

timg236, 2024-01-18:
@vmmon Please could you try the attached fix candidate?

[pi4-fat-max-clusters-2024-01-18-14-01-54-ae876639fd38ac823f7d18907aaec396c27d7488.zip](https://github.com/raspberrypi/rpi-eeprom/files/13977573/pi4-fat-max-clusters-2024-01-18-14-01-54-ae876639fd38ac823f7d18907aaec396c27d7488.zip)

vmmon, 2024-01-18:
I put both the `recovery.bin`  and the `pieeprom.bin` on the SD-Card. With FAT12 they were simply ignored, as the ROM does not support this older format.  With FAT16 and FAT32 it stops with

`FATAL error-code 24`

timg236, 2024-01-18:
Apollogies, this didn't include the pieeprom.sig file so you would have had to use "rpi-eeprom-config --edit pieeprom.bin" to install it - 24 is complaining about the missing signature.

Attached is a regular rpi eeprom recovery.zip file

[fat-rpi-eeprom-recovery.zip](https://github.com/raspberrypi/rpi-eeprom/files/13979164/fat-rpi-eeprom-recovery.zip)
prom.bin you would need to run "rpi-eeprom-config --edit pieeprom.bin" from the OS.

vmmon, 2024-01-18:
No problem, I have figured this out. I have now a workflow to generate eeprom update images from my laptop, which is nice for deploying network boot to a fleet of machines...

The bug is fixed in the `pieeprom.bin`  from today.  Good work. Thanks for your rapid response.

vmmon, 2024-01-19:
@timg236 Sorry, I was a bit to fast.

The bug is only triggered if the start4.elf occupies the last cluster of the filesystem. In my tests yesterday I had a `start4cd.elf` there as well, which shadowed the bug.

The underlying cause for the bug is that cluster numbers in the data-area are counted from 2. So the out-of bounds comparision in the read-function needs to be something like:
```
if (nr >= data_clusters + 2)  {
   // out-of-bounds error
}
```

timg236, 2024-01-19:
That's odd. According to the MSFT FAT spec FAT12 clusters are in the range 0x2 to MAX.
where MAX is defined to be

The Maximum Valid Cluster Number (MAX) for the volume is (CountofClusters + 1).
https://archive.org/details/FAT_Specification/page/16/mode/2up

minimal.img has 4552 data sectors and I make MAX to be 569 with start4.elf hitting that and wrapping back around to cluster 2.

Am I looking at the wrong image / missing something about CountofClusters?

vmmon, 2024-01-19:
There is something off in your numbers.
- the total sectors are 4552 as seen in the two byte value 0x11c8 at offset 0x1013 in the image
- we have only one FAT starting at 0x1a00 in the image
   - it is 4 sectors large which has space for 682 clusters
   - the last entry is at 0x1d57
- the data clusters are calculated as:
  `DataSec = TotSec – (BPB_ResvdSecCnt + (BPB_NumFATs * FATSz) + RootDirSectors)`

timg236, 2024-01-19:
I think that's what I said. The max data cluster is 569

vmmon, 2024-01-19:
let me recheck the different images I have...

timg236, 2024-01-19:
Thanks. If you have an example where you think it's wrong I'll run it back through or test code :)

vmmon, 2024-01-19:
Ok, so the SDCard path works with the minimal image and a slightly larger one.  Its via HTTP_BOOT that I get the errors:

```
type: 12 lba: 8 oem: ' alpico ' volume: '            '
rsc 5 fat-sectors 2 c-count 576 c-size 8
root dir cluster 1 sectors 1 entries 16
FAT12 clusters 576
Read config.txt bytes       27 hnd 0x11
FAT read failed @ 232
Read start4.elf failed
Firmware not found
ERROR: 4
```

timg236, 2024-01-22:
> Ok, so the SDCard path works with the minimal image and a slightly larger one.
> 
> Its via HTTP_BOOT that I get the errors:
> 
> ```
> Verifying
> RSA verify
> rsa-verify pass (0x0)
> MBR: 0x00000008,    4552 type: 0x0c
> MBR: 0x00000000,       0 type: 0x00
> MBR: 0x00000000,       0 type: 0x00
> MBR: 0x00000000,       0 type: 0x00
> Trying partition: 0
> type: 12 lba: 8 oem: ' alpico ' volume: '            '
> rsc 5 fat-sectors 2 c-count 568 c-size 8
> root dir cluster 1 sectors 1 entries 16
> FAT12 clusters 568
> Read config.txt bytes       16 hnd 0x11
> FAT read failed @ 168
> Read start4.elf failed
> Firmware not found
> ERROR: 4
> ```

Thanks, I can try minimal.img as a boot.img on a local server. Odd that it's complaining about sector 168, the HTTP boot ramdisk should just be a replacement at the block device level but obviously something weird is happening.

timg236, 2024-01-22:
Got it. Will post updated bootloader shortly.

timg236, 2024-01-22:
Should be fixed in this image
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/latest/pieeprom-2024-01-22.bin

vmmon, 2024-01-22:
Flashed the image and it works. Thanks!

What was it?

timg236, 2024-01-22:
Conflict between the code doing the read-ahead in the FAT and the ramdisk bounds checking

closed 2024-02-02 by vmmon