raspberrypi/rpi-eeprom #523: SD card detection broken in pieeprom-2024-01-05

trejan, 2024-01-19:
### Describe the bug

Bootloader screen shows SD detected when no card is present. Blank space for the CID.

Tested with two separate Pi 5 boards. No hardware attached except for power, HDMI and an active cooler.

SD detection does work with previous pieeprom-2024-01-05.

### Steps to reproduce the behaviour

Flash pieeprom-2024-01-05. Plug in power + HDMI only.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461

### System

```
2024/01/15 19:02:28
version f0aa07150e6f3ae21a88e5224cb9f57aba147348 (release)
timestamp 1705345348
update-time 1705672184
capabilities 0x0000007f
```
```
2024/01/15 19:02:28 
Copyright (c) 2012 Broadcom
version f0aa0715 (release) (embedded)
```

timg236, 2024-01-19:
Should the title say 2024/01/15 since you say "SD detection does work with previous pieeprom-2024-01-05."

trejan, 2024-01-19:
> Should the title say 2024/01/15 since you say "SD detection does work with previous pieeprom-2024-01-05."
Oops. Cut and paste fail.

> Also, to confirm you saying that the "card detect is broken" rather than doesn't boot. I can see how card-detect could have changed / been broken but boot seems to work for me.
Yes. It is just the bootloader screen showing that the card is detected when its not present. Everything works fine with a card inserted.

timg236, 2024-01-19:
Thanks, sounds like fallout from changes support the 2712D0 GPIO register churn.

timg236, 2024-01-19:
The changes required were a bit more invasive than expected, I've posted a pre-release. Will need review before pushing to rpi-update.

[rpi-eeprom-recovery.zip](https://github.com/raspberrypi/rpi-eeprom/files/13992111/rpi-eeprom-recovery.zip)

trejan, 2024-01-19:
> The changes required were a bit more invasive than expected, I've posted a pre-release. 

Thanks. The SD detection in the bootloader is fixed now.

> Will need review before pushing to rpi-update.

I'll have to revert to 2024-01-15 as net install doesn't work in this debug build. Using HTTP_HOST to point it at my local server doesn't help as it still fails the signature check.

Booting from NVMe and USB is okay.

timg236, 2024-01-22:
> > The changes required were a bit more invasive than expected, I've posted a pre-release.
> 
> Thanks. The SD detection in the bootloader is fixed now.
> 
> > Will need review before pushing to rpi-update.
> 
> I'll have to revert to 2024-01-15 later as net install doesn't work in this debug build. Using HTTP_HOST to point it at my local server doesn't help as it still fails the signature check.
> 
> Booting from NVMe and USB is okay.

Thanks. I think net-install was broken in the local build. Will push out a proper fix soon.

timg236, 2024-01-22:
Fix should now be available in rpi-update

trejan, 2024-01-22:
Yep. SD card detection and net install are both working with this new version. Thanks.

closed 2024-01-22 by trejan

microfx, 2024-02-02:
> Fix should now be available in rpi-update

What does this mean? I am searching for 5 hours straight for any trace of a solution... I just cloned this repo and tried it with those binaries (eeprom-update) 

``` 
This tool only works with a Raspberry Pi 4
```

```
 sudo ~/rpi-eeprom/rpi-eeprom-update -a -d ./firmware-2712/latest/pieeprom-2024-01-24.bin 
BOOTLOADER: up to date
   CURRENT: Mon 30 Oct 16:45:10 UTC 2023 (1698684310)
    LATEST: Thu  1 Jan 00:00:00 UTC 1970 (0)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.
```
```
 sudo ~/rpi-eeprom/rpi-eeprom-config --edit
Updating bootloader EEPROM
 image: /lib/firmware/raspberrypi/bootloader/default/pieeprom-2023-01-11.bin
config_src: blconfig device
config: /tmp/tmpumk3cyb6/boot.conf
################################################################################
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf41

################################################################################

*** To cancel this update run 'sudo rpi-eeprom-update -r' ***

This tool only works with a Raspberry Pi 4
```

It's not even working .. although it's working a bit better than the standard EEPROM-config that just says "oooh Raspberry Pi 4 onlyyyyy" but it's still not saving anything.... 

Can somebody explain me what my problem is? Any why am I the only one having this problem?

timg236, 2024-02-02:
Deleted off topic comments and locking since OP verified as fixed.

pelwell, 2024-02-02:
(Hint: `rpi-update` is not the same as `rpi-eeprom-update`)