raspberrypi/rpi-eeprom #526: NVME phison controller causing RPI 5 boot problems?

washran, 2024-01-22:
### Describe the bug

Western Digital seems to be the worst nvme ssd for the RPI 5. Why is the phison controller an issue? Nvme drives tested that work are:     AData Legend 700
    Axe Memory
    Crucial P2 M.2
    Crucial P3 Plus M.2
    Inland PCIe NVMe SSD
    Kingston KC3000
    Kioxia Exceria 2.5
    Lexar NM620
    Lexar NM710
    Netac M.2 NVMe SSD
    Origin Inception TLC830 Pro NVMe
    Patriot P310
    PNY CS1030
    Sabrent Rocket 4.0
    Sabrent Rocket Nano
    Samsung 980
    Samsung 980 Pro
    Team MP33
    Western Digital Black SN750 SE Series

I have a TeamGroup nvme ssd that boots fine but has unstable speeds. Has a phison controller though.

### Steps to reproduce the behaviour

Need a sd card to boot Western Digital nvme ssd drives. It seems that some of the issues have to do with the coding of the bootloader and the kernel. I'm sure the kinks will be ironed out.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf416
ENABLE_SELF_UPDATE=1

washran, 2024-01-23:
[[0;1;31mFAILED[0m] Failed to start [0;1;39mnvmf-autoc…systems automatically during boot.
See 'systemctl status nvmf-autoconnect.service' for details. This is a nvme-fabric module that is missing from kernel modules.

[boot.txt](https://github.com/raspberrypi/rpi-eeprom/files/14019822/boot.txt)

closed 2024-01-24 by washran

peterharperuk, 2024-01-24:
@washran I have a possible for that works for me for a WD SN550 driver if you're able to help test it?

janstaunton, 2024-02-06:
Have a similar issue with the Pi 4 Model B, Model Number: Western Digital SN730E in a SanDisk Extreme Pro

Can boot from the drive fine attached via USB, haven't tried speeds, but this service errors on boot: 

> 12:53 PM Failed to start nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot. systemd
> 12:53 PM nvmf-autoconnect.service: Failed with result 'exit-code'. systemd
> 12:53 PM nvmf-autoconnect.service: Control process exited, code=exited, status=1/FAILURE systemd
> 12:53 PM modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/6.1.0-rpi8-rpi-v8 modprobe
> 12:53 PM Starting nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot...