raspberrypi/rpi-eeprom #527: Cannot boot from MXT USB2 uSD card reader

procount, 2024-01-23:
### Describe the bug

This USB card reader works fine in Raspberry Pi OS, but it is not recognised by the USB boot loader.

### Steps to reproduce the behaviour

Tried booting a working uSD card using this adaptor on USB2 and USB3 ports of Pi4 and Pi5. Neither recognised it.
It booted straight away when I changed the adaptor for an Integral USB3 adaptor.
The diagnostic screen reported:
"MSD error [01:00] 2.00 000000:01 error 2"

### Device (s)

Raspberry Pi 5, Raspberry Pi 4 Mod. B

### Bootloader configuration.

Pi4:
┌─────────────────────────┤ Display EEPROM Status ├──────────────────────────┐
│                                                                            │ 
│ BOOTLOADER: up to date                                                     │ 
│    CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)                      │ 
│     LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)                      │ 
│    RELEASE: default (/lib/firmware/raspberrypi/bootloader-2711/default)    │ 
│                                                                            │ 
│   VL805_FW: Using bootloader EEPROM                                        │ 
│      VL805: up to date                                                     │ 
│    CURRENT: 000138c0                                                       │ 
│     LATEST: 000138c0                                                       │ 
│                                                                            │ 
│                                   <Ok>                                     │ 
│                                                                            │ 
└────────────────────────────────────────────────────────────────────────────┘ 

Pi5:
┌─────────────────────────┤ Display EEPROM Status ├──────────────────────────┐
│                                                                            │ 
│ BOOTLOADER: up to date                                                     │ 
│    CURRENT: Wed 13 Dec 16:20:43 UTC 2023 (1702484443)                      │ 
│     LATEST: Wed 18 Oct 17:30:17 UTC 2023 (1697650217)                      │ 
│    RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)    │ 
│                                                                            │ 
│                                                                            │ 
│                                   <Ok>                                     │ 
│                                                                            │ 
└────────────────────────────────────────────────────────────────────────────┘ 

### System

N/A

### USB boot

Here are the USB id codes for the reader from dmesg:

[  224.137287] usb 1-1.3.4: new high-speed USB device number 6 using xhci_hcd
[  224.241811] usb 1-1.3.4: New USB device found, idVendor=aaaa, idProduct=8816, bcdDevice=13.08
[  224.241831] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  224.241842] usb 1-1.3.4: Product: MXT USB Device

pi@raspberrypi:~ $ lsusb -vvv

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
[... 499 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

timg236, 2024-05-01:
Do you have a link to the product?

procount, 2024-05-01:
https://www.ebay.co.uk/itm/386153943967

I bought 2 Integral USB3 card readers from amazon https://www.amazon.co.uk/gp/product/B07PN54VQ2/ to replace one I lost.
These work fine on the RPi (prolific), but wouldn't work on my Win10 laptop (well, they worked, but they kept disconnecting after a few 10s of seconds.) So I bought this USB2 card reader from ebay, which worked fine on my laptop and in Raspberry Pi OS, but wouldn't work in the RPi boot loader.

timg236, 2024-07-29:
[pi5-rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/16408964/pi5-rpi-eeprom-recovery.zip)
[pi4-rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/16408977/pi4-rpi-eeprom-recovery.zip)

@procount Please could you try these bootloaders (flash using rpi-imager). I tweaked the order of the MSD operations because some SD-USB adapters don't wake up until after read-capacity has been issued.

timg236, 2024-07-29:
A better fix - the previous version caused an issue with multiple slot SD card readers where the card was not in LUN0

[pi4-rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/16415136/pi4-rpi-eeprom-recovery.zip)
[pi5-rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/16415137/pi5-rpi-eeprom-recovery.zip)

procount, 2024-08-13:
> A better fix - the previous version caused an issue with multiple slot SD card readers where the card was not in LUN0

Sorry Tim, my original SD card reader has stopped working altogether, so I can no longer test.

timg236, 2024-08-14:
No problem. Closing as fixed.

closed 2024-08-14 by timg236

Hellsnomad, 2024-10-30:
Hi,

This is not working on rasp 4 8gb model.

I have the same USB error when trying to boot image from sd card reader

Have tried the [pi4-rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/16415136/pi4-rpi-eeprom-recovery.zip) but 'firmware not available' is on the screen when switching on the device

let me know