raspberrypi/rpi-eeprom #529: NVME (0x6) removal from BOOT_ORDER not honored

AlphaDE, 2024-01-25:
### Describe the bug

I added a Pineberry Pi BM1L HatDrive! Bottom NVME adapter with a 512GB NVME to my Raspberry Pi 5. After setting up, I changed the boot-order with `sudo rpi-eeprom-config --edit` to `BOOT_ORDER=0xf416`. Later due to an error in fstab, I got stuck in emergency mode and tried to recover. Despite removing "0x6" from the BOOT_ORDER, the Pi 5 bootet from the NVME as long as it was connected making it impossible to boot from sdcard, mount the NVME partition and fix the issue. Also changing the order (61) which should boot from sdcard first does boot from NVME despite a sdcard is inserted.

### Steps to reproduce the behaviour

When 0x6 is removed from BOOT_ORDER or moved back, Pi 5 should not try to boot from NVME anymore or honor boot order. 

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
PCIE_PROBE=1

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

2024/01/22 14:44:36
version e891ded664f30eae0d95f44db449a2f70dd66c29 (release)
timestamp 1705934676
update-time 1706198011
capabilities 0x0000007f

2024/01/22 14:44:36
Copyright (c) 2012 Broadcom
version e891ded6 (release) (embedded)

Linux raspberrypi5-alfter 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

### NVMe boot

/dev/nvme0n1          /dev/ng0n1            73CPH60RQMZK         KBG50ZNS512G KIOXIA                      1         512,11  GB / 512,11  GB    512   B +  0 B   ANGA0102

NVME Identify Controller:
vid       : 0x1e0f
ssvid     : 0x1e0f
sn        : 73CPH60RQMZK
mn        : KBG50ZNS512G KIOXIA
fr        : ANGA0102
rab       : 2
ieee      : 8ce38e
cmic      : 0
  [3:3] : 0     ANA not supported
  [2:2] : 0     PCI
  [1:1] : 0     Single Controller
  [0:0] : 0     Single Port

mdts      : 9
cntlid    : 0
ver       : 0x10400
rtd3r     : 0x186a0
rtd3e     : 0x7a120
oaes      : 0x200
  [31:31] : 0   Discovery Log Change Notice Not Supported
  [27:27] : 0   Zone Descriptor Changed Notices Not Supported
  [15:15] : 0   Normal NSS Shutdown Event Not Supported
  [14:14] : 0   Endurance Group Event Aggregate Log Page Change Notice Not Supported
  [13:13] : 0   LBA Status Information Notices Not Supported
  [12:12] : 0   Predictable Latency Event Aggregate Log Change Notices Not Supported
  [11:11] : 0   Asymmetric Namespace Access Change Notices Not Supported
  [9:9] : 0x1   Firmware Activation Notices Supported
  [8:8] : 0     Namespace Attribute Changed Event Not Supported
[... 279 lines cut ...]
  [3:3] : 0     Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0x1   Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 255
nawupf  : 0
nacwu   : 255
nabsn   : 255
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 512.110.190.592
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : 00000000000000008ce38e04044f8bf1
eui64   : 8ce38e04044f8bf1
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x2 Good (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0x1 Better

19:41 25.01.2024

AlphaDE, 2024-01-25:
The instructions I found refer to use this and docs say to use it, if the hat is not HAT+. I found no reference whether the Pineberry BM1L is Hat+ or not.

timg236, 2024-01-25:
Please can you attach boot loader logs showing the behaviour and verify that you don’t have the same partition uuid on both the sd and the sd card. Otherwise, it’s not possible to predict which partition the kernel will mount.

AlphaDE, 2024-01-25:
@timg236 The partition IDs might be the reason. The howtos for setting up an NVMe recommend to setup the system on sdcard and then use SD Card Copier to transfer the data to the NVMe. But that leads obviously to identical partition UUIDs on both media and then the "broken" NVME took precende.  I'll see how I can modify the partition UUIDs on the sdcard.

timg236, 2024-01-25:
That would be explain it.
I normally use rpi imager on pc with usb nvme adapter or boot from a one off rpi is sd boot and run the imager on rpi os to flash the local nvme.

AlphaDE, 2024-01-25:
Digging deeper I found that partition 1 of the sdcard was mount to /boot/firmware and partition 2 of the NVMe as /, I changed the sdcard uuid with fdisk which should have fixed that. The NVMe How-To need to be reworked to avoid that confusion.

Should be fixed.

closed 2024-01-25 by AlphaDE