raspberrypi/rpi-eeprom #531: changing value from gen=2 to gen=3 in config.txt (dtparam=pciex1_gen=3) sometimes causes the raspbery pi 5 not to boot

kitopopo, 2024-02-02:
### Describe the bug

I use a HatDrive! Bottom from Pineberry pi in a raspberry pi 5 8gb with a nvme lexar nm620 256Gb. The nvme boots correctly with dtparam=pciex1_gen=2 in config.txt. I can change the dtparm to dtparam=pciex1_gen=3 the first timeand the nvme boots correctly with gen 3 and i make a speed test with hdparm and i get 830 MB/s, but sometimes if i power off the raspberry pi 5 and after power on the nvme does not boot (HatDrive leds are on and rasp fan is with high speed), If I power off and power on again the raspberry the HatDrive leds are always off (unrecoverable). I can solve the problem but  i need plug the nvme in a computer and change the dtparam from gen3 to gen 2 and all works correctly. Sometimes change again from gen2 to gen3 and sometimes boot and sometimes the boot crash.

### Steps to reproduce the behaviour

With the raspberrypi5 and the nvme working with (dtparam=pciex1_gen=3) powering off the raspberry some times and the nvme does not boot

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf416	
PCIE_PROBE=1

### System

BOOTLOADER: up to date
   CURRENT: lun 22 ene 2024 14:44:36 UTC (1705934676)
    LATEST: lun 22 ene 2024 14:44:36 UTC (1705934676)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
-------------------------------------------------------------------------
2024/01/22 14:44:36
version e891ded664f30eae0d95f44db449a2f70dd66c29 (release)
timestamp 1705934676
update-time 1706876734
capabilities 0x0000007f
----------------------------------------------------------------
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf416       
PCIE_PROBE=1
-------------------------------------------------
dtparam=nvme
dtparam=pciex1_gen=2
-------------------------------------------------
2024/01/22 14:44:36 
Copyright (c) 2012 Broadcom
version e891ded6 (release) (embedded)
---------------------------------------------
Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
-------------------------------------------------------------------------------------
dpkg -l | grep linux-image
ii  linux-image-6.1.0-rpi7-rpi-2712      1:6.1.63-1+rpt1                     arm64        Linux 6.1 for Raspberry Pi 2712, Raspberry Pi
ii  linux-image-6.1.0-rpi7-rpi-v8        1:6.1.63-1+rpt1                     arm64        Linux 6.1 for Raspberry Pi v8, Raspberry Pi
ii  linux-image-6.1.0-rpi8-rpi-2712      1:6.1.73-1+rpt1                     arm64        Linux 6.1 for Raspberry Pi 2712, Raspberry Pi
ii  linux-image-6.1.0-rpi8-rpi-v8        1:6.1.73-1+rpt1                     arm64        Linux 6.1 for Raspberry Pi v8, Raspberry Pi
ii  linux-image-rpi-2712                 1:6.1.73-1+rpt1                     arm64        Linux for Raspberry Pi 2712 (meta-package)
ii  linux-image-rpi-v8                   1:6.1.73-1+rpt1                     arm64        Linux for Raspberry Pi v8 (meta-package)

### NVMe boot

Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            NKE133R020578P110N   Lexar SSD NM620 256GB                    1         256,06  GB / 256,06  GB    512   B +  0 B   SN10660 
---------------------------------------------------------------------------------------------------------------------
sudo nvme id-ctrl -H /dev/nvme0
NVME Identify Controller:
vid       : 0x1e4b
ssvid     : 0x1e4b
sn        : NKE133R020578P110N  
mn        : Lexar SSD NM620 256GB                   
fr        : SN10660 
[... 301 lines cut ...]

dlfeat  : 0
  [4:4] : 0	Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0	Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0	Bytes Read From a Deallocated Logical Block and its Metadata are Not Reported

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr	: 0
nvmsetid: 0
endgid  : 0
nguid   : 0000000623103133caf25b0350000e31
eui64   : caf25b0350000e31
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)

pelwell, 2024-02-02:
For reasons I don't recall, BCM2712's PCIE1 can't quite manage gen3 timing. You can get away with for some devices under some circumstances - which is why there is a parameter to enable it - but it isn't safe to enable by default. In other words, this is expected behaviour.

timg236, 2024-02-02:
This also isn't under the control of the firmware

closed 2024-02-02 by timg236

lurch, 2024-02-02:
See the Warning at https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html#pcie-gen-3-0