raspberrypi/rpi-eeprom #532: rpi-eeprom-update: Use raspi-config to change the release. --> there is no option in rasp-config for a release

microfx, 2024-02-02:
### Describe the bug

the text " Use raspi-config to change the release." is not helping me. Where? I searched for over 10 minutes. There is nothing with a release. Maybe give better clues when writing stuff like this to not waste so many precious lifetime of other people with such hints. How can I change the release? I am completely stuck because this tool is weird right now in its current state with Raspberry Pi 4. Even the latest binary tells me it's only for Raspberry Pi 4 and doesn't do anything. 

### Steps to reproduce the behaviour

sudo rpi-eeprom-update -a

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

sudo rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Mon 30 Oct 16:45:10 UTC 2023 (1698684310)
    LATEST: Thu  1 Jan 00:00:00 UTC 1970 (0)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

### System

OS: Raspbian GNU/Linux 11 (bullseye) aarch64 Kernel: Linux 6.1.69-v8-16k+

timg236, 2024-02-02:
Closing. This is not the place for random questions

peterharperuk, 2024-02-02:
sudo raspi-config
Choose Advanced Options
Choose Bootloader Version

It allows you to change the bootloader version between default and latest.

timg236, 2024-02-02:
" Maybe give better clues when writing stuff like this to not waste so many precious lifetime of other people with such hints. "

We have better things to do than waste time on spurious questions that are already covered elsewhere. As the bug report template will have told you. This is not a bug report.

pelwell, 2024-02-02:
https://www.google.com/search?q=raspi-config+bootloader+version