raspberrypi/rpi-eeprom #535: Pi 5: booting from Intel SSDPEKKF256G7L NVMe drive fails on first attempt

andrum993, 2024-02-03:
### Describe the bug

When attempting to boot from an Intel SSDPEKKF256G7L NVMe drive, the bootloader fails to boot from the device at the first attempt. After then attempting to boot from USB then SD card (and failing to do so), booting from the NVMe drive then succeeds on the second attempt.

I have the drive attached to my Pi 5 using a Pimoroni NVMe base (https://shop.pimoroni.com/products/nvme-base?variant=41219587178579).

### Steps to reproduce the behaviour

Attempt to boot from the above noted device with 6 (NVMe) first in the BOOT_ORDER, and other boot types in the boot order, for example BOOT_ORDER=0xf146.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
BOOT_MODE=0f146

I have also added the following to capture debug info:
NETCONSOLE=6665@192.168.1.111/,6666@/

Booting with this line included shows the same behaviour.

### NVMe boot

andrew@tarmachan:~ $ sudo nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            BTPY71361KFM256D     INTEL SSDPEKKF256G7L                     1         256.06  GB / 256.06  GB    512   B +  0 B   131P
andrew@tarmachan:~ $ sudo nvme id-ctrl -H /dev/nvme0
NVME Identify Controller:
vid       : 0x8086
ssvid     : 0x8086
sn        : BTPY71361KFM256D
mn        : INTEL SSDPEKKF256G7L
fr        : 131P
rab       : 6
ieee      : 5cd2e4
cmic      : 0
  [3:3] : 0     ANA not supported
  [2:2] : 0     PCI
  [1:1] : 0     Single Controller
  [0:0] : 0     Single Port

mdts      : 5
cntlid    : 0x1
ver       : 0x10200
rtd3r     : 0x249f0
rtd3e     : 0x13880
oaes      : 0
  [31:31] : 0   Discovery Log Change Notice Not Supported
  [27:27] : 0   Zone Descriptor Changed Notices Not Supported
  [15:15] : 0   Normal NSS Shutdown Event Not Supported
  [14:14] : 0   Endurance Group Event Aggregate Log Page Change Notice Not Supported
  [13:13] : 0   LBA Status Information Notices Not Supported
  [12:12] : 0   Predictable Latency Event Aggregate Log Change Notices Not Supported
  [11:11] : 0   Asymmetric Namespace Access Change Notices Not Supported
  [9:9] : 0     Firmware Activation Notices Not Supported
  [8:8] : 0     Namespace Attribute Changed Event Not Supported

ctratt    : 0
  [19:19] : 0   Flexible Data Placement Not Supported
  [15:15] : 0   Extended LBA Formats Not Supported
  [14:14] : 0   Delete NVM Set Not Supported
  [13:13] : 0   Delete Endurance Group Not Supported
  [12:12] : 0   Variable Capacity Management Not Supported
  [11:11] : 0   Fixed Capacity Management Not Supported
[... 271 lines cut ...]

dlfeat  : 0
  [4:4] : 0     Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0     Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0     Bytes Read From a Deallocated Logical Block and its Metadata are Not Reported

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : 00000000000000000000000000000000
eui64   : 5cd2e41471a0935d
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)

andrum993, 2024-02-03:
Bootloader logs as requestion in form - Github would not allow me to submit with this included since it objects to a form submission greated than 65535 characters in length, so here it is separately:

System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

Linux tarmachan 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
Revision	: d04170
Serial		: 00afddee8e47ff82
Model		: Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

2024/01/24 12:16:01 
Copyright (c) 2012 Broadcom
version 1aeae380 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"

Filesystem information
----------------------
Filesystem                  1K-blocks      Used Available Use% Mounted on
udev                          3952000         0   3952000   0% /dev
tmpfs                          824704      7408    817296   1% /run
/dev/nvme0n1p2              245679712  11365328 221821840   5% /
tmpfs                         4123504         0   4123504   0% /dev/shm
tmpfs                            5120        48      5072   1% /run/lock
/dev/nvme0n1p1                 522232     62390    459842  12% /boot/firmware
zdata                       984416256      1024 984415232   1% /zdata
zdata/pi-backups2           984416256      1024 984415232   1% /zdata/pi-backups2
zdata/pi-backups            984416256      1024 984415232   1% /zdata/pi-backups
zdata/newtempmon            984417280      2048 984415232   1% /zdata/newtempmon
zdata/Software Library     1019437056  35021824 984415232   4% /zdata/Software Library
zdata/data                  984650752    235520 984415232   1% /zdata/data
zdata/learning              984593408    178176 984415232   1% /zdata/learning
zdata/Videos               1641869312 657454080 984415232  41% /zdata/Videos
zdata/circuitpython         984417280      2048 984415232   1% /zdata/circuitpython
zdata/Music                1026690048  42274816 984415232   5% /zdata/Music
zdata/pi-backups/lomond     987578368   3163136 984415232   1% /zdata/pi-backups/lomond
zdata/pi-backups2/pi4b      988699648   4284416 984415232   1% /zdata/pi-backups2/pi4b
zdata/ossian-backups       1113224192 128808960 984415232  12% /zdata/ossian-backups
zdata/pi-backups/tarmachan  986915840   2500608 984415232   1% /zdata/pi-backups/tarmachan
zdata/VMs                  1011632128  27216896 984415232   3% /zdata/VMs
tmpfs                          824688         0    824688   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		4194288		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20240123
openbox:
  Installed: (none)
lxpanel:
  Installed: (none)
[... 1154 lines cut ...]
[    4.999097] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    5.010743] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    5.012907] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.013809] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.049103] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 1
[    5.053832] spl: loading out-of-tree module taints kernel.
[    5.071546] [drm] Got external EDID base block and 1 extension from "edid/lr_tv.dat" for connector "HDMI-A-1"
[    5.099305] Console: switching to colour frame buffer device 240x67
[    5.117570] vc4-drm axi:gpu: [drm] fb0: vc4drmfb frame buffer device
[    5.150371] [drm] Got external EDID base block and 1 extension from "edid/lr_tv.dat" for connector "HDMI-A-1"
[    5.165998] [drm] Got external EDID base block and 1 extension from "edid/lr_tv.dat" for connector "HDMI-A-1"
[    5.296205] zfs: module license 'CDDL' taints kernel.
[    5.296213] Disabling lock debugging due to kernel taint
[    6.635627] ZFS: Loaded module v2.2.2-4~bpo12+1, ZFS pool version 5000, ZFS filesystem version 5
[   11.325698] Adding 4194288k swap on /var/swap.  Priority:-2 extents:13 across:7938048k SSFS
[   11.370724] Process accounting resumed
[   11.665490] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[   11.665501] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[   11.668010] pps pps0: new PPS source ptp0
[   11.668096] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   15.771319] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
[   15.886564] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   15.886575] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[  341.503583] [drm] Got external EDID base block and 1 extension from "edid/lr_tv.dat" for connector "HDMI-A-1"
[  341.535060] [drm] Got external EDID base block and 1 extension from "edid/lr_tv.dat" for connector "HDMI-A-1"

andrum993, 2024-02-03:
Oops - forgot the bootloader logs. Here's a packet capture of the NETCONSOLE output:

[intel_ssd_boot.zip](https://github.com/raspberrypi/rpi-eeprom/files/14151444/intel_ssd_boot.zip)

timg236, 2024-02-29:
The latest default bootloader release has a short delay in the NVMe initialisation which is required for some drives, particularly Western Digital. It might be worth trying that to see if it helps.

andrum993, 2024-03-01:
With bootloader EEPROM from 2024-02-16 the same thing happens - first attempt to boot from NVMe fails, then succeeds on the second attempt after trying SD card and USB.

andrum993, 2024-03-01:
Something weird is going on here - after playing about with it and logging the bootloader's debug output using the UART, I can actually boot successfully from the NVMe drive first time. Very odd. I will therefore close this one as it seems to have fixed itself 🤷

closed 2024-03-01 by andrum993

timg236, 2024-03-01:
It sounds like it is very sensitive to startup time, possibly variable if it's doing some maintenance on startup? Worth keeping an eye on. The UART output might slow things down enough that the NVMe firmware wakes up in time.

andrum993, 2024-03-01:
The strange thing is, it was still showing the problem for several reboots with bootloader debug logging enabled, then suddenly started working with it still enabled.

One thing I did notice is that now it is working, I see entries in the bootloader debug output about `PCI2 init` etc just after the first entry, then USB init, then the entries just before it boots correctly from the NVMe drive it mentions `PCI1 init` etc. It looks like PCI1 is the interface connected to the NVMe connector. Previously, when it was failing, there was no mention of PCI2 in the bootloader debug output. I have no idea what changed, but it looks to me like it is something in the Pi that has changed in terms of the PCIe initialisation. Why it would suddenly change I have no idea - all I have been doing with successive reboots is editing the BOOT_ORDER variable in the bootloader config.

peterharperuk, 2024-03-01:
The problem drive  would work ok if BOOT_UART=1 was set in the eeprom config. It had problems when logging was disabled. It just needed the delay that logging added to work properly. Anyway, I've ordered one to test here.

andrum993, 2024-03-01:
> The problem drive would work ok if BOOT_UART=1 was set in the eeprom config. It had problems when logging was disabled. It just needed the delay that logging added to work properly. Anyway, I've ordered one to test here.

Nope - it failed both with and without logging enabled - it only started working correctly several reboots after switching logging on: I spent a long time testing various different combinations of boot order.

Just switched bootloader logging off to confirm, and it correctly booted from the NVMe drive at the first attempt.

timg236, 2024-03-01:
Sorry can’t explain that :)

andrum993, 2024-03-01:
Can you explain whether PCI2 is the link that connects BCM2712 to the RP1?

In the case where the first boot attempt from NVMe failed, my logs show no messages related to PCI2 before the bootloader went onto the next boot mode after NVMe, and there is nothing in the log to show it even attempting boot from NVMe. Whereas in the case of boot success on first attempt from NVMe, the log shows PCI2 init-ed before boot from PCI1 ( apparently the interface connected to the NVMe drive) succeeds.

C0rn3j, 2024-03-03:
If it starts happening again, you could possibly try unplugging the fan and other USB devices if any.

The fan being plugged in prevents USB boot in some cases due to power contention, which may or may not be causing the issue.

Is that with the official Pi 5 adapter?

timg236, 2024-03-03:
Removed off topic  USB comments - this is about NVMe

timg236, 2024-03-03:
PCIe2 is RP1 - it is a separate RC and doesn't get re-initialised as part of the boot-order loop. If that fails you would an RP1 not found error.
If it didn't attempt to boot from NVMe then either it's missing from the BOOT_ORDER or more likely there's an issue with PCIe detect pins. It might be worth seeing if PCIE_PROBE=1 (aka ignore detect pins) helps.

andrum993, 2024-03-08:
> PCIe2 is RP1 - it is a separate RC and doesn't get re-initialised as part of the boot-order loop. If that fails you would an RP1 not found error. If it didn't attempt to boot from NVMe then either [...]

Thanks. The issue has resolved itself, but if it happens again I will give that a try. Maybe EMI was affecting the detect signal? I did have an issue in this location that I put down to EMI, but that was a while ago with a usb to usb + ethernet hub for the Zero.