raspberrypi/rpi-eeprom #540: Raspberry pi 5 in standby fan is running at full speed, usb is also on

Zuluuk, 2024-02-08:
### Describe the bug

The system is booting normally with no issue.

When the system is halt / shutdown and the power light is red. The fan is running at full speed. USB devices attached is still getting power because lights on the joystick is still on.

### Steps to reproduce the behaviour

I only have the official case and fan, this is not an active cooler. The usb controller i use is a dragon rise with LED switches.

1. Using raspberry os 64 bit, start up and go into gui << the fan at this time should just stop because there is no loads.
2. Shut down / halt the system. The red light power should come on.
3. At this time the fan for some reason goes full speed, the LED on the joystick is still on.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf14

BOOTLOADER: up to date
   CURRENT: Mon  5 Feb 14:38:34 UTC 2024 (1707143914)
    LATEST: Fri  5 Jan 15:57:40 UTC 2024 (1704470260)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

lam@raspberrypi:~ $ rpi-eeprom-config pieeprom.upd
ERROR: Failed to read 'pieeprom.upd'
[Errno 2] No such file or directory: 'pieeprom.upd'

### System

lam@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

lam@raspberrypi:~ $ vcgencmd bootloader_version
2024/02/05 14:38:34
version 0e79493d7768139f1868e8cdcf664962d56c62a8 (release)
timestamp 1707143914
update-time 1707409155
capabilities 0x0000007f

lam@raspberrypi:~ $ vcgencmd version
2024/02/05 14:38:34
Copyright (c) 2012 Broadcom
version 0e79493d (release) (embedded)

lam@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

### Bootloader logs

I dont have GPIO header for this log

### USB boot

lam@raspberrypi:~ $ lsusb -vvv

Bus 004 Device 002: ID 0781:5581 SanDisk Corp. Ultra
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
[... 382 lines cut ...]
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

### NVMe boot

Not used

### Network (TFTP boot)

Not used

Zuluuk, 2024-02-08:
Video demonstration: https://www.veed.io/view/6dd96d53-2d58-459a-bec2-7068367b8c44?panel=showcase

timg236, 2024-02-08:
Try setting POWER_OFF_ON_HALT=1 in the EEPROM config to see if that helps

Zuluuk, 2024-02-08:
> Try setting POWER_OFF_ON_HALT=1 in the EEPROM config to see if that helps

I tried this and it made no difference. I thought the same at the time, I was reading Jeff's guide and thought that it could be a power management issue.

timg236, 2024-02-08:
Sounds like an issue with the USB devices. See if it happens with that disconnected. Or possibly it's not shutting down at all so systemd / OS issue

Zuluuk, 2024-02-09:
Confirmed that I have nothing connected and POWER_OFF_ON_HALT=1 also. 

The camera work is a bit shaky. 

https://www.veed.io/view/cc337f04-b6d6-489a-89d2-f7a543f5bacd?panel=showcase

timg236, 2024-02-09:
It seems to be specific to your setup. Works on normal RPi OS mouse key board SSD.

We would need to see a uart log showing that it had entered shutdown.

It might be worth using a clean rpi  OS install and verifying that the power button boots the system when it is in this state.

Also are you using a Pi4 or Pi5 power supply ?

Zuluuk, 2024-02-09:
> It seems to be specific to your setup. Works on normal RPi OS mouse key board SSD.
>
> We would need to see a uart log showing that it had entered shutdown.
>
> It might be worth using a clean rpi OS [...]

It can't be my setup, as there are nothing connected except for the usb c power, there is no OS there. I dont have a microsd in the bay or do i have a usb.

If it is not going into shutdown with a powerbutton press and standby on with no os then its a bug. No other computer system does that.

I was in the batocera chat and another user had similar issues. I never had this issue before, it only happen recently when i tried to use the GPIO to test the power feature to program a power button, I confirmed with the dev that he hasn't implemented the GPIO for pi5 yet, so i rolled back. This is when I noticed the issue, and tried to update eeprom etc.

Please see attached. https://github.com/batocera-linux/batocera.linux/issues/10844

To be honest, I think I have a fault hardware if you dont think its the eeprom, the weird and wonderful world of early adoption.

timg236, 2024-02-09:
If the OS is not running then you have to hold the power button for at least 7 seconds to put the PMIC into standby. The short press is implemented by the OS in software.

By GPIO are you referring to the power button pads? It's not possible to use the 40-pin header for a power button on Pi5 because that's connected to RP1 and not the SOC.

There's a small chance that an issue with a USB-PD power supply could wakeup the board again, a plain USB-C power supply would rule that out.

My guess is that it has never entered STANDBY mode or was immediately woken up (hardware issue) because in STANDBY mode the PMIC disables 3.3V output (so fan will stop) and that's done in the PMIC IC itself.

Zuluuk, 2024-02-09:
I have forced a shutdown by holding the power button for 7s+.

For the gpio I gathered as much as I couldn't get to work, and reported it to the dev already.

I have tested a generic usb a to c charging block and my cable. The same happens when forcing into standby with the power button.

timg236, 2024-02-09:
Just to confirm the sequence

1. Disconnect power cable
2. DUT is powered on with no SD card and no peripherals and no RTC battery
3. Green LED is on
4. Press power button for 7 seconds
5. Green LED goes off
6. Red LED is on
7. Green LED immediately switches on  with USB and FAN active

If that is happening then either there is a hardware fault with the PMIC OR there is an issue with  USB-C power supply is resetting and causing DUT to power on.

Zuluuk, 2024-02-09:
> Just to confirm the sequence
>
>     1. Disconnect power cable
>
>     2. DUT is powered on with no SD card and no peripherals and no RTC battery
>
>     3. Green LED is on
>
>     4. Press power button for 7 [...]

The steps you mentioned is exactly how I have reproduced the issue. I have tested now with 3 different power brick, 1 official psu. All give me the same issue. As I have chatted with another user, he didn't experience this issue either, so I would have to say its a hardware issue.

pelwell, 2024-02-09:
I'm looking at the fan behaviour at the moment for other reasons using one of our active coolers, and with the latest bootloader EEPROM image (8th February 2024), `sudo poweroff` (or pushing the power button - twice if the GUI is running) causes the fan to stop, whether or not it was running at the time of shutdown. The POWER_OFF_ON_HALT only affects the USB power.

Zuluuk, 2024-02-09:
> I'm looking at the fan behaviour at the moment for other reasons using one of our active coolers, and with the latest bootloader EEPROM image (8th February 2024), `sudo poweroff` (or pushing the [...]

I will try this later and reply directly. I don't think there's an issue with the fan controller. When I am in the os, the fan does not even start 😂. I also never had this issue before as I remember I shutdown and everything was pretty gucci, fan off, usb light off etc. This only happen recently when I tried the gpio and updated the eeprom to rule out any bugs. I am happy to test and help out as I do believe that contribution is why and how the raspberry pi have been so successful.

timg236, 2024-02-09:
> The steps you mentioned is exactly how I have reproduced the issue. I have tested now with 3 different power brick, 1 official psu. All give me the same issue. As I have chatted with another user, [...]

A couple of questions for hardware engineers.

- Please could we have a photo of the top side of the board showing the PMIC
- Does this still happen if the case fan is not connected - i.e. immediate wakeup

Zuluuk, 2024-02-09:
> A couple of questions for hardware engineers.
>
>     * Please could we have a photo of the top side of the board showing the PMIC
>
>     * Does this still happen if the case fan is not connected - i.e. [...]

Hi timg236,
https://www.veed.io/view/d4280b33-0f48-4f17-8a25-0dd71906ad44?panel=share this is the check from red to green. It's not instant, it's approximately 1s.

Photo of the board attached. As I don't have macro mode. I took a few just in case some details are missing. 
[image]
[image]
[image]
[image]

Zuluuk, 2024-02-09:
> I'm looking at the fan behaviour at the moment for other reasons using one of our active coolers, and with the latest bootloader EEPROM image (8th February 2024), `sudo poweroff` (or pushing the [...]

Hi pelwell,
When I am in the os the fan stopped. As soon as I entered that command the light turn red and the fan goes full speed.

pelwell, 2024-02-09:
I understand - I was just confirming that something is not right with your system.

timg236, 2024-02-09:
I checked the code and USB + RP1 are not initialised until after DDR init so not quite instant. Something is resetting the PMIC and causing it to leave standby, probably hardware related but I can't think what.

Zuluuk, 2024-02-09:
> Out of interest, what does it say on the chip on the bottom left of the board by the USB-C port (the PMIC). Probably not significant though!

Not sure if you able to see this
[image]

timg236, 2024-02-09:
Thanks. It’s the part that we would expect so that’s not the issue. This failure mode has got me stumped. 
I’ll follow on Monday about whether we could get the board back (and send you replacement etc) if you at happy to do that and logistics are feasible.

Zuluuk, 2024-02-10:
> Thanks. It’s the part that we would expect so that’s not the issue. This failure mode has got me stumped. I’ll follow on Monday about whether we could get the board back (and send you replacement etc) if you at happy to do that and logistics are feasible.

Happy to help, I live in Dublin(Ireland), do let me know. At this time i am also waiting for pihut to reply to my rma request, which is taking a while to even have someone engaged.

timg236, 2024-02-12:
@Zuluuk Please can you contact info@raspberrypi.com mentioning this issue with your name, email and phone number for shipping - add "Tim Gover"

For this Pi5 please don't use the RMA process, we want to get the board directly and as soon as we receive it we can send you a replacement board plus something to compensate for postage.

If you contact info@raspberrypi.com with your name, address, email and phone number mentioning this bug we can sort it out via email.

Zuluuk, 2024-02-12:
> "Tim Gover"

Thanks Tim, I have sent the email to info. At the weekend, I have checked with the developer for Batocera. He has confirmed that there are customisation / flashing of the eeprom from any of his image. During my conversation another user also reported a similar issue to mine. Hope you can rootcause it.

I will update pihut now to cancel my RMA.

Have a good day.

timg236, 2024-02-15:
The board arrived. Thanks!

The issue was related to a hardware fault related to the USB current limiter / VBUS switch. This is why the FAN was on when the PMIC is in standby since USB VBUS is not a PMIC output, it's from 5V but goes via a switch.

Closing this as a software bug and the board is with hardware team. 

Please let us know if there's any issue with the replacement since you have my email.

closed 2024-02-15 by timg236

Lewiscowles1986, 2024-08-06:
I'd like to understand this so that I can make mine spin all the time. I Altered /boot/firmware/config.txt to lower the thresholds all to 1000, and set the fan speed value to 255. When it's powering on or off, I get fan. When it is not, I've not seen the fan doing anything, and I think it's because someone over at RaspberryPi "was helping" (and should be prevented from helping anyone else).

I want the fan. Even when it is not needed, all the time!

pelwell, 2024-08-06:
> someone over at RaspberryPi "was helping" (and should be prevented from helping anyone else)

Wow.

lurch, 2024-08-06:
> I want the fan. Even when it is not needed, all the time!

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#fan-connector-pinout shows the fan-pinout, so I guess if you want it running all the time the easiest way of achieving that would be to connect the 5V and GND wires of the fan directly to the 5V and GND pins on the Pi's main GPIO header? Although I've not tried this myself, so I dunno if the fan would actually work when wired like that.

pelwell, 2024-08-06:
> I guess if you want it running all the time the easiest way of achieving that would be to connect the 5V and GND wires of the fan directly to the 5V and GND pins on the Pi's main GPIO header?

Nope - even when the fan is stopped, the 5V signal will be on - changing it to a different 5V source won't make any difference. The fan has a speed control signal that is normally driven by a PWM. Driving it low will cause the fan to constantly spin at maximum speed:
```
pi@raspberrypi:~$ sudo pinctrl | grep FAN
29: no    pu | hi // FAN_TACH/GPIO29 = none
45: a0    pd | hi // FAN_PWM/GPIO45 = PWM1_CHAN3
pi@raspberrypi:~$ sudo pinctrl FAN_PWM op dl
```
Note that this will need to be done after each reboot. If you want to restore the kernel driver's control of the fan speed:
```
pi@raspberrypi:~$ sudo pinctrl FAN_PWM a0
```

Lewiscowles1986, 2024-08-08:
> sudo pinctrl FAN_PWM op dl

This was the trick. Apologies I missed all the replies. I got it from stackoverflow same day I posted this and forgot to come back here.

I use systemd to start it on boot. I was hoping that there would be a systemd service, or other alteration I can run to prevent the attempt at being helpful (which I still don't appreciate)

joeuhren, 2024-10-11:
> The board arrived. Thanks!
>
> The issue was related to a hardware fault related to the USB current limiter / VBUS switch. This is why the FAN was on when the PMIC is in standby since USB VBUS is not a [...]

@timg236 Curious that this is mentioned as a software bug. Are there steps I can take to stop this from happening to my rpi5? It was working great for months and now out of the blue the fan spins at full blast and doesn't stop when I shut down. Everything else seems to work fine. Running newest eeprom from Sep-23 tried POWER_OFF_ON_HALT=1 all the same troubleshooting as above but can't seem to stop this behavior.

ToKe79, 2025-02-28:
First off all sorry for reviving this closed issue.

@joeuhren: As the issue of the OP is **not** a software/firmware bug, the issue was closed and handed over to the HW dev team. So there is nothing in SW/FW (I suppose) to fix this behavior.

@timg236: If I may ask, what is the recommended route to handle these issues for other users? Should they RMA the board where it was purchased, or should they contact Raspberry directly? I have a friend in the US with what looks same issue - after running `shutdown` or `poweroff` the OS turns off, green led switches to red and the fan starts spinning at 100 %. Thank you in advance for any response, even a negative one.

forkymcforkface, 2025-03-05:
Ive been having this same issue, any fixes other than RMA? Sadly I bought this right when it came out and is now out of warranty. Id love for my pi5 to be able to shutdown without the fans going full blast.

elcatedratico, 2025-03-28:
I have the same problem as mentioned before. 
As soon as I power off the system, the fan starts spinning full power. There is no other hardware than the original power supply connected. 
Don't know what to do now.

psp631, 2025-04-28:
Is your problem solved? I'm having the exact same issue.

elcatedratico, 2025-04-28:
In fact, the problem has been solved. I sent the Pi back to the manufacturer and there the problem was checked. As already written by many users, it is a hardware defect.

I was able to complain about the Pi and got a new one. This works flawlessly.

forkymcforkface, 2025-04-28:
I was able to solve this by doing a full eeprom and firmware update