raspberrypi/rpi-eeprom #541: Support booting "zboot" kernel images?

kemurphy, 2024-02-11:
### Describe the bug

Fedora IoT ships kernels using some EFI "zboot" format, which doesn't appear to be recognized by the rpi bootloader.  Looks like it should be easy to support though.  Here are a few links:

https://www.phoronix.com/news/Linux-6.1-Generic-EFI-Zboot
https://lwn.net/Articles/929431/
https://www.mail-archive.com/qemu-devel@nongnu.org/msg947068.html

### Steps to reproduce the behaviour

Download the kernel and initrd at https://mirror.fcix.net/fedora-alt/iot/39/IoT/aarch64/os/images/pxeboot/ and try to boot them (I used PXE but presumably it's the same result directly from SD card).  Expect to boot Linux and instead watch a blinking cursor forever.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400

### Bootloader configuration.

kernel=vmlinuz
initramfs initrd.img

timg236, 2024-02-11:
gzip is supported

closed 2024-02-11 by timg236

kemurphy, 2024-02-14:
Hi sorry I think you misunderstand -- it's not gzip, it's that there's now different header magic in the kernel image format that needs to be parsed.

timg236, 2024-02-14:
It would be in start4.elf but there's no EFI entry points so I can't see how this can work. That would explain why you get the flashing cursor, kernel is running but stuck. 
Maybe in future this could happen if the ARM stubs (on Pi5 now ARM TFA firmware) were extended to implement some of the EFI stubs.

kemurphy, 2024-02-15:
Well, not quite, yes EFI is one way to run them but the format leaves room
for non-EFI bootloaders to parse the header and decompress the payload
directly if the compression algorithm is supported.

timg236, 2024-02-15:
Yes. I can’t see us adding that for Pi4.