raspberrypi/rpi-eeprom #543: Raspberry PI 5 issue with TV Hat

clementmoine, 2024-02-13:
### Describe the bug

Several Raspberry Pi 5 users with the TV Hat have reported issues, such as kernel panics and crashes, when the TV Hat is installed on the Pi 5 and when tvheadend or any other program attempts to access /dev/dvb. I am one of them; when I plug in the module, the Pi starts up normally, but as soon as tvheadend is launched, the Pi freezes and becomes inaccessible.

A thread is open on the raspberry pi forum but we can't get any help from there. I can't assure that issue come from the firmware but dmseg logs seems to confirm that. 

I tried moving the SDCard from RPI 5 to RPI 4, everything works as expected. Turning back to the RPI 5 get the PI frozen after seconds. 

https://forums.raspberrypi.com/viewtopic.php?t=359876 

### Steps to reproduce the behaviour

- Proper new install of Rapsberry Pi 5 with Raspberry Pi Imager.
- Install the TV Hat on the Rapsberry PI 5
- Install tvheadend
- Run tvheadend
- Raspberry PI will freeze after seconds
- Running `dmseg` gives some errors about /dev/dvb read overflow.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf41

### System

https://pastebin.com/raw/tEmhhAAK

timg236, 2024-02-13:
The firmware doesn’t get involved with TV HAT stuff. It’s a pure Linux driver

closed 2024-02-13 by timg236

clementmoine, 2024-02-13:
Oh ok ! I was not sure ! Thanks !

clementmoine, 2024-02-13:
Moved the issue to the 

> @clementmoine In which case, I guess you'll want to report this at https://github.com/raspberrypi/linux/issues instead.

Moved the issue to the proper repo thank you !

 https://github.com/raspberrypi/linux/issues/5953