raspberrypi/rpi-eeprom #552: Raspiblitz Enable of device-initiated u1 failed / ext4-fs error (device sda1) ext4_find_entry

fabianp99, 2024-03-06:
### Describe the bug

After clicking reboot the raspiblitz shut down and don't start again. On the LCD there is a text :Enable of device-initiated u1 failed. Sometimes there is also a text : ext4-fs error (device sda1) ext4_find_entry. 

### Steps to reproduce the behaviour

reboot the raspiblitz

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 07:38:04 up 5 min,  3 users,  load average: 3.68, 2.45, 1.11

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-03-06 07:35:53 GMT; 2min 11s ago
    Process: 6848 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6887 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 7057 (bitcoind)
      Tasks: 19 (limit: 9352)
        CPU: 2min 21.721s
     CGroup: /system.slice/bitcoind.service
             └─7057 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Mar 06 07:33:52 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Mar 06 07:35:53 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Tue 2024-03-05 00:34:10 GMT, ends at Wed 2024-03-06 07:38:05 GMT. --
Mar 06 07:33:52 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Mar 06 07:35:53 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-03-06T07:35:30Z Verification: No coin database inconsistencies in last 6 blocks (15460 transactions)
2024-03-06T07:35:30Z  block index           97251ms
2024-03-06T07:35:30Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2024-03-06T07:35:31Z Opened LevelDB successfully
2024-03-06T07:35:31Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2024-03-06T07:35:31Z txindex thread start
2024-03-06T07:35:31Z txindex is enabled at height 833379
2024-03-06T07:35:31Z txindex thread exit
2024-03-06T07:35:31Z Using SQLite Version 3.38.5
2024-03-06T07:35:31Z Using wallet /mnt/hdd/bitcoin/satstack
2024-03-06T07:35:31Z init message: Loading wallet…
2024-03-06T07:35:31Z [satstack] Wallet file version = 10500, last client version = 250000
2024-03-06T07:35:31Z [satstack] Setting spkMan to active: id = d102fa83270b045196246cf04661314e1f565814463205c5aaa58e5607e497a1, type = legacy, internal = false
2024-03-06T07:35:31Z [satstack] Setting spkMan to active: id = 6204c82845bee0fba256762dfbf4a6b95dc6756d6623db74d83b814b35234ab7, type = p2sh-segwit, internal = false
[... 387 lines cut ...]
ln_default_tor="0"
ln_default_error_short=""
ln_default_channels_pending="0"
ln_default_channels_active="0"
ln_default_version="0.16.4-beta"
btc_default_activated="1"
btc_default_error_short=""
btc_default_sync_progress="0.9999918235004017"
btc_default_running="1"
btc_default_synced="1"
btc_default_mempool_transactions="7589"
btc_default_blocks_behind="0"
btc_default_sync_initialblockdownload="0"
btc_default_peers="11"
btc_default_error_full=""
btc_default_blocks_data_kb="614803368"
btc_default_blocks_headers="833379"
btc_default_online="1"
btc_default_port="8333"
btc_default_ready="1"
btc_default_sync_initial_done="1"
btc_default_version="v25.0.0"
btc_default_blocks_verified="833379"
btc_default_sync_percentage="100.00"
*** POSSIBLE ERROR REPORTS ***

timg236, 2024-03-06:
If the OS is booting then it's not a bootloader issue. Please see forums for interop issues / other distros

fabianp99, 2024-03-06:
After rebooting I can't build a connection with ssh to the Raspiblitz. The only thing I can do is to unplug the Raspiblitz and start IT again and then everything works.