raspberrypi/rpi-eeprom #554: Cannot use rpi-eeprom-update under mainline kernel

ArchGuyWu, 2024-03-17:
### Describe the bug

`[root@ArchRPi ~]# rpi-eeprom-config
BOOT_UART=1
BOOT_ORDER=0xe1

[root@ArchRPi ~]# rpi-eeprom-update
Device does not a have a Raspberry Pi bootloader EEPROM (e.g. Pi 4 or Pi 5). Skipping bootloader update.
[root@ArchRPi ~]# uname -a
Linux ArchRPi 6.7.9-1-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Tue Mar 12 11:20:09 MDT 2024 aarch64 GNU/Linux`

### Steps to reproduce the behaviour

`[root@ArchRPi ~]# rpi-eeprom-config
BOOT_UART=1
BOOT_ORDER=0xe1

[root@ArchRPi ~]# rpi-eeprom-update
Device does not a have a Raspberry Pi bootloader EEPROM (e.g. Pi 4 or Pi 5). Skipping bootloader update.
[root@ArchRPi ~]# uname -a
Linux ArchRPi 6.7.9-1-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Tue Mar 12 11:20:09 MDT 2024 aarch64 GNU/Linux`

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

unrelated

### System

archlinux arm

### Bootloader logs

unrelated

### USB boot

unrelated

### NVMe boot

unrelated

### Network (TFTP boot)

unrelated

timg236, 2024-03-18:
This is something that you need to take up with the maintainer of that distro

ArchGuyWu, 2024-03-18:
> This is something that you need to take up with the maintainer of that distro

But I just changed linux-rpi to linux-aarch64, and other tools like fastfetch can recognize the host is rpi4b under linux-aarch64.

lurch, 2024-03-18:
> But I just changed linux-rpi to linux-aarch64, and other tools like fastfetch can recognize the host is rpi4b under linux-aarch64.

Feel free to have a go at getting https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L377 working with your choice of distro and kernel, and we _may_ consider including it if it doesn't break compatibility with Raspberry Pi OS.

ArchGuyWu, 2024-03-18:
> > But I just changed linux-rpi to linux-aarch64, and other tools like fastfetch can recognize the host is rpi4b under linux-aarch64.
> 
> Feel free to have a go at getting https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L377 working with your choice of distro and kernel, and we _may_ consider including your change if it doesn't break compatibility with Raspberry Pi OS.

I found that fastfetch uses [/sys/firmware/devicetree/base/model](https://github.com/fastfetch-cli/fastfetch/blob/7c5eb021ee81e42a77b1eae73c9691e23cdd12ab/src/detection/host/host_linux.c#L10) to check the host, and uses /sys/firmware/devicetree/base/serial-number to check serial number, maybe this also works for Raspberry Pi OS.

ArchGuyWu, 2024-03-18:
The result on my rpi4:
cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.4
cat /sys/firmware/devicetree/base/serial-number
100000000f1b63d1

zetxx, 2024-04-29:
Same issue here.
2 Rpis with 8Gb mem, everything is ok
1 Rpi with memory 4Gb: Device does not a have a Raspberry Pi bootloader EEPROM (e.g. Pi 4 or Pi 5). Skipping bootloader update.
also, when i try: cat: /sys/firmware/devicetree/base/system/linux,revision: No such file or directory: No such file or directory
and from 8Gb versions: cat /scat /sys/firmware/devicetree/base/system/linux,revision                                                                                                                                                                         1 ↵
�1

lurch, 2024-04-29:
@zetxx The `rpi-eeprom` tool has been written by Raspberry Pi, and we only officially support Raspberry Pi OS. If you want to see it working on ArchLinux, you'll have to ask the ArchLinux developers to create and test a patch, and get them to create a Pull Request to this repo.

AdelKS, 2026-09-06:
For anyone who finds this from a search: if your Pi boots through U-Boot, the fix is a small change to the U-Boot boot script. It takes one edit and one reboot.

`rpi-eeprom-update` looks up your board in the device tree. The Raspberry Pi firmware writes that information, but U-Boot then loads its own device tree from disk and the information is lost. Tell U-Boot to carry some values across.

On Arch Linux ARM the boot script is `/boot/boot.txt`, shipped by `uboot-raspberrypi`. Here is the change against the stock file:

```diff
--- a/boot/boot.txt
+++ b/boot/boot.txt
@@ -7,6 +7,18 @@

 if load ${devtype} ${devnum}:${bootpart} ${kernel_addr_r} /Image; then
   if load ${devtype} ${devnum}:${bootpart} ${fdt_addr_r} /dtbs/${fdtfile}; then
+    fdt addr ${fdt_addr_r}
+    fdt resize 8192
+    if fdt addr ${fdtcontroladdr}; then
+      fdt get value fwrev /system linux,revision
+      fdt get value blts /chosen/bootloader build-timestamp
+      fdt addr ${fdt_addr_r}
+      fdt mknode / system
+      fdt set /system linux,revision <${fwrev}>
+      fdt mknode /chosen bootloader
+      fdt set /chosen/bootloader build-timestamp <${blts}>
+    fi
+    fdt addr ${fdt_addr_r}
     if load ${devtype} ${devnum}:${bootpart} ${ramdisk_addr_r} /initramfs-linux.img; then
       booti ${kernel_addr_r} ${ramdisk_addr_r}:${filesize} ${fdt_addr_r};
     else
```

Rebuild the compiled script and reboot:

```
cd /boot && sudo ./mkscr
sudo reboot
```

`rpi-eeprom-update` should now work:

```
$ sudo rpi-eeprom-update
BOOTLOADER: update available
   CURRENT: Fri May 17 11:26:58 UTC 2024 (1715945218)
    LATEST: Sun May 17 19:13:18 UTC 2026 (1779045198)
   RELEASE: default

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

Tested on a Pi 4B rev 1.1 with Arch Linux ARM, `linux-aarch64 7.2.3` and
`uboot-raspberrypi-2025.01`