raspberrypi/rpi-eeprom #559: Raspberry Pi 4 PXE boot client not able to acquire IP address from dnsmasq

aggo15, 2024-04-05:
### Describe the bug

I tried to boot a Raspberry Pi 4 from network. Two Pis are involved in this situation and connected via ethernet, both are same hardware, refer to sections below for more details about the setup and config. My observation below:

1. When PXE Client boot. By right it will try to acquire IP address from dnsmasq hosted in another Raspberry Pi 4, but it fail to do so.
2. After 2 attempts it will boot into SD card. When it boot into SD card, it can get IP address from dnsmasq without issue.
    1. Here is the log display on the screen
      ```
      Boot mode: NETWORK (02) order f145
      Boot mode: NETWORK (02) order f145
      USB2[1] 400202e1 connected
      USB2 root HUB port 1 init
      NETWORK e4:5f:01:1f:02:a5 wait for link TFTP: 0.0.0.0
      Link ready
      DHCP src: e4:5f:01:0d:24:96 192.168.25.1
      DHCP src: e4:5f:01:0d:24:96 192.168.25.1
      NETBOOT init failed
      Boot mode: BCM-USB-MSD (05) order f14
      ```
5. Based on the log from dnsmasq, it did respond to PXE client DHCP request with information below:
    1. dnsmasq log:
      ```
      Apr  5 05:49:00 dnsmasq[32]: started, version 2.85 cachesize 150
      Apr  5 05:49:00 dnsmasq[32]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP 
      conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
      Apr  5 05:49:00 dnsmasq[32]: warning: no upstream servers configured
      Apr  5 05:49:00 dnsmasq-dhcp[32]: DHCP, proxy on subnet 192.168.25.255
      Apr  5 05:49:00 dnsmasq-dhcp[32]: DHCP, IP range 192.168.25.3 -- 192.168.25.10, lease time 1d
      Apr  5 05:49:00 dnsmasq-tftp[32]: TFTP root is /source/dist/tftpboot
      Apr  5 05:49:00 dnsmasq[32]: cleared cache
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 available DHCP range: 192.168.25.3 -- 192.168.25.10
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 available DHCP subnet: 192.168.25.255/255.255.255.0
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 vendor class: PXEClient:Arch:00000:UNDI:002001
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 PXE(eth0) e4:5f:01:1f:02:a5 proxy
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 tags: eth0
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 broadcast response
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 sent size:  1 option: 53 message-type  2
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 sent size:  4 option: 54 server-identifier  192.168.25.1
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:14:31:b0:00:01:1f:02:a5:70...
      Apr  5 05:49:54 dnsmasq-dhcp[32]: 1957337301 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 available DHCP range: 192.168.25.3 -- 192.168.25.10
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 available DHCP subnet: 192.168.25.255/255.255.255.0
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 vendor class: PXEClient:Arch:00000:UNDI:002001
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 PXE(eth0) e4:5f:01:1f:02:a5 proxy
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 tags: eth0
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 broadcast response
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 sent size:  1 option: 53 message-type  2
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 sent size:  4 option: 54 server-identifier  192.168.25.1
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:14:31:b0:00:01:1f:02:a5:70...
      Apr  5 05:50:10 dnsmasq-dhcp[32]: 1957337301 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
      ```
4. For verification, I've tried to manually set IP address via bootloader config CLIENT_IP and it is able to boot into network without issue. At this step it is pretty obvious the PXE client is not able to get IP address from dnsmasq for some reason.

I've already upgrade the bootloader to latest available binary in "firmware-2711" directory on this repo before attempt to netboot. Please refer to sections below for more details.

### Steps to reproduce the behaviour

Here is the setup for server and client that is having issue.

# Server config
Install dnsmasq on another Raspberry Pi 4 as PXE boot server with config below:

```
interface=eth0
domain-needed
bogus-priv
no-resolv
no-hosts
[... 182 lines cut ...]
NA

### NVMe boot

NA

### Network (TFTP boot)

dnsmasq config below:
```
interface=eth0
domain-needed
bogus-priv
no-resolv
no-hosts
log-dhcp
enable-tftp
tftp-root=/source/dist/tftpboot
pxe-service=0,"Raspberry Pi Boot"
dhcp-range=eth0,192.168.25.3,192.168.25.10,1d
dhcp-range=192.168.25.255,proxy
dhcp-option=6,192.168.25.1
dhcp-option=1,255.255.255.0
dhcp-option=3,192.168.25.1
```

timg236, 2024-04-05:
Network boot has been stable for some years now on Pi4 so this sounds like something specific with this setup. 

It's probably worth starting with this tutorial as a baseline and adding in features as necessary to work out what's different
https://www.raspberrypi.com/documentation/computers/remote-access.html#network-boot-your-raspberry-pi

aggo15, 2024-04-08:
@timg236 I figured out the issue with the setup. Seems like there is an additional line in dnsmasq config that is not required. 

```
dhcp-range=192.168.25.255,proxy
```

After remove the line above, the PXE client able to acquire IP address without issue. 

Since it is not related to the bootloader, I'll close the issue.

closed 2024-04-08 by aggo15

lurch, 2024-04-08:
Yeah, DHCP proxy mode is used when you already have _another_ DHCP server (e.g. a home broadband router) on the same network, and not when dnsmasq is the _only_ DHCP server on the network. See https://wiki.archlinux.org/title/dnsmasq#Proxy_DHCP and https://wiki.fogproject.org/wiki/index.php?title=ProxyDHCP_with_dnsmasq#How_ProxyDHCP_works for more details.

@nathan-contino Perhaps this is worth a small clarification NOTE in the documentation that Tim pointed to above?

aggo15, 2024-04-09:
Perhaps a little note on the code like below would be nice.

Current documentation:
```
Then replace the contents of dnsmasq.conf with:

# Note: comment out port if you want DNS services for systems on the network.
port=0
dhcp-range=10.42.0.255,proxy
log-dhcp
enable-tftp
tftp-root=/tftpboot
pxe-service=0,"Raspberry Pi Boot"
```

Change to:
```
Then replace the contents of dnsmasq.conf with:

# Note: comment out port if you want DNS services for systems on the network.
port=0
# Note: comment out line below if your are using dnsmasq as the sole DHCP server in the network
dhcp-range=10.42.0.255,proxy
# Note: Uncomment line below if you are using dnsmasq as the sole DHCP server in the network and
#       you like to limit the range of IP offered to PXE clients.
# dhcp-range=eth0,10.42.0.10,10.42.0.20,30d
log-dhcp
enable-tftp
tftp-root=/tftpboot
pxe-service=0,"Raspberry Pi Boot"
```

lurch, 2024-04-09:
@aggo15 Please feel free to submit a PR to https://github.com/raspberrypi/documentation ? :slightly_smiling_face:

aggo15, 2024-04-15:
@lurch I don't have the experience doing the documentation update but I love to learn how to do it. Currently reading the CONTRIBUTING.md, will create a pull request if everything goes well on my end.