raspberrypi/rpi-eeprom #571: rpi-eeprom:arm64 update from 21.6-1 to 22.0-1 leads to extreme high lags and 100% GPU load in chromium-browser and firefox

qrp73, 2024-04-19:
### Describe the bug

After update rpi-eeprom:arm64 22.0-1 it starting to lags like hell, sometimes chromium browser freezes for 10-20 seconds and don't respond even on close tab

With rpi-eeprom:arm64 21.6-1 it worked much more smooth and without these long many second freezes

Firefox also affected, I notice very long freezes (the same as for chromium-browser) in firefox despite the fact that hw acceleration is disabled. Before update rpi-eeprom to 22.0-1 I never seen so long freezes in firefox with disabled hw acceleration, it always worked smooth.

### Steps to reproduce the behaviour

Steps to reproduce:
1) use system with rpi-eeprom:arm64 21.6-1
2) run sudo apt update && sudo apt upgrade to update rpi-eeprom to the latest 22.0-1 version
3) open chromium-browser (Version 122.0.6261.89)
4) open some page with videos, for example: https://t.me/Nature/13536

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
$ rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
```

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd bootloader_version
2024/04/17 13:51:36
version 0b7b6f28b0a25311e6d75518aec0e29ecfb36ec0 (release)
timestamp 1713358296
update-time 1713449182
capabilities 0x0000007f

$ vcgencmd version
Feb 29 2024 12:24:53 
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)

$ uname -a
Linux raspi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```

qrp73, 2024-04-19:
Is there a way to downgrade rpi-eeprom to 22.0-1 until fix will be issued?

pelwell, 2024-04-19:
How certain are you that the EEPROM change is to blame (and why), given that it sounds as though you haven't managed to revert it?

timg236, 2024-04-19:
It’s possible that you are seeing some cpu usage increase whilst flashrom is running and doing the upgrade. About 20 seconds worst case

qrp73, 2024-04-19:
> How certain are you that the EEPROM change is to blame (and why), given that it sounds as though you haven't managed to revert it?

Before last update there was no lags. It starts to happens after last update, which include rpi-eeprom update:
```
Start-Date: 2024-04-18  17:05:38
Commandline: apt upgrade
Requested-By: pi (1000)
Upgrade: rpi-eeprom:arm64 (21.6-1, 22.0-1), firefox:arm64 (124.0.1-1+rpt4, 125.0.1-1+rpt4), firefox-esr:arm64 (115.9.1esr-1~deb12u1, 115.10.0esr-1~deb12u1)
End-Date: 2024-04-18  17:06:25
```

It also include firefox, but the main issue with lags appears in chromium-browser. 
The extreme high lags with freezes for 10-20 seconds happens in chromium-browser which is not changed.

The issue happens on certain pages which include videos, like telegram. 
For example, this link leads to 96% GPU load: https://t.me/s/channels_tg?q=%23%D0%9F%D1%80%D0%B8%D1%80%D0%BE%D0%B4%D0%B0&before=3607

But this link is not worse case, it don't stops mouse cursor movements like it happens on other pages and don't leads to long freezes.
I will post another example links later, needs to find where it is reproduced very noticeable with a long app freezes.

qrp73, 2024-04-19:
> It’s possible that you are seeing some cpu usage increase whilst flashrom is running and doing the upgrade. About 20 seconds worst case

There is no high CPU usage, it shows high 100% GPU usage during these lags and freezes.

qrp73, 2024-04-19:
Just to be clear, there is no throttling and no voltage issues. CPU temperature is below 40°C. My rpi4 is placed into metal shell with a large heatsink on the top. During long run ```stress -c4``` temperature don't exceed 50°C.

This is software issue and somehow related to video decoding, because it always happens on web pages with videos. 
Needs to investigate it more deep to find better pages to show the worse case of this issue.

timg236, 2024-04-19:
Thanks. It would be useful to know the last version that works as expected.

qrp73, 2024-04-19:
> cd /lib/firmware/raspberrypi/bootloader-2712/latest sudo rpi-eeprom-config --edit pieeprom-2024-02-16.bin

OMG! it broke my system :(

I executed these commands:
```
$ cd /lib/firmware/raspberrypi/bootloader-2712/latest
$ sudo rpi-eeprom-config --edit pieeprom-2024-02-16.bin
```

and I cannot boot anymore... It shows red screen just after power on.

I booted from another sdcard and read your command more careful and it appears that I set rpi-eeprom for RPI5 (chip 2712), but I'm using RPI4 (chip 2711)... Is it correct?

Is it possible to restore the system on my sdcard? 
I can modify card content with card reader while booted in bullseye from another sdcard...

timg236, 2024-04-19:
Yes use the 2711 bootloader. Remove files called 'recovery.bin' 'pieeprom.*' from the boot partition

timg236, 2024-04-19:
I see exactly the same GPU load behavior on rpi-eeprom version 2023-05-11 (vcgencmd bootloader_version).  

This makes sense because the only job of the bootloader on Pi4 is to load start4.elf. After that it's discarded so it's unlikely that could be the root cause.

If you have a reference known good system then it might be worth comparing the kernel and start4.elf versions
"uname -a" and
"vcgencmd version"

donteatsoclose2me, 2024-04-19:
I can add anecdotal evidence to support the OP's claim.  I use my RPi-5 as a zoneminder server.  After the latest update of rpi-eeprom (21.6-1 -> 22.0.1), the video decoding, live feeds and recorded video are awful -- choppy, dropped frames...  At first I though it was the zoneminder update, so I rolled that back.  But, the problem remains.  'top' shows 2.0 idle, where it used to have >50% idle.

This extreme drop in performance directly correlates to the rpi-eeprom update.  I'd roll that back if I knew how.

qrp73, 2024-04-19:
In order to restore the system I copied files from bootfs partition to backup, then replace it with the same files from sdcard with working os. It allowed me to boot the system, then I applied eeprom update from proper folder 2711 and compared what files are changed, it shows that it add pieeprom.sig and pieeprom.upd. So I restored original filed for bootfs and removed these files. It works, now it boot ok.

I tried to rollback rpi-eeprom for my RPI4 with these commands:
```
$ cd /lib/firmware/raspberrypi/bootloader-2711/latest
$ sudo rpi-eeprom-config --edit pieeprom-2024-04-15.bin
```

after that it rebooted ok and it seems that these lags disappeared. 
It still shows high GPU load on problematic pages, but it shows a little bit shorter load and there is no high lags like it happens with last update. 

I will see how it works more and let you know about results.

timg236, 2024-04-19:
> I can add anecdotal evidence to support the OP's claim. I use my RPi-5 as a zoneminder server. After the latest update of rpi-eeprom (21.6-1 -> 22.0.1), the video decoding, live feeds and recorded [...]

The thread already describes how to use specific versions of the boot loader.
Since Pi4 bootloader has no involvement in GPU drivers (it’s only a loader) the most likely explanation is that other system packages were updated at the same time.

donteatsoclose2me, 2024-04-19:
> The thread already describes how to use specific versions of the boot loader. Since Pi4 bootloader has no involvement in GPU drivers (it’s only a loader) the most likely explanation is that other system packages were updated at the same time.

cat /var/log/apt/history.log
(pertinent entry):
Start-Date: 2024-04-19  09:16:56
Commandline: apt upgrade
Upgrade: rpi-eeprom:arm64 (21.6-1, 22.0-1), apache2-data:arm64 (2.4.57-2, 2.4.59-1~deb12u1), apache2-bin:arm64 (2.4.57-2, 2.4.59-1~deb12u1), zoneminder:arm64 (1.37.59~20240415.28-bookworm, 1.37.59~20240417.31-bookworm), apache2-utils:arm64 (2.4.57-2, 2.4.59-1~deb12u1), zoneminder-doc:arm64 (1.37.59~20240415.28-bookworm, 1.37.59~20240417.31-bookworm), apache2:arm64 (2.4.57-2, 2.4.59-1~deb12u1)
End-Date: 2024-04-19  09:17:41

rpi-eeprom:arm64 (21.6-1, 22.0-1) seems to be the only system package that was updated.

timg236, 2024-04-19:
The op reported an issue on Pi4 and Pi5 is a completely different chip and firmware.
It does not appear to be reproducible because the GPU load is equally high on the previous firmware.
Marking as Awaiting Information until someone can indicate an objective benchmark eg Geekbench where there is a clear difference in performance between two versions of the boot loader with the exactly the same start4.elf and kernel

qrp73, 2024-04-20:
After some testing I found that this issue is floating. With exactly the same steps to reproduce it can lead to a long system freeze in some cases, but if I try to repeat it with the same steps I can't reproduce it.

I tried to switch from pieeprom-2024-04-15.bin to pieeprom-2024-04-17.bin and back and tested them. In most cases it works almost the same. But in some cases I catch long system freeze (3-5 seconds) and even complete system halt (display image is freezed, no response to mouse and keyboard, don't respond on network connections and there is no errors in logs). And it happened when pieeprom-2024-04-17.bin were applied. I didn't catch the same issue with pieeprom-2024-04-15.bin.

timg236, 2024-04-20:
> > The thread already describes how to use specific versions of the boot loader. Since Pi4 bootloader has no involvement in GPU drivers (it’s only a loader) the most likely explanation is that other system packages were updated at the same time.
> 
> cat /var/log/apt/history.log (pertinent entry): Start-Date: 2024-04-19 09:16:56 Commandline: apt upgrade Upgrade: rpi-eeprom:arm64 (21.6-1, 22.0-1), apache2-data:arm64 (2.4.57-2, 2.4.59-1~deb12u1), apache2-bin:arm64 (2.4.57-2, 2.4.59-1~deb12u1), zoneminder:arm64 (1.37.59~20240415.28-bookworm, 1.37.59~20240417.31-bookworm), apache2-utils:arm64 (2.4.57-2, 2.4.59-1~deb12u1), zoneminder-doc:arm64 (1.37.59~20240415.28-bookworm, 1.37.59~20240417.31-bookworm), apache2:arm64 (2.4.57-2, 2.4.59-1~deb12u1) End-Date: 2024-04-19 09:17:41
> 
> rpi-eeprom:arm64 (21.6-1, 22.0-1) seems to be the only system package that was updated.
> 
> Just to clarify, I'm using a Pi5, not 4.
> 
> I'll happily try rolling back. Are the above commands still valid for my Pi5? `$ cd /lib/firmware/raspberrypi/bootloader-2711/latest` `$ sudo rpi-eeprom-config --edit pieeprom-2024-04-15.bin`

Coincidentally, I did spot something in the Pi5 code (not part of Pi4 so not really this issue!). If you are already happy with rpi-update then you could install the firmware from there. Or download it via wget.

```
wget https://github.com/raspberrypi/rpi-eeprom/raw/master/firmware-2712/default/pieeprom-2024-04-20.bin
sudo rpi-eeprom-config --edit pieeprom-2024-04-20.bin
sudo reboot
```
This should be in APT next week if you want to use the normal install route

donteatsoclose2me, 2024-04-20:
Excellent!

I did end up rolling back to 4-15, and that alleviated the problem I was seeing.  Now, I've just installed 23.0-1 (already available in the repos), and I'm back to normal GPU performance.

Thanks for looking into this and promptly providing a solution.

qrp73, 2024-04-22:
I reverted to pieeprom-2024-04-15.bin and keep running it for 2 days. The freeze issue didn't happened.

timg236, 2024-04-23:
> cat /var/log/apt/history.log (pertinent entry): Start-Date: 2024-04-19 09:16:56 Commandline: apt upgrade Upgrade: rpi-eeprom:arm64 (21.6-1, 22.0-1), apache2-data:arm64 (2.4.57-2, 2.4.59-1~deb12u1), [...]

Great thanks for confirming.

timg236, 2024-04-23:
> I reverted to pieeprom-2024-04-15.bin and keep running it for 2 days. The freeze issue didn't happened.

I've not seen a freeze but if you see the freeze again it might be useful to capture the output of 'sudo vclog -m' and dmesg after the freeze to see if there there is any unusual messages there.

qrp73, 2024-04-23:
Just caught freeze with pieeprom-2024-04-15.bin, so it seems that this issue happens due to other update.

dmesg:
```
...
[ 7488.596120] vc4-drm gpu: swiotlb buffer is full (sz: 466944 bytes), total 32768 (slots), used 140 (slots)
[ 8298.369472] vc4-drm gpu: swiotlb buffer is full (sz: 339968 bytes), total 32768 (slots), used 0 (slots)
[ 8298.952917] vc4-drm gpu: swiotlb buffer is full (sz: 339968 bytes), total 32768 (slots), used 0 (slots)
[ 8995.089351] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1256 (slots)
[ 8995.094219] vc4-drm gpu: swiotlb buffer is full (sz: 3883008 bytes), total 32768 (slots), used 0 (slots)
[ 8995.100938] vc4-drm gpu: swiotlb buffer is full (sz: 3932160 bytes), total 32768 (slots), used 0 (slots)
[ 8995.107912] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1256 (slots)
[ 8995.646293] vc4-drm gpu: swiotlb buffer is full (sz: 3883008 bytes), total 32768 (slots), used 0 (slots)
[ 8995.695292] vc4-drm gpu: swiotlb buffer is full (sz: 3932160 bytes), total 32768 (slots), used 0 (slots)
[ 9389.549806] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 950 (slots)
[ 9389.557478] vc4-drm gpu: swiotlb buffer is full (sz: 1159168 bytes), total 32768 (slots), used 0 (slots)
[ 9389.561700] vc4-drm gpu: swiotlb buffer is full (sz: 1421312 bytes), total 32768 (slots), used 0 (slots)
[ 9389.570557] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 950 (slots)
[ 9390.069309] vc4-drm gpu: swiotlb buffer is full (sz: 1159168 bytes), total 32768 (slots), used 0 (slots)
[ 9390.112088] vc4-drm gpu: swiotlb buffer is full (sz: 1421312 bytes), total 32768 (slots), used 0 (slots)
```

sudo vclog -m
```
...
009058.657: brfs: File read: 181 bytes
010713.401: brfs: File read: /mfs/sd/kernel8.img
010713.439: Loaded 'kernel8.img' to 0x200000 size 0x8d4b33
013557.604: Device tree loaded to 0x2df9c400 (size 0xdbbf)
013565.903: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
013674.070: sdram: sdram refresh 1562->6248 (1)

015826.965: vchiq_core: vchiq_init_state: slot_zero = 0xce080000, is_master = 1
015836.843: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000023
4138759.242: video_decode:8:RIL: resolution changing
4198149.332: video_decode:10:RIL: resolution changing
4198171.143: video_decode:9:RIL: resolution changing
4209734.845: video_decode:11:RIL: resolution changing
4209770.305: video_decode:12:RIL: resolution changing
014777.762: video_decode:13:RIL: resolution changing
083073.846: video_decode:14:RIL: resolution changing
327064.209: video_decode:15:RIL: resolution changing
327066.834: video_decode:16:RIL: resolution changing
398047.960: video_decode:17:RIL: resolution changing
398164.372: video_decode:18:RIL: resolution changing
409671.972: video_decode:19:RIL: resolution changing
448552.675: video_decode:20:RIL: resolution changing
459928.803: video_decode:21:RIL: resolution changing
478468.263: video_decode:22:RIL: resolution changing
484360.495: video_decode:23:RIL: resolution changing
489754.934: video_decode:24:RIL: resolution changing
511184.775: video_decode:25:RIL: resolution changing
810461.234: video_decode:26:RIL: resolution changing
870067.462: video_decode:28:RIL: resolution changing
870070.142: video_decode:27:RIL: resolution changing
954759.292: video_decode:29:RIL: resolution changing
995014.531: video_decode:30:RIL: resolution changing
995021.391: video_decode:31:RIL: resolution changing
1009085.581: video_decode:32:RIL: resolution changing
1062500.785: video_decode:33:RIL: resolution changing
1062503.097: video_decode:34:RIL: resolution changing
1081350.894: video_decode:35:RIL: resolution changing
1083778.984: video_decode:36:RIL: resolution changing
1096868.510: video_decode:37:RIL: resolution changing
1102990.970: video_decode:38:RIL: resolution changing
1105547.495: video_decode:39:RIL: resolution changing
1224169.351: video_decode:40:RIL: resolution changing
1254784.058: video_decode:41:RIL: resolution changing
2885924.714: video_decode:42:RIL: resolution changing
2888980.518: video_decode:43:RIL: resolution changing
[... 28 lines cut ...]
4190009.100: video_decode:71:RIL: resolution changing
4190010.218: video_decode:72:RIL: resolution changing
036257.714: video_decode:73:RIL: resolution changing
036277.130: video_decode:74:RIL: resolution changing
045177.055: video_decode:75:RIL: resolution changing
045372.921: video_decode:76:RIL: resolution changing
130639.718: video_decode:77:RIL: resolution changing
138819.080: video_decode:78:RIL: resolution changing
167450.104: video_decode:79:RIL: resolution changing
307664.157: video_decode:80:RIL: resolution changing
407580.100: video_decode:81:RIL: resolution changing
407702.944: video_decode:82:RIL: resolution changing
411625.898: video_decode:83:RIL: resolution changing
414100.500: video_decode:84:RIL: resolution changing
414101.497: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
426080.882: video_decode:85:RIL: resolution changing
722759.342: video_decode:86:RIL: resolution changing
742814.618: video_decode:87:RIL: resolution changing
796429.020: video_decode:88:RIL: resolution changing
796445.853: video_decode:89:RIL: resolution changing
809643.801: video_decode:90:RIL: resolution changing
834833.446: video_decode:91:RIL: resolution changing
884994.568: video_decode:92:RIL: resolution changing
885730.915: video_decode:93:RIL: resolution changing
```

qrp73, 2024-05-06:
it is reproduced with old rpi-eeprom, so I close it because it looks like it don't related to rpi-eeprom update

closed 2024-05-06 by qrp73