raspberrypi/rpi-eeprom #574: Network install doesn't start with wireless keyboard/trackpad

jmhalder, 2024-04-27:
### Describe the bug

I've seen others having the same issue. It works as expected with a HP wired keyboard, never shows the red splash screen when I have Logitech K400r plugged in.

### Steps to reproduce the behaviour

Plug in and boot with a Logitech K400r (probably other K400 variants) with no boot device.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

#vcgencmd bootloader_version output
2024/04/20 11:53:30
version d1744d218076d7eee7c14763bbf3d3cf6278191a (release)
timestamp 1713610410
update-time 1714182005
capabilities 0x0000007f

#rpi-eeprom-config output
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
NET_INSTALL_ENABLED=1

timg236, 2024-04-27:
Please could you supply boot loader uart logs and full details of the HP keyboard ie Amazon (or similar) link

HiassofT, 2024-04-27:
This sounds a bit like the issue I reported in the beta forum a while ago https://forums.raspberrypi.com/viewtopic.php?p=2186020#p2186020

Here's the report with logs (I haven't tested with newer bootloaders yet):

There seems to be some issue with HID detection / handling:

Using a Logitech K400 plus (also tested with the original K400 version) and it's unifying receiver the net install screen doesn't come up, only the "black" detail status screen.

On an RPi4 those Logitech keyboards work fine and I can initiate netinstall by pressing shift.

Using a plain wireed HP USB keyboard on RPi5 brings up the netinstall screen though and I could install LibreELEC just fine.

With the Logitech keyboard I see the bootloader registering a HID device in the UART log:

```
RPi: BOOTSYS release VERSION:1aeae380 DATE: 2024/01/24 TIME: 12:16:01
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1706098561 serial e933002c boardrev d04170 stc 1622116
AON_RESET: 00000003 PM_RSTS 00001000
RP1_BOOT chip ID: 0x20001927
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
PMIC reset-event 00000000 rtc 00000001 alarm 00000000 enabled 0
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
DDR 4267 1 0 64 152
RP1_BOOT chip ID: 0x20001927

RP1_BOOT chip ID: 0x20001927
RP1_BOOT: fw size 25968
PCI2 init
PCI2 reset
PCIe scan 00001de4:00000001
RP1_CHIP_INFO 20001927

RPi: BOOTLOADER release VERSION:1aeae380 DATE: 2024/01/24 TIME: 12:16:01
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1706098561 serial e933002c boardrev d04170 stc 4589732
AON_RESET: 00000003 PM_RSTS 00001000
M.2 PCIe HAT not detected.
usb_pd_init status 3
USB_PD CONFIG 0 41
XHCI-STOP
xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
USBSTS 11
xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
xHC0 ports 3 slots 64 intrs 4
XHCI-STOP
xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
USBSTS 1
xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
xHC1 ports 3 slots 64 intrs 4
USB-PD: src-cap PDO object1 0x0a0191f4
Current 5000 mA
Voltage 5000 mV
USB-PD: src-cap PDO object2 0x0002d12c
Current 3000 mA
Voltage 9000 mV
USB-PD: src-cap PDO object3 0x0003c0e1
Current 2250 mA
Voltage 12000 mV
USB-PD: src-cap PDO object4 0x0004b0b4
Current 1800 mA
Voltage 15000 mV
Boot mode: SD (01) order f4
USB2[2] 000206e1 connected
USB2 root HUB port 2 init
DEV [01:00] 2.00 000000:02 class 0 VID 046d PID c52b
HID [01:00] 2.00 000000:02 register HID
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
[... 145 lines cut ...]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
Device Status:     0x0000
  (Bus Powered)
```

jmhalder, 2024-04-27:
lsusb shows the exact same information for me, including the same PID/VID for the root device(Which would make that it's the same "unified receiver" for the K400r and K400+. The only exception is the iConfiguration:
`    iConfiguration          4 RQR12.11_B0032`
I don't have 1mm jst connectors, and don't have anything to plug into the uart port during boot. The "normal" GPIO pins 14/15 for uart on the Pi 4 don't seem to give any output on the Pi 5.

Under the BOOT_UART [setting description](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#configuration-properties), the documentation makes no mention of using the jst uart port. Maybe it should make that distinction.

Since opening this issue, I've set NET_INSTALL_KEYBOARD_WAIT=4000
This had no effect.

timg236, 2024-04-29:
That looks familiar I found one in my attic :)

peterharperuk, 2024-04-30:
Yes, I can reproduce this. Works on Pi4 but not Pi5

closed 2024-05-13 by timg236