raspberrypi/rpi-eeprom #575: Raspberry Pi 5 /chosen/bootloader/partition is set to 0

gportay, 2024-04-30:
### Describe the bug

The system boots.

And, unless I am mistaken, the bootloader of the pi 5 (in its latest version, as of today ~`a482d84ba68bf5e0c150434d56061e6afa967b96`) does not set the `boot_partition` number to `/chosen/bootloader/partition`. I have `0` instead of `2`.

Here are the logs:

```
[ 1163.764341] reboot: Restarting system

RPi: BOOTSYS release VERSION:a482d84b DATE: 2024/04/05 TIME: 11:41:19
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1712313679 serial 64b59de4 boardrev d04170 stc 1031956
AON_RESET: 00000003 PM_RSTS 00001020
RP1_BOOT chip ID: 0x20001927
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
PMIC reset-event 00000000 rtc 6630f047 alarm 00000000 enabled 0
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
DDR 4267 1 0 64 152
OTP boardrev d04170 bootrom a a
Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
VC-JTAG unlocked
RP1_BOOT chip ID: 0x20001927

RP1_BOOT chip ID: 0x20001927
RP1_BOOT: fw size 25968
PCI2 init
PCI2 reset
PCIe scan 00001de4:00000001
RP1_CHIP_INFO 20001927

RPi: BOOTLOADER release VERSION:a482d84b DATE: 2024/04/05 TIME: 11:41:19
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1712313679 serial 64b59de4 boardrev d04170 stc 4052355
AON_RESET: 00000003 PM_RSTS 00001020
usb_pd_init status 1
XHCI-STOP
xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
USBSTS 1
xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
xHC0 ports 3 slots 64 intrs 4
XHCI-STOP
xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
USBSTS 1
xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
xHC1 ports 3 slots 64 intrs 4
Boot mode: SD (01) order f4
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [417]
CID: 001234565344000000000000000c0127
CSD: 400e00325b59000074937f800a400000
SD: bus-width: 4 spec: 2 SCR: 0x02b58443 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00000001,    8192 type: 0x0c
MBR: 0x00002001,   65536 type: 0x0c
MBR: 0x00012001,   65536 type: 0x0c
MBR: 0x00022001,  573442 type: 0x0f
Trying partition: 0
type: 12 lba: 1 'mkfs.fat' '  V       ^ ' clusters 2036 (4)
rsc 1 fat-sectors 6 root dir cluster 1 sectors 32 entries 512
FAT12 clusters 2036
Read autoboot.txt bytes       64 hnd 0x2
Select partition rsts 0 C(boot_partition) 2 EEPROM config 0 result 2
Trying partition: 2
type: 16 lba: 8193 'mkfs.fat' '  V       ^ ' clusters 16343 (4)
rsc 4 fat-sectors 64 root dir cluster 1 sectors 32 entries 512
FAT16 clusters 16343
Read config.txt bytes      548 hnd 0x29
[sdcard] pieeprom.upd not found
usb_max_current_enable default 0 max-current 900
Read bcm2712-rpi-5-b.dtb bytes    75189 hnd 0x4
dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712
Selecting USB low current limit

buildroot login: root
# cat /sys/firmware/devicetree/base/chosen/bootloader/version 
a482d84ba68bf5e0c150434d56061e6afa967b96# 
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/partition 
00000000  00 00 00 00                                       |....|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/build-timestamp 
00000000  66 0f d5 4f                                       |f..O|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/update-timestamp 
00000000  66 30 e9 fc                                       |f0..|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/                 
boot-mode         name              tryboot           
build-timestamp   partition         update-timestamp  
capabilities      rsts              version           
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/boot-mode 
00000000  00 00 00 01                                       |....|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/rsts      
00000000  00 00 10 20                                       |... |
00000004
# cat /boot/autoboot.txt 
[all]
tryboot_a_b=1
boot_partition=2
[tryboot]
boot_partition=3
```

### Steps to reproduce the behaviour

Boot the Pi 5 and check out the file `/sys/firmware/devicetree/base/chosen/bootloader/partition`, it should be different from `0`.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

### System

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

It is buildroot `Linux buildroot 6.1.61-v8_16k #1 SMP PREEMPT Mon Apr 29 15:25:00 CEST 2024 aarch64 GNU/Linux`.

### Bootloader logs

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

### USB boot

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

### NVMe boot

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

### Network (TFTP boot)

I am not using the Raspberry Pi OS. I can provide the inputs if requested.

gportay, 2024-05-15:
@timg236, thanks it works!

Firmware was updated via `Raspberry Pi OS`.

```
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/build-timestamp 
00000000  66 42 2e 75                                       |fB.u|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/version         
00000000  38 33 62 35 65 35 65 36  31 61 37 30 61 39 33 63  |83b5e5e61a70a93c|
00000010  61 65 65 36 35 63 62 64  36 61 65 61 37 37 62 64  |aee65cbd6aea77bd|
00000020  64 34 38 35 65 35 62 30  00                       |d485e5b0.|
00000029
```

```
# cat /boot/autoboot.txt 
[all]
tryboot_a_b=1
boot_partition=2
[tryboot]
boot_partition=3
```

Default boot, i.e. with tryboot unset:

```
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/partition 
00000000  00 00 00 02                                       |....|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/tryboot   
00000000  00 00 00 00                                       |....|
00000004
```

And with tryboot set:

```
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/partition 
00000000  00 00 00 03                                       |....|
00000004
# hexdump -C /sys/firmware/devicetree/base/chosen/bootloader/tryboot   
00000000  00 00 00 01                                       |....|
00000004
```

How it was updated (by memory):
```
git clone https://github.com/raspberrypi/rpi-eeprom.git
rpi-eeprom-update -f firmware-2712/latest/pieeprom-2024-05-13.bin
```

timg236, 2024-05-15:
Great! Thanks for verifying this.

This should be in APT fairly soon. Closing as fixed.