raspberrypi/rpi-eeprom #580: rpi-eeprom-update Wont update to latest

Fyb3roptik, 2024-05-22:
### Describe the bug

Using Raspberry OS on a 5. Tried to update to the latest 5/17 or even 4/20 for the hat+ support, but all it says is latest is Feb 16. If I try to download it and manually install I simply get an error ILLEGAL NUMBER invalid date '@'

### Steps to reproduce the behaviour

Use Raspberry OS on a flash drive, boot into it and try to run an eeprom update

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf41

timg236, 2024-05-22:
You'll need to provide more information about the exact command that you typed and the output (see Issue template).

Alternatively, you could use the Raspberry Pi Imager to create an EEPROM update SD card (use a spare card)
"Choose OS -> Misc Utility Images -> Bootloader EEPROM configuration"

Fyb3roptik, 2024-05-22:
I have no way of getting the info off of it and into here unfortunately. But I ran `sudo rpi-eeprom-update` and `sudo rpi-eeprom-update -f filename.bin` for manual install. Neither are working for latest Raspberry OS. This is already on a USB drive since my main SD Card uses Ubuntu and it doesn't work there either. On Ubuntu it says latest is Dec!

Fyb3roptik, 2024-05-22:
Here is from the Ubuntu 24.04 OS at least

```
sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed Dec  6 18:29:25 UTC 2023 (1701887365)
    LATEST: Wed Dec  6 18:29:25 UTC 2023 (1701887365)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

```
sudo rpi-eeprom-update -f pieeprom-2024-04-20.bin
*** CREATED UPDATE pieeprom-2024-04-20.bin  ***

/usr/bin/rpi-eeprom-update: 194: [: Illegal number:
   CURRENT: Wed Dec  6 18:29:25 UTC 2023 (1701887365)
date: invalid date ‘@’
    UPDATE:  ()
    BOOTFS: /boot/firmware
```

lurch, 2024-05-22:
I assume that you're running Raspberry Pi OS Bookworm on your USB flash drive? Are all the packages up to date with `apt` ?

Fyb3roptik, 2024-05-22:
Yes I am and no they're not. It was taking forever to run an upgrade. Let me use this other sd card I have so this will be faster. I had the same assumption that I needed to update. Will report back

timg236, 2024-05-22:
Since the commands are known to work on a clean install I suspect an environment variable conflict. Please can you attach your /etc/default/rpi-eeprom-update file and the output of this command

`sudo bash -x rpi-eeprom-update`

Fyb3roptik, 2024-05-22:
yeah looks like it was at version 21.4-1 and is getting 23.1-1 now

Fyb3roptik, 2024-05-22:
That did the trick. Updating eeprom now shows 5/17 image! Perfect! Sorry for not trying this before. Seems Ubuntu repos may be outdated a bit

closed 2024-05-22 by Fyb3roptik