raspberrypi/rpi-eeprom #584: Booting NVME with PCIE Gen3

yoramrotbach, 2024-06-15:
### Describe the bug

Sabrent Rocket 2230 NVME SSD, works great with pcie-gen3 speed once the system is already booted from uSD card.
As this drive drive most likely doesn't support pcie-Gen2 speeds, the eeprom code doesn't see it and the loading process is stuck. Please add a switch to enable pcie-gen3 switch in eeprom. The dtparam in config.txt is too late for the boot process.

### Steps to reproduce the behaviour

Use any pcie hat for raspberry pi 5. I'm using Geekworm x1003.
Use NVME SSD drive that doesn't support gen2 speeds. I'm using Sabrent rocket 2230, supporting gen3 and up speeds.
The NVME has been formatted to boot. The source of the image was a working uSD card.
Try to boot - will not succeed for many minutes. Black screen. after several minutes some text appears letting me know that all possible boot devices failed.
Turn PCIE switch in eeprom config - no effect.
boot from uSD card with pcie-gen3 configuration - NVME drive works great.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
PCIE_PROBE=1

BOOTLOADER: up to date
   CURRENT: Wed Jun  5 03:41:49 PM UTC 2024 (1717602109)
    LATEST: Wed Jun  5 03:41:49 PM UTC 2024 (1717602109)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.

### System

Latest Raspberry pi os 64 bit
bootloader latest 2712
firmware version 6fe0b091
Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### NVMe boot

Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            48803179100692       Sabrent SB-2130-256                      1         256.06  GB / 256.06  GB    512   B +  0 B   R21B47.1

______________________________________________________________________________________________

NVME Identify Controller:
vid       : 0x1987
ssvid     : 0x1987
sn        : 48803179100692      
mn        : Sabrent SB-2130-256                     
fr        : R21B47.1
rab       : 4
ieee      : 6479a7
cmic      : 0
  [3:3] : 0	ANA not supported
  [2:2] : 0	PCI
  [1:1] : 0	Single Controller
  [0:0] : 0	Single Port

mdts      : 6
cntlid    : 0
ver       : 0x10400
rtd3r     : 0x186a0
rtd3e     : 0x4c4b40
oaes      : 0x200
  [31:31] : 0	Discovery Log Change Notice Not Supported
  [27:27] : 0	Zone Descriptor Changed Notices Not Supported
  [15:15] : 0	Normal NSS Shutdown Event Not Supported
[... 286 lines cut ...]
dlfeat  : 9
  [4:4] : 0	Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0x1	Deallocate Bit in the Write Zeroes Command is Supported
  [2:0] : 0x1	Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr	: 0
nvmsetid: 0
endgid  : 0
nguid   : 00000000000000016479a74820c00937
eui64   : 6479a77a20c00937
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x1 Better (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best

lurch, 2024-06-15:
> boot from uSD card with pcie-gen3 configuration - NVME drive works great.

It would also be interesting to see if you get any errors in `dmesg` if you boot from uSD card with pcie-gen2 configuration.

peterharperuk, 2024-06-15:
I think I have a sabrent rocket. Let me check if I can get it to work.

yoramrotbach, 2024-06-15:
@lurch - when gen=2, the nvme isn't seen at all by the system.
nothing in dmesg.
sudo nvme list - returns nothing.

@pelwell - working on getting the serial debug information as suggested

timg236, 2024-06-15:
I have Sabrent Rocket 2230 and it works fine at Gen 2 with the Raspberry Pi M.2 HAT+