raspberrypi/rpi-eeprom #585: RPi 5 bootloader doesn't load kernel with GPT partition array not at LBA 2

sairon, 2024-06-27:
### Describe the bug

Home Assistant OS started using [Genimage](https://github.com/pengutronix/genimage) for creating OS images in release 12.4. After that change we started to get reports that freshly imaged SD card fails to boot on Raspberry Pi 5 (https://github.com/home-assistant/operating-system/issues/3437). The change that was introduced by using Genimage was that the `first LBA` field in the GPT header was changed from 34 (which is typically what `sgdisk` sets) to 2048 (which is what's actually more appropriate with 1MiB partition alignment) - something that looked quite harmless.

This worked perfectly fine when the image was flashed to the card from Linux, however, it failed to boot when Raspberry Pi Imager was used on Windows. Turned out the Imager itself isn't to blame, but the Windows kernel (presumably on Windows 10+, Windows 7 don't show that behavior), which alters the partition table **even when a drive is just plugged into the computer**. The change it does is that it moves the backup LBA to the real end of the drive, and *if the first LBA isn't 34*, it also moves the partition array start to a different block (seems to be `first_LBA - 32`). In the second case, the actual array of partition entries still exists on both places - at LBA 2 and LBA 2016 (i.e. the old data from the image is not nulled).

The problem is that Raspberry Pi bootloader can't boot if the partition entries are relocated. It doesn't seem that it makes assumptions the partition table is at LBA 2 - because it is still there, so maybe something is miscalculated or read is attempted outside of expected boundaries? Anyway, although what Windows do is quite a nasty thing, the GPT table is still perfectly valid, so I'd expect it shouldn't cause trouble.

Here's the diff of the part of the GPT table before and after it's modified by Windows - note the change in the last 8 bytes:
       
```diff
-00000210: 7e39 180f 0000 0000 0100 0000 0000 0000  ~9..............
+00000210: cf3d ab66 0000 0000 0100 0000 0000 0000  .=.f............
          | crc32   | reserved|current LBA        |
-00000220: ffff 3f00 0000 0000 0008 0000 0000 0000  ..?.............
+00000220: ffff ba03 0000 0000 0008 0000 0000 0000  ................
          | backup LBA        | first LBA         |
-00000230: deff 3f00 0000 0000 a211 3e2d a149 fa44  ..?.......>-.I.D
+00000230: deff ba03 0000 0000 a211 3e2d a149 fa44  ..........>-.I.D
          | last LBA          | GUID  0-7B        |
-00000240: a850 55b6 af53 4f62 0200 0000 0000 0000  .PU..SOb........
+00000240: a850 55b6 af53 4f62 e007 0000 0000 0000  .PU..SOb........
          | GUID 8-15B        | partition array   |
```

I can confirm that the partition entries start at LBA 2016 (0x7e0 = byte offset 0xfc000) after that change:

```
-000fc000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+000fc000: 2873 2ac1 1ff8 d211 ba4b 00a0 c93e c93b  (s*......K...>.;
```

For completeness, this is what Windows do with the partition table if the first LBA is set to 34:

```diff
-00000210: 99f9 2492 0000 0000 0100 0000 0000 0000  ..$.............
+00000210: deb0 923f 0000 0000 0100 0000 0000 0000  ...?............
          | crc32   | reserved|current LBA        |
-00000220: ffff 3f00 0000 0000 2200 0000 0000 0000  ..?.....".......
+00000220: ffff ba03 0000 0000 2200 0000 0000 0000  ........".......
          | backup LBA        | first LBA         |
-00000230: deff 3f00 0000 0000 faec 4e7f 7d9a 0f4b  ..?.......N.}..K
+00000230: deff ba03 0000 0000 faec 4e7f 7d9a 0f4b  ..........N.}..K
          | GUID 8-15B        | partition array   |
 00000240: 8bbc 2d37 8b85 a2f0 0200 0000 0000 0000  ..-7............
```       

### Steps to reproduce the behaviour

Install Home Assistant OS 12.4 image (e.g. using Raspberry Pi Imager) on Windows.

\- alternatively -

Run system image with GUID partition table with array of partition entries starting at LBA != 2.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Bootloader versions up to latest 6fe0b091 2024/06/05.

### Bootloader logs

[image]

timg236, 2024-07-02:
No plans to support this right now but leaving it open as possibly part of 4K native sector / large boot drive support if/when that's  supported.

sairon, 2024-07-02:
@timg236 Okay, it's a pity but I planned on accommodation to it anyway. However, do you have some more insights _why_ that happens? The GPT records are perfectly valid and the drive can be used in Windows and Linux without issues, it's just the bootloader which can't read data from the boot partition then.

timg236, 2024-07-02:
The bootloader has never supported the backup LBA and IIRC expects it to be able to use 2, maybe that constraint can be relaxed on Pi5 (only) since it's only read by the SPI bootloader.

timg236, 2024-07-02:
No promises but if someone could post images then we could see if a quick fix is possible. The image just needs to include all the gpt table rather than all the data

sairon, 2024-07-03:
Not sure we're on the same page. The backup LBA is not the problem here - the problem is when the [PartitionEntryLBA](https://uefi.org/specs/UEFI/2.10/05_GUID_Partition_Table_Format.html#gpt-header) (of the primary GPT) points to LBA 2016. Usually this is LBA 2 but if `FirstUsableLBA` is not 34, Windows relocates it. Also, the actual partition entries are both at LBA 2 and LBA 2016 then, so it seems something else goes wrong there.

The issue can be reproduced by flashing [Home Assistant OS 12.4](https://github.com/home-assistant/operating-system/releases/download/12.4/haos_rpi5-64-12.4.img.xz) image from Windows environment (or plugging an SD card with that image to Windows 10+ PC). If you want, I can prepare a minimal image from that for reproducing the bootloader issue, i.e. without the whole rootfs.

timg236, 2024-07-03:
I think the bootloader only understands the normal/default case. That might be fixable.
I've asked someone to reproduce it and look at it has a background task since this sounds like something that possibly used to work until Windows started doing relocation. Probably a couple of weeks though due to other projects.

learmj, 2024-10-02:
Hi @sairon 
Thanks for reporting this. A fix is in the works. fyi it can be reproduced without involving Windows. Use genimage to create a disk image with GPT and set the location of the Partition Entry Array to something other than LBA 2.
For example:

```
image disk.img {
   hdimage {
      partition-table-type = "gpt"
      gpt-location = 2048
   }
```
This locates it at LBA 4 (512B sectors) rather than the default of 2, which prevents the partition entries from being read.

learmj, 2024-10-03:
The attached should resolve the problem you're having with the HA images. Please feel free to give it a try. It would be great to know it resolves the problem you reported.
```
sudo rpi-eeprom-update -d -f /path/to/new/pieeprom.bin
```
Should yield:
```
$ sudo vcgencmd bootloader_version
2024/10/03 11:45:08
version 5fe3f5dc0a3ea9983fb42927170827c0935727ce (release)
```
The usual caveats apply about updating your eeprom.
[rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/17243132/rpi-eeprom-recovery.zip)

sairon, 2024-10-09:
Hi @learmj, thanks for looking into this. I can confirm the provided bootloader fixes the problem - a card that failed to boot (right before I updated the bootloader) now boots without issues. Thank you!

timg236, 2024-10-17:
Thanks @sairon for the test closing this issue as fixed

closed 2024-10-17 by timg236

martinetd, 2025-12-26:
This is a bit old, sorry for reviving this thread, but we ran into a similar problem (on another platform that expects the bootloader to be at the 1k offset) due to this:
> Turned out the Imager itself isn't to blame, but the Windows kernel (presumably on Windows 10+, Windows 7 don't show that behavior), which alters the partition table even when a drive is just plugged into the computer

Did you or someone else try reporting this to microsoft? I don't understand how modifying partition table without any prompt can fly...
I tried very hard to report it on their "feedback hub" (here: https://aka.ms/AAyj5qe , can apparently only be opened from windows..........) and haven't gotten the slightest acknowledgement in 2 months, but given that behaviour is apparently over a year old I doubt they're going to change now... Anyway, if someone did report (or knows a better way to report), please let me know!
Thanks!!