raspberrypi/rpi-eeprom #586: Disk corruption using USB3.0

damige, 2024-06-27:
### Describe the bug

When using the Pi 5 i have a large amount of irrecoverable corruption of data on my usb drives.

I use my pi5 as a NAS. On this device i used two USB3.0 14TB seagate disks (ironwolf pro and exos x18).
Both disks experience corruption as shown by btrfs in dmesg.
Smart data of all disks shows no signs of errors.

What i tried to fix it:

- Switched from luks+btrfs to plain btrfs.
- Removed the USB hub.
- Switched from btrfs to zfs.
- removed the disks from the enclosure and used a separate enclosure
- Upgraded the firmware from v2024.04.20-2712 to v2024.06.05-2712
- bought a ICY BOX IB-3805-C31 , and placed the disks into this enclosure
- used a triple mirror (4TB ironwolf pro) for data

and still data corruption appeared almost instantly. Even on the triple mirror some data was unrecoverable within the hour. (As stated by dmesg during scrub)

corruption is silent using ext4(due to no check summing?), but when using btrfs or zfs it shows up quite quick after doing a scrub.
I have tried switching from the UAS driver to usb-storage but this did not solve the issue.

What fixed the problem:
Attached the enclosures to an older usb 3.0 x86 system. Now no corruption for the last days.
This leads me to believe there is an firmware issue with the Pi5.

Because of this i lost about 20TB of linux ISO's which obviously is quite the hassle. Hopefully this can be fixed. For now storage has been migrated of the pi platform.

### Steps to reproduce the behaviour

Attach two (large?) USB 3.0 drives
use BTRFS or ZFS
Copy ~500GB+ of data
Execute a scrub

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf416
PCIE_PROBE=1

### System

Ubuntu 24.04 LTS 

2024/06/05 16:41:49
version 6fe0b091c7cb1c5da26b7f41cc24c42840531a83 (release)
timestamp 1717602109
update-time 1719396331
capabilities 0x0000007f

2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)

Linux earth 6.8.0-1005-raspi #5-Ubuntu SMP PREEMPT_DYNAMIC Wed May 22 15:01:35 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

lurch, 2024-06-28:
If this isn't a power-issue, then this is probably a [Linux](https://github.com/raspberrypi/linux) driver issue, rather than a bootloader issue.

damige, 2024-06-28:
> If this isn't a power-issue, then this is probably a [Linux](https://github.com/raspberrypi/linux) driver issue, rather than a bootloader issue.

Pi5 was powered using the supplied power adapter. External drives both have their own power source, and don't rely on pi5 board power. Power does seem unlikely.

I have tried the UAS and usb-storage driver, corruption happens on both. So it seems to be more generic then those settings.

damige, 2024-06-30:
Alright, so the issue was not pi5 firmware related.
It seems that in the closet where the hardware was located the pi5 @5000mbps usb3 gave corruption.
old x86 @5000mbps gave no corruption
new x86 @10000mbps gave corruption again.

I tried various stock supplies cables (usb-c to usb-c , usb-a to usb-c) etc.
Using a very short (~20cm) usb3.2 cable new x86 @10000mbps no longer corrupts.
So it seems to be a cable and interference problem.

closed 2024-06-30 by damige

kubrickfr, 2024-08-06:
I know your problem seems to be solved, but were you using disk encryption with LUKS?

damige, 2024-08-06:
> I know your problem seems to be solved, but were you using disk encryption with LUKS?

Yes, but i also had corruption without luks, and with plain zfs

pelwell, 2024-08-06:
What is the power supply arrangement in your system? By using the crypto extensions you are increasing the amount of work the system can do (good), but also increasing the amount of power that is required (and consuming power in a different part of the chip). Almost all memory corruption (or apparent memory corruption) problems usually turn out to be due to an inadequate supply of power. That may be because of a poor power supply, a bad cable, or because peripherals are trying to draw more current than the Pi can spare.

pelwell, 2024-08-06:
That is an example of a good power configuration.

kubrickfr, 2024-08-06:
Official RPI5 27W power supply, powering the PI, an nvme drive, and a fan. The 5-drive USB enclosure I am using has its own power supply.

pelwell, 2024-08-06:
That doesn't sound excessive. Is that using our M.2 HAT+?

kubrickfr, 2024-08-06:
Whatever is in [that box](https://wiki.geekworm.com/images/3/3d/FOR-WEB-INSTRUCTION-MANUAL-NEO-5-M.2-NVMe-20240222.pdf)

actraiser30, 2024-10-22:
It's a pi5 issue, whatever is plugged in usb ports hangs the system,and corruption occurs.
Still cannot use any of the usb2,3 ports,system stucks.