raspberrypi/rpi-eeprom #588: [CM4-S] BCM-USB-MSD boot mode fails to boot, with no feedback

ChilldyneTrevor, 2024-07-25:
### Describe the bug

I am using a Revolution Pi Connect S, which is based on CM4-S. I have flashed a bootloader config to enable BCM-USB-MSD boot. With bootloader versions up to `pieeprom-2024-04-15.bin`  the PI boots normally from various USB drives. 

However, starting with `pieeprom-2024-04-17.bin`, the Pi cannot boot from a USB MSD at power-on. When this happens, there is no video output at all, the disk access light on the USB drive goes out, and eventually LED A3 turns on solid red on the front of the RevPi Connect S enclosure (unfortunately I'm unsure to which pins this might be wired.)

I have found that if the USB drive is _not_ connected at power-on and instead inserted _after_ video output shows that USB boot is being attempted, the system will boot. But if the USB drive is connected at power-on, the system hangs as described above.

Strangely, there is _one_ configuration of USB device that I have on hand that appears to work, which is a micro SD card in the microSD slot of a multi-card SD to USB adapter. Putting the same card in a full-size SD sleeve in the other slot of the adapter doesn't work. All flash/thumb/pen drives and SSDs do not work.

### Steps to reproduce the behaviour

**To reproduce the failure condition:**

1. Flash a boot configuration prioritizing BCM-USB-MSD onto CM4-S with  `pieeprom-2024-04-17.bin` or newer.
2. Power off Pi.
3. Connect a bootable USB drive, especially a "thumb"/"pen" drive or SSD.
4. Power on Pi.
5. Observe failure: No video output, disk access light blinks and then goes out, LED A3 turns on solid red.

**To reproduce the working edge case:**

1. Flash a boot configuration prioritizing BCM-USB-MSD onto CM4-S with  `pieeprom-2024-04-17.bin` or newer.
2. Power off Pi with no USB devices connected.
3. Wait for video output to indicate BCM USB MSD boot is being attempted.
4. Connect bootable USB drive.
5. Observe Pi booting normally.

### Device (s)

Raspberry Pi CM4-S

### Bootloader configuration.

[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf15

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-12
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbaa2a26c810dc53305754575f299bc2319d46f2, stage2
```

Problematic bootloader: `pieeprom-2024-04-17.bin` and newer
Last working bootloader: `pieeprom-2024-04-15.bin`

```
$ vcgencmd version
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

```
$ uname -a
Linux RevPi110019 6.1.46-rt13-v8 #1 SMP PREEMPT_RT Mon, 22 Apr 2024 08:37:46 +0200 aarch64 GNU/Linux
```

### Bootloader logs

The Revolution Pi Connect S does not appear to have a UART connection that is set up correctly to output boot information, unless I am using it incorrectly. Boot details _are_ output over the HDMI connection during a successful boot, but when this problem occurs there is no HDMI output.

### USB boot

Various USB flash/pen/thumb drives and one SSD all failed. A full-size SD card in two different SD to USB adapters failed.

The only configuration that works on the newer bootloader versions is a 32 GB SanDisk microSD card in the microSD slot of an SD-USB adapter. If I put the card in a passive microSD-SD sleeve, and then put that in the full-size SD slot of the SD-USB adapter, the problem occurs.

Link to working uSD to USB adapter: https://www.digikey.com/en/products/detail/startech-com/SDMSDRWU3AC/21398884
Link to working SD card: https://www.digikey.com/en/products/detail/raspberry-pi/SC0251L/15212334

pelwell, 2024-07-26:
@nbuchwitz Is this something you've encountered?

nbuchwitz, 2024-07-26:
I will have a look, but probably not for next week.

nbuchwitz, 2024-07-26:
RevPi Connect S has a LAN9514 hub/ethernet phy and downstream a LAN912 hub/ethernet phy. The front ports are connected to the LAN9514 on downstream 1&2. 

@pelwell, any changes regarding device enumeration between both firmware versions?

pelwell, 2024-07-26:
Looking at the release history (https://github.com/raspberrypi/rpi-eeprom/commits/master/) and the 2024-04-17 release note (https://github.com/raspberrypi/rpi-eeprom/commit/c0a207f4524418d4f7d7266934dc7a1ae92110bf#diff-4bb6018cbd55737981d3920b5a74c870f352cd3c56f4065cc24901e0c580a049R3), the only change that looks like it might be relevant is:
```
  * Switch to building the Pi4 firmware from the common Pi4/Pi5
    mainline release. This doesn't change the Pi4 features
    but should make it quicker to release bug fixes in common code.
```
We'll have to dig into what the practical differences might have been.

nbuchwitz, 2024-07-26:
Lets let me first build a test setup and see if I can verify the report. I also have a serial console attached and can check if there might be any clues in the log. Might save you some time in the dirt.

nbuchwitz, 2024-07-26:
Can confirm that the screen stays black with 72caf667. Serial console as following:

```
  0.05 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  0.09 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 026544fd boardrev a03150 stc 509138
  0.26 PM_RSTS: 0x00001020
  0.26 part 00000000 reset_info 00000000
  0.29 uSD voltage 3.3V
  0.55 Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
  0.58 DDR 3200 0 0 8 152
  1.88 OTP boardrev a03150 bootrom 48b0 48b0
  1.90 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  1.97 VC-JTAG unlocked

  1.60 RPi: BOOTLOADER release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  1.65 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 026544fd boardrev a03150 stc 1765132
  1.75 Unsupported boot order 4
  1.94 Unsupported boot order 6
  1.97 Unsupported boot order 2
  1.05 Boot mode: BCM-USB-MSD (05) order f1
  1.10 XHCI-STOP
  1.12 xHC0 ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
  1.15 xHC0 ports 1 slots 64 intrs 1
  1.23 USB2[1] 000206e1 connected
  1.74 USB2[1] 00200e03 connected enabled
  1.75 USB2 root HUB port 1 init
  1.78 DEV [01:00] 2.00 000000:01 class 9 VID 0424 PID 9514
  1.83 HUB init [01:00] 2.00 000000:01
  2.87 HUB [01:00] 2.00 000000:01 init port 1 speed 3
  2.89 DEV [02:01] 2.00 000001:01 class 255 VID 0424 PID ec00
  2.95 HUB [01:00] 2.00 000000:01 init port 2 speed 3
  2.98 DEV [03:01] 2.00 000002:01 class 0 VID 14af PID af01
  2.02 MSD device [03:01] 2.00 000002:01 conf 0 iface 0 ep 81#512 02#512
  2.08 MSD [03:01] 2.00 000002:01 register MSD
  2.12 MSD [03:01] 2.00 000002:01 LUN 0
  2.09 HUB [01:00] 2.00 000000:01 init port 5 speed 3
  2.11 DEV [04:01] 2.00 000005:01 class 9 VID 0424 PID 9512
  2.16 HUB init [04:01] 2.00 000005:01
  2.20 HUB [04:01] 2.00 000005:01 init port 1 speed 3
  2.23 DEV [05:04] 2.00 000015:01 class 255 VID 0424 PID ec00
  2.28 HUB [04:01] 2.00 000005:01 init port 2 speed 1
  2.31 DEV [06:04] 2.00 000025:01 class 0 VID 0403 PID 6001
  2.36 MSD INQUIRY [03:01] 2.00 000002:01
  2.39 MSD [03:01] 2.00 000002:01 lun 0 block-count 15720448 block-size 512
```

I also noticed that they relay (on GPIO1) in the device switches after some additional seconds.

The next older release e608a69d boots without any issue:
```
RPi: BOOTLOADER release VERSION:e608a69d DATE: 2024/04/15 TIME: 14:12:14
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1713186734 serial 026544fd boardrev a03150 stc 472117
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
DDR 3200 0 0 8 152

Unsupported boot order 4
Unsupported boot order 6
Unsupported boot order 2
Boot mode: BCM-USB-MSD (05) order f1
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
xHC ports 1 slots 64 intrs 1
USB2[1] 000206e1 connected
USB2[1] 00200e03 connected enabled
USB2 root HUB port 1 init
DEV [01:00] 2.00 000000:01 class 9 VID 0424 PID 9514
HUB init [01:00] 2.00 000000:01
HUB [01:00] 2.00 000000:01 init port 1 speed 3
DEV [02:01] 2.00 000001:01 class 255 VID 0424 PID ec00
HUB [01:00] 2.00 000000:01 init port 2 speed 3
DEV [03:01] 2.00 000002:01 class 0 VID 14af PID af01
MSD device [03:01] 2.00 000002:01 conf 0 iface 0 ep 81#512 02#512
MSD [03:01] 2.00 000002:01 register MSD
MSD [03:01] 2.00 000002:01 LUN 0
HUB [01:00] 2.00 000000:01 init port 5 speed 3
DEV [04:01] 2.00 000005:01 class 9 VID 0424 PID 9512
HUB init [04:01] 2.00 000005:01
HUB [04:01] 2.00 000005:01 init port 1 speed 3
DEV [05:04] 2.00 000015:01 class 255 VID 0424 PID ec00
MSD INQUIRY [03:01] 2.00 000002:01
MSD [03:01] 2.00 000002:01 lun 0 block-count 15720448 block-size 512
MBR: 0x00000800, 2097152 type: 0x0c
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 2048 c-count 261628 c-size 8
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 261628
Trying partition: 0
type: 32 lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 2048 c-count 261628 c-size 8
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 261628
Read config.txt bytes     1926 hnd 0x109
HUB [04:01] 2.00 000005:01 init port 2 speed 1
DEV [06:04] 2.00 000025:01 class 0 VID 0403 PID 6001
HUB [04:01] 2.00 000005:01 init port 3 speed 1
DEV [07:04] 2.00 000035:01 class 0 VID 0403 PID 6001
Read start4.elf bytes  2251392 hnd 0x157c
Read fixup4.dat bytes     5399 hnd 0x11b
0x00a03150 0x00000000 0x00001fff
MEM GPU: 76 ARM: 947 TOTAL: 1023
Firmware: 82f3750a65fadae9a38077e3c2e217ad158c8d54 Mar 17 2023 10:50:39
Starting start4.elf @ 0xfeb00200 partition 0
+
```

pelwell, 2024-07-26:
This could be due to a timing change in the bootloader, which might be hard to pinpoint. The advice for now is to stay on `pieeprom-2024-04-15.bin` - there should be no reason to upgrade at the moment.

timg236, 2024-07-26:
The difference between these versions is essentially pre/post Pi5 fork/merge. Whilst the XHCI, MSD/SCSI and file-system code is almost identical there will be differences in timing which can upset some drives.

I've just booted a CM4-S with an SD card in a one of these U-Green SD card readers to verify that the BCM2711 XHCI driver is still functional
https://www.amazon.co.uk/UGREEN-Adapter-Windows-Surface-Chromebook/dp/B01EFPX9XA/

Revision        : a03150
Serial          : 10000000b33619c5
Model           : Raspberry Pi Compute Module 4S Rev 1.0
pi@tim-cm4s:~$ vcgencmd bootloader_version
2024/05/17 12:26:58
version 72caf66729df313801bcefe9b1ff7099c71bb5ce (release)
timestamp 1715945218
update-time 1722008570
capabilities 0x0000007f

The failure point after the INQUIRY command might suggest that the USB drive isn't ready and then never becomes ready because it was asked too early.
This EEPROM parameter might help
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_STARTUP_DELAY

As @pelwell says upgrading beyond the last stable release doesn't add much more functionality except for some unification of secure-boot code so it might be best to stick with that for the moment.

If it is as I suspect timing related differences at power on this invariably breaks some drives / fixes others so a quick fix is unlikely unless someone spots another issue e.g. a protocol bug that's slipped through.

ChilldyneTrevor, 2024-07-26:
@pelwell, @timg236 : Thank you, and understood. It's no trouble for us to use the older release for our project, particularly since we need to flash the bootloader with the new boot order before deployment anyway.

@timg236: I did try various combinations of `USB_MSD_STARTUP_DELAY`, `USB_MSD_PWR_OFF_TIME`, and `USB_MSD_DISCOVER_TIMEOUT`. I was not able to find settings that worked, but my testing was not particularly methodical.

Regarding breaking some/fixing others, all of the following boot on the old firmware but fail on the new firmware:

- An Inland 2.5" SSD on a Sabrent SATA to USB adapter.
- A PNY 32 GB USB 3.0 flash drive
- A Memorex 64GB USB 3.1 flash drive
- SanDisk 32GB Ultra Fit flash drive: https://www.amazon.com/dp/B09FFK1QLR
- A SanDisk 32GB Micro-SD in a passive full-size SD sleeve in a generic SD to USB adapter
- The same SanDisk 32GB Micro-SD in a passive full-size SD sleeve in the full-size SD slot of the StarTech USB adapter in my original post

The only combination that worked was that same SanDisk 32GB Micro-SD in the Micro-SD slot of the StarTech adapter.

timg236, 2024-07-26:
Thanks for the information. To rule out one more thing please could you see if 2024/05/17 is any better?

ChilldyneTrevor, 2024-07-26:
@timg236: confirmed, the problem still occurs with pieeprom-2024-05-17.bin

nbuchwitz, 2024-07-26:
Did also some tests with `USB_MSD_STARTUP_DELAY`, `USB_MSD_PWR_OFF_TIME`, and `USB_MSD_DISCOVER_TIMEOUT`. Unfortunately without any success.

 After that I ran a comparison of good and bad firmware with `XHCI_DEBUG=0x10`. During initialization I've spotted:

```
  2.53 MSD [03:01] 2.00 000002:01 LUN 0
  2.56 HUB [01:00] 2.00 000000:01 state: 9
```

On the working firmware, the order is different, and the HUB reaches state 9 before the MSD. I guess the states references to the internal state machine? What does happend before / in stage 9?

[bad-boot.txt](https://github.com/user-attachments/files/16395815/bad-boot.txt)
[good-boot.txt](https://github.com/user-attachments/files/16395816/good-boot.txt)

timg236, 2024-07-29:
The firmware on this issue may help but can't rule out other issues
https://github.com/raspberrypi/rpi-eeprom/issues/527#issuecomment-2255165612

nbuchwitz, 2024-07-29:
Thanks, seems to do the trick:

```
  0.07 RPi: BOOTSYS release VERSION:4000fad6 DATE: 2024/07/29 TIME: 08:10:00                                                                                
  0.11 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1722237000 serial 026544fd boardrev a03150 stc 511677                                            
  0.29 PM_RSTS: 0x00001000                                                                                                                                  
  0.29 part 00000000 reset_info 00000000                                                                                                                    
  0.32 uSD voltage 3.3V                                                                                                                                     
  0.58 Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200                                                                                          
  0.61 DDR 3200 0 0 8 152                                                                                                                                   
  1.91 OTP boardrev a03150 bootrom 48b0 48b0                                                                                                                
  1.93 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000                                                                   
  1.00 VC-JTAG unlocked                                                                                                                                     
                                                                                                                                                            
                                                                                                                                                            
  1.64 RPi: BOOTLOADER release VERSION:4000fad6 DATE: 2024/07/29 TIME: 08:10:00                                                                             
  1.68 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1722237000 serial 026544fd boardrev a03150 stc 1768398                                           
  1.78 Unsupported boot order 4                                               
  1.98 Unsupported boot order 6                                                                                                                             
  1.00 Unsupported boot order 2                                                                                                                             
  1.09 Boot mode: BCM-USB-MSD (05) order f1                                   
  1.14 XHCI-STOP                                                                                                                                            
  1.15 xHC0 ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65                                                                                          
  1.18 xHC0 ports 1 slots 64 intrs 1                                                                                                                        
  1.27 USB2[1] 000206e1 connected                                             
  1.77 USB2[1] 00200e03 connected enabled                                                                                                                   
  1.78 USB2 root HUB port 1 init                                                                                                                            
  1.81 DEV [01:00] 2.00 000000:01 class 9 VID 0424 PID 9514                                                                                                 
  1.86 HUB init [01:00] 2.00 000000:01                                        
  2.90 HUB [01:00] 2.00 000000:01 init port 1 speed 3                         
  2.93 DEV [02:01] 2.00 000001:01 class 255 VID 0424 PID ec00                 
  2.98 HUB [01:00] 2.00 000000:01 init port 2 speed 3                         
  2.01 DEV [03:01] 2.00 000002:01 class 0 VID 14af PID af01                   
  2.05 MSD device [03:01] 2.00 000002:01 conf 0 iface 0 ep 81#512 02#512                                                                                    
  2.12 MSD [03:01] 2.00 000002:01 register MSD                                                                                                              
  2.16 MSD [03:01] 2.00 000002:01 LUN 0                                                                                                                     
  2.12 HUB [01:00] 2.00 000000:01 init port 5 speed 3                                                                                                       
  2.14 DEV [04:01] 2.00 000005:01 class 9 VID 0424 PID 9512                                                                                                 
  2.19 HUB init [04:01] 2.00 000005:01                                                                                                                      
  2.23 HUB [04:01] 2.00 000005:01 init port 1 speed 3                                                                                                       
  2.26 DEV [05:04] 2.00 000015:01 class 255 VID 0424 PID ec00                                                                                               
[... cut ...]
  2.44 HUB [04:01] 2.00 000005:01 init port 3 speed 1                                                                                                       
  2.48 DEV [07:04] 2.00 000035:01 class 0 VID 0403 PID 6001                                                                                                 
  2.93 Read start4.elf bytes  2251392 hnd 0x157c                                                                                                            
  3.00 Read fixup4.dat bytes     5399 hnd 0x11b 
```

timg236, 2025-01-07:
Closing as resolved.

closed 2025-01-07 by timg236