raspberrypi/rpi-eeprom #596: `rpi-eeprom-update` confusingly reports firmware versions for external VL805 controllers

srcshelton, 2024-08-18:
### Describe the bug

The Turing Pi Cluster board supports up to four CM4 modules, the fourth of which has a VL805 controller on its PCIe bus.

The first three modules report:

```
$ sudo rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Tue 30 Jul 13:02:35 UTC 2024 (1722344555)
    LATEST: Tue 30 Jul 13:02:35 UTC 2024 (1722344555)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 
    LATEST: 
```

… whilst the fourth reports:

```
$ sudo rpi-eeprom-update -a
Password:
BOOTLOADER: up to date
   CURRENT: Tue 30 Jul 13:02:35 UTC 2024 (1722344555)
    LATEST: Tue 30 Jul 13:02:35 UTC 2024 (1722344555)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 00013600
    LATEST: 00013600
```

… which is both incorrect (this controller isn't using the boot loader EEPROM) and confusing (the latest RPi VL805 firmware is `000138c0`).

If the VL805 firmware upgrade doesn't apply to a given model of RPi, perhaps that section should be elided from the output regardless of whether there is a VL805 present?

(Would I be correct in saying that there is no intersection between the devices which have VL805 firmware updatable by `rpi-eeprom-update` and those which have an accessible PCIe lane to which an external controller could be attached?)

### Steps to reproduce the behaviour

Run `rpi-eeprom-update` on a system with a PCIe-connected VL805 controller.

### Device (s)

Raspberry Pi CM4

### Bootloader configuration.

```
$ sudo rpi-eeprom-config
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```

### System

```
$ sudo vcgencmd bootloader_version
2024/07/30 14:02:35
version 5f18ffb61682740ed1751ba59952843d7521314f (release)
timestamp 1722344555
update-time 1719242772
capabilities 0x0000007f

$ sudo vcgencmd version
Apr 17 2024 17:27:22 
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start_cd)
$ uname -a
Linux rpi-cm4-8-32 6.6.20_p20240307 #3 SMP Fri Apr 12 07:23:47 BST 2024 aarch64 GNU/Linux
```

### Bootloader logs

n/a

### USB boot

n/a

### NVMe boot

n/a

### Network (TFTP boot)

n/a

srcshelton, 2024-08-18:
```
$ lspci -d 1106:3483 -xxx
01:00.0 USB controller: VIA Technologies, Inc. VL805/806 xHCI USB 3.0 Controller (rev 01)
00: 06 11 83 34 00 00 10 00 01 30 03 0c 00 00 00 00
10: 04 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 06 11 83 34
30: 00 00 00 00 80 00 00 00 00 00 00 00 00 01 00 00
```

srcshelton, 2024-08-18:
Related to #469 (in that not outputting VL805 details unless relevant would resolve this issue).

timg236, 2024-08-19:
It's not ideal, but realistically, not going to change this.

closed 2024-08-19 by timg236