raspberrypi/rpi-eeprom #597: CM4 stuck in infinite loop if EEPROM write protection is set and firmware upgrade is present

uplogix-mmcclain, 2024-08-19:
### Describe the bug

System is stuck in infinite loop when bootloader attempts to upgrade the firmware, but the EEPROM is write protected.

I would expect the bootloader to not attempt a firmware update when the EEPROM is write protected.
```
276.96 pieeprom.sig
276.96 hash: 7d24337713cf5f617f67fc65accbaf27a69b299669d38a168acdb6bc3a00a098
276.02 ts: 1724092212
276.92 SELF-UPDATE timestamp current 1724092073 new 1724092212 
276.94 Updating bootloader EEPROM
276.17 Reading EEPROM: 524288 bytes 0x00b60000
276.66 347ms
276.00 EEPROM is write protected (bc)
276.01 Reading EEPROM: 524288 bytes 0x00b60000
277.33 628ms
277.68 EEPROM is write protected (bc)
283.91 pieeprom.sig
283.91 hash: 7d24337713cf5f617f67fc65accbaf27a69b299669d38a168acdb6bc3a00a098
283.97 ts: 1724092212
283.87 SELF-UPDATE timestamp current 1724092073 new 1724092212 
283.90 Updating bootloader EEPROM
283.13 Reading EEPROM: 524288 bytes 0x00b60000
283.62 347ms
283.96 EEPROM is write protected (bc)
283.96 Reading EEPROM: 524288 bytes 0x00b60000
284.29 628ms
284.63 EEPROM is write protected (bc)
```

### Steps to reproduce the behaviour

This is with a CM4 and CM4io board.

Add the following to config.txt.

```
eeprom_write_protect=1

[cm4]
dtparam=spi=on
dtoverlay=audremap
dtoverlay=spi-gpio40-45
```

Put a jumper across the EEPROM_nWP pin to ground.

Put new pieeprom.upg and pieeprom.sig files in firware directory.

sync and umount the filesystem.

Unplug the power and reconnect.

Below is output from latest firmware 2024/07/30.
This is also broken in the default firmware dated 2023/01/11 and 2024/04/15.
I didn't test any other versions.

### Device (s)

Raspberry Pi CM4

### Bootloader configuration.

eeprom_write_protect=1

[cm4]
dtparam=spi=on
dtoverlay=audremap
dtoverlay=spi-gpio40-45

### Bootloader logs

[infinite_loop_firmware_20240730.txt](https://github.com/user-attachments/files/16664222/infinite_loop_firmware_20240730.txt)
[infinite_loop_firmware_20230111.txt](https://github.com/user-attachments/files/16664225/infinite_loop_firmware_20230111.txt)

timg236, 2024-08-20:
Sounds like a bug but there's no way for the bootloader to sense the nWP pin directly. Therefore, it will have to try the update at least once before giving up and booting anyway

timg236, 2024-10-17:
Closing as fixed

closed 2024-10-17 by timg236