raspberrypi/rpi-eeprom #606: rpi-eeprom-update unable to find default directory

L4wnmower, 2024-09-13:
### Describe the bug

After latest updates and EEPROM bump to rpi-eeprom 25.1-1, rpi-eeprom-update is unable to find the default directory

### Steps to reproduce the behaviour

Entering
`sudo rpi-eeprom-update`
output
`EEPROM updates directory /usr/bin/firmware/default not found.`

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
```

### System

```
**cat /etc/rpi-issue**

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

**vcgencmd bootloader_version**

2024/09/10 14:40:30
version 5be4f304b152e06cbcb9c08c26aafc9da755ae47 (release)
timestamp 1725975630
update-time 1726224794
capabilities 0x0000007f

**vcgencmd version**

2024/09/10 14:40:30 
Copyright (c) 2012 Broadcom
version 5be4f304 (release) (embedded)

**uname -a**

Linux RasPi5 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux
```

### Bootloader logs

NA

### USB boot

NA

### NVMe boot

NA

### Network (TFTP boot)

NA

lurch, 2024-09-13:
Ahhh, the `EXT4-fs error` indicates that your SD-card has started experiencing filesystem corruption (which could also explain why `/usr/bin/firmware/default` has "disappeared"). Your safest course of action is probably to stop using that SD-card as soon as possible, make a backup (onto a different drive or device!) of any files you need from that card, and then write a fresh image onto that SD card using Raspberry Pi Imager. If you're very unlucky, the Imager "verify" stage might fail, indicating that your SD card has developed hardware errors.

L4wnmower, 2024-09-13:
I will try with a fresh SSD/USB 3.0 installation to check if this is reproducible or not. It is a bit odd though: I don't have a bunch of scattered filesystem errors - just this one on this specific inode. The SD card is a fresh one, a 256GB Kingston Canvas Go Plus, with no more than 30 - 40 hours of intermittent operation in total.

timg236, 2024-09-13:
What does your environment say e.g. do have a variable called FIRMWARE_ROOT ?

L4wnmower, 2024-09-13:
No. Should there be a variable with that name?
To retrace back what happened
I used apt to update this afternoon. From `/var/log/apt/history.log` entries are
```
Start-Date: 2024-09-13  13:42:04
Commandline: apt full-upgrade
Requested-By: george (1000)
Install: libupower-glib3:arm64 (0.99.20-2, automatic), chromium-sandbox:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, automatic), libc++1-16:arm64 (1:16.0.6-15~deb12u1, automatic), notification-daemon:arm64 (3.20.0-4+b1, automatic), libunwind-16:arm64 (1:16.0.6-15~deb12u1, automatic), chromium:arm64 (1:128.0.6613.137-1~deb12u1+rpt2), chromium-l10n:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, automatic), chromium-common:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, automatic), libc++abi1-16:arm64 (1:16.0.6-15~deb12u1, automatic), upower:arm64 (0.99.20-2, automatic)
Upgrade: raspi-utils-eeprom:arm64 (20240402-4, 20240903-1), rpi-eeprom:arm64 (24.0-1, 25.1-1), raspi-utils-otp:arm64 (20240402-4, 20240903-1), python3-gpiozero:arm64 (2.0-1, 2.0.1-0+rpt1), raspi-utils-core:arm64 (20240402-4, 20240903-1), rc-gui:arm64 (1.83, 1.85), chromium-browser:arm64 (126.0.6478.164-rpt1, 1:128.0.6613.137-1~deb12u1+rpt2), rpi-chromium-mods:arm64 (20230925, 20240904), raspinfo:arm64 (20240402-4, 20240903-1), raspi-utils:arm64 (20240402-4, 20240903-1), raspi-utils-dt:arm64 (20240402-4, 20240903-1), code:arm64 (1.92.2-1723659531, 1.93.0-1725457763), raspberrypi-sys-mods:arm64 (20240902, 20240911), chromium-browser-l10n:arm64 (126.0.6478.164-rpt1, 1:128.0.6613.137-1~deb12u1+rpt2)
End-Date: 2024-09-13  13:47:44
```
There were no apt or dpkg error messages of any kind. Just to make sure, I rechecked the log files tonight.
After updating, I rebooted and since I saw that an EEPROM update package was installed, I ran `sudo rpi-eeprom-update -d -a` to install the new one. I rebooted again and got the 
`EEPROM updates directory /usr/bin/firmware/default not found.`
output from `sudo rpi-eeprom-update`.
I attempted to reproduce the error on a pristine  USB-SSD installation. When I tried `raspi-config` to change the boot order to USB, I got an error message about `rpi-eeprom` not being found. I reinstalled it using apt. Executed the test on the ssd w/o errrors. Reinstated to boot from SD without any errors, neither from the filesystem or from `rpi-eeprom-update`. 
Obviously something went sideways during the original sequence. 
Is there something else in the logs, that I forgot to check? Most important of all, though: Pi appears to work normally so far, but can I somehow verify if the EEPROM was flashed properly?

timg236, 2024-09-13:
No just checking in case it had overridden a default value. Maybe you could add ‘set -x’ to the top of the script and capture the output.
The usr/bin bit suggests something is getting mangled.
Please can you also attach /etc/default/rpi-erprom-update

L4wnmower, 2024-09-13:
Nothing unusual in  `/etc/default/rpi-eeprom-update`: 
`FIRMWARE_RELEASE_STATUS="default"`
Is there any way to check a posteriori the flashed EEPROM's hash? I thought about reflashing, but  I'm not sure about any side effects.

timg236, 2024-09-16:
You can use `vcgencmd bootloader_version` and `vcgencmd bootloader_config` to get information about the current bootloader (current as in the one that booted not the current flash contents).

Looking at the paths I think you are missing files under `/lib/firmware/raspberrypi/` causing the script to think that it's running from a direct git clone rather than the system version.

Try running this command to see if the package is corrupted
`sudo dpkg --verify rpi-eeprom`

If so run
```
sudo apt purge rpi-eeprom
sudo apt install rpi-eeprom
```

L4wnmower, 2024-09-16:
I got

```
>vcgencmd bootloader_version
2024/09/10 14:40:30
version 5be4f304b152e06cbcb9c08c26aafc9da755ae47 (release)
timestamp 1725975630
update-time 1726252498
capabilities 0x0000007f
```

```
>vcgencmd bootloader_config
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
```

`sudo dpkg --verify rpi-eeprom` did not find anything wrong, but that's expected since I had already reinstalled `rpi-eeprom` using apt. 
Unfortunately, I did not check `/lib/firmware/raspberrypi/` prior to reinstalling `rpi-eeprom`. Now it appears populated by the various firmware files.

L4wnmower, 2024-09-17:
Tonight `rpi-eeprom` was updated from 25.1-1 to 25.2-1. 
Since this was probably a glitch and not really a bug, I'm closing this issue.

closed 2024-09-17 by L4wnmower