raspberrypi/rpi-eeprom #607: rpi-otp-private-key fails for Pi 3B+ and Pi Zero 1.3 despite them having support for OTP

sunetfreitag, 2024-09-15:
### Describe the bug

The otp helper rpi-otp-private-key is written that it should also have support for most legacy hardware as stated in the documentation "Using the One-Time Programmable Memory on Raspberry Pi Single-Board Computers" (2023-02-10: githash: c65fe9c-clean).
Running the script on at least the Zero 1.c and 3B+ results in "Chip not supported". If you remove the BOARD_INFO check and set MAX_ROW_COUNT=8 it works.

BOARD_INFO from my boards:
- Pi Zero 1.3: 00900093
- Pi 3B+ 1.4: 00a020d4

### Steps to reproduce the behaviour

1. Copy [rpi-otp-private-key](https://github.com/raspberrypi/rpi-eeprom/blob/4c5aebdb200bc9a2ffd2a0158efffb9603c33be7/tools/rpi-otp-private-key) to your local file system
2. Execute rpi-otp-private-key

### Device (s)

Other

### Bootloader configuration.

(empty)

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2
$ vcgencmd bootloader_version
unknown

$ vcgencmd version
Feb 29 2024 12:26:47 
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)
$ uname -a
Linux udmd 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

timg236, 2024-09-15:
These are different chips with different otp reservations that also don’t support secure boot so this  is not possible

ysi-camerona, 2024-10-28:
@timg236 I ran into this unexpected behavior today too. The [docs](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#device-specific-private-key) suggest that reading/programming a device-specific private key is possible on all Raspberry Pi computers. From your perspective, should the docs be updated to indicate it is only supported on >= Pi 4? I'm happy to submit a docs PR but want to make sure we're all on the same page.

timg236, 2024-10-29:
Right now it requires >= Pi4 or newer so we can add a note.
It's possible that those rows could be made available on older models to store a device-private key. There's no secure-boot on those models so the security benefit is less obvious but there may be applications where that is ok e.g. if physical access to the device is not possible.

lurch, 2024-10-29:
ping @nathan-contino

ysi-camerona, 2024-10-30:
Thanks Tim, I'll aim to have a documentation PR submitted with that change on Friday.

ysi-camerona, 2024-11-01:
PR has been submitted: https://github.com/raspberrypi/documentation/pull/3900