raspberrypi/rpi-eeprom #611: Metadata gathering clashes with recovery reboot

tdewey-rpi, 2024-09-24:
### Describe the bug

As noted on https://github.com/raspberrypi/rpi-sb-provisioner/issues/45, using the 'latest' published recovery.bin (25.2-1), if I set both flags and then rpiboot the recovery image, I don't observe a reboot.

### Steps to reproduce the behaviour

Using rpi-eeprom (25.2-1), Compute Module 4 on CM4 IO Board:

```sh
cp "/lib/firmware/raspberrypi/bootloader-2711/latest/recovery.bin" "${FLASHING_DIR}/bootcode4.bin"
echo "recovery_reboot=1" >> "${FLASHING_DIR}/config.txt"
echo "recovery_metadata=1" >> "${FLASHING_DIR}/config.txt"
echo "program_pubkey=1" > "${FLASHING_DIR}/config.txt"

rpiboot -d "${FLASHING_DIR}"
```

Observe the device does not reboot.

### Device (s)

Raspberry Pi CM4

### Bootloader configuration.

N/A, uses only rpiboot with this firmware

### Bootloader logs

Attached to https://github.com/raspberrypi/rpi-sb-provisioner/issues/45 (unfortunately as a screenshot, sorry!)

timg236, 2024-09-24:
Thanks will check, I though I'd fixed this on Pi5 but perhaps something on 2711

timg236, 2024-09-25:
It seems to work for me, possibly I failed to bump the debian version in usbboot or missed the rpi-eeprom version change

pi@tim-pi4:~/usbboot/secure-boot-recovery $ strings bootcode4.bin | grep DATE
BOOT-EEPROM: UPDATED
ENABLE_SELF_UPDATE
DATE: 2024/09/05

tdewey-rpi, 2024-09-25:
Tested this a few more times - this is a false report, apologies.

It appears this actually impacts a race between udev and rpi-sb-provisioner, and removing metadata gathering tilted the odds but didn't actually resolve the problem.

closed 2024-09-25 by tdewey-rpi