raspberrypi/rpi-eeprom #619: [RaspberryPi 5] Unable to load signed boot.img w/initramfs > 30MB

SyedFaisalAli, 2024-10-16:
### Describe the bug

This seems similar to https://github.com/raspberrypi/rpi-eeprom/issues/375

The understanding from docs is "The maximum size for a ramdisk file is 96MB", but this appears to be limited to Raspberry Pi4 with observed behavior. On Pi5, this fails to load past 30MB. Below 30MB, the kernel and a small initramfs shell (1.3MB) is successfully loaded.

If the signed boot is disabled but ramdisk is set in config, the boot.img (Tested around 60MB) *is* able to be read and kernel loads, however, the initramfs fails to load and continues to main root partition. Below 30MB, the kernel and initramfs shell successfully loads.

Some clarifications on the size limits over ramdisk vs signed boot would be helpful. If possible, raising this limit like what was done with 2711 would help allow for bigger initramfs with tools to support LUKS encryption and decryption before main rootfs mounting.

### Steps to reproduce the behaviour

Build a signed boot.img greater than 30MB.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=1

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB (RP1), NVMe PCIe, then network
BOOT_ORDER=0xf146

# Disable self-update mode - to prevent automatic updates of unsigned bootloader images
ENABLE_SELF_UPDATE=0

# Select signed-boot mode in the EEPROM. This can be used to during development
# to test the signed boot image. Once secure boot is enabled via OTP this setting
# has no effect i.e. it is always 1.
SIGNED_BOOT=1

### System

OS/Version: Yocto-based Kirkstone Distribution
Bootloader Version:  
```
2024/06/05 16:41:49
version 6fe0b091c7cb1c5da26b7f41cc24c42840531a83 (release)
timestamp 1717602109
update-time 1729103195
capabilities 0x0000007f
```

Firmware Version:
```
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
```

Kernel Version: `Linux CAM-2CCF67509B94 6.1.77-v8-16k #1 SMP PREEMPT Thu Feb  8 15:07:10 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`

### Bootloader logs

```
  1.27 RPi: BOOTSYS release VERSION:6fe0b091 DATE: 2024/06/05 TIME: 16:41:49
  1.31 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1717602109 serial 8f7b63c3 boardrev d041
70 stc 1631177
  1.41 AON_RESET: 00000003 PM_RSTS 00001000
  1.48 RP1_BOOT chip ID: 0x20001927
  1.51 PM_RSTS: 0x00001000
  1.51 part 00000000 reset_info 00000000
  1.55 PMIC reset-event 00000000 rtc 671043a8 alarm 00000000 enabled 0
  1.61 uSD voltage 3.3V
  1.80 Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
[... 345 lines cut ...]

dlfeat  : 1
  [4:4] : 0	Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0	Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0x1	Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 1000204886016
mssrl   : 0
mcl     : 0
msrc    : 0
nsattr	: 0
nvmsetid: 0
anagrpid: 0
endgid  : 1
nguid   : 00000000000000000000000000000000
eui64   : 002538b34140c14a
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
```

learmj, 2024-10-17:
> ramdisk is set in config, the boot.img (Tested around 60MB) is able to be read and kernel loads, however, the initramfs fails to load and continues to main root partition. Below 30MB, the kernel and initramfs shell successfully loads.

Hi. To avoid any doubt, for both signed and signed boot scenarios, please state if your initramfs archive is embedded into the kernel image (image is a single monolithic blob), or if it's stored as a separate file (kernel and initramfs are separate)?

To verify if authentication is working correctly on a monolithic blob, you could create a file containing random data of a certain size, sign it, then check if the bootloader authenticates it upon load.

SyedFaisalAli, 2024-10-17:
Hi!

> Hi. To avoid any doubt, for both signed and signed boot scenarios, please state if your initramfs archive is embedded into the kernel image (image is a single monolithic blob), or if it's stored as a separate file (kernel and initramfs are separate)?

The Initramfs archive is stored as a separate file within boot.img. I forgot to show my config.txt in boot.img:

```
enable_uart=1
uart_2ndstage=1
disable_overscan=1
dtoverlay=dwc2,dr_mode=peripheral,nohdmi
start_file=start4.elf
fixup_file=fixup4.dat
initramfs initramfs.cpio.xz followkernel
```

Outer config in boot partition:
```
enable_uart=1
boot_ramdisk=1
uart_2ndstage=1
```

> To verify if authentication is working correctly on a monolithic blob, you could create a file containing random data of a certain size, sign it, then check if the bootloader authenticates it upon load.

So, I created a 30MB file and signed:
```
dd if=/dev/urandom of=boot.img bs=1024 count=30000
rpi-eeprom-digest -i boot.img -o boot.sig -k private.pem
```

After copying to the boot partition in our nvme stick, it verifies successfully:
```
 43.37 secure-boot
 43.39 Loading boot.img ...
 43.42 boot.sig
 43.43 hash: 01d752b596c5c8d277482a4e93b2e77f26cd2e3106363a91e67033e55eeddf9c
 43.50 ts: 1729176130
 43.52 rsa2048: 49adf45767915153ab089e886cd2e9bc4e88f5074f7b333c09b84d17f317fb08968216daee2a0d448b1
dc80ab267cdbddc8eae890724c95257c1d6bbe9aeb38a2a3f65813177f3c06e5e4e5b671b0ab16aa6ce8df895509ee1bffe
144b963093bbbe3c7907168aa6015015d1a7692fff1044edc11775c017d08f3ff3ddf0f8c398a04e30f9705a1570521a28d
17643beec93a13061f2e32b697aa4d3fb7757975d1f8fd006dcb8212034d03956cf9128e571d39335fc90a4e22a2192fcfd
b24c456e8b177f81bb9306a91d7b7d56d81bee712a47abd6516e2c7d700c1dbeff066132f8bfaf944fa0d98896eac4b91c7
05d63818ac1e2c9e07fa4a461b585ea7f
 44.14 Verifying
 47.07 RSA verify
 47.19 rsa-verify pass (0x0)
```

I then create a 31MB file and sign:
```
dd if=/dev/urandom of=boot.img bs=1024 count=31000
rpi-eeprom-digest -i boot.img -o boot.sig -k private.pem
```

Fails to load with bad signature.
```
134.69 type: 16 lba: 16384 'mkfs.fat' '  V       ^ ' clusters 42551 (8)
134.75 rsc 8 fat-sectors 168 root dir cluster 1 sectors 32 entries 512
134.82 FAT16 clusters 42551
134.84 secure-boot
134.86 Loading boot.img ...
134.89 boot.sig
134.90 Failed to parse signature file
134.93 Bad signature boot.sig
134.96 Error 12 loading boot.img
```

learmj, 2024-10-18:
Your original description seems different to the 31 MB test, though.
```
  6.27 Loading boot.img ...
  6.30 boot.sig
...
  7.47 Verifying
 14.00 Bad signature boot.img
 14.54 Error 12 loading boot.img
```
vs
```
134.86 Loading boot.img ...
134.89 boot.sig
134.90 Failed to parse signature file
134.93 Bad signature boot.sig
134.96 Error 12 loading boot.img
```
The actual authentication check of the image doesn't look like it actually happened.

timg236, 2024-10-21:
I just tried the 31MB image on a 2024/09/23 bootloader using the same command as you and it worked fine on SD, USB, NVMe

```
dd if=/dev/urandom of=boot.img bs=1024 count=30000
rpi-eeprom-digest -i boot.img -o boot.sig -k private.pem
```

Possibly a mismatch in the key or an issue with the NVMe. 

It might be worth updating the bootloader and replacing the mass-storage-gadget64/boot.img file to see if you can still reproduce the issue.

SyedFaisalAli, 2024-10-21:
Hi all, thanks so far!

Yes, it looks like I was mistaken (I might not have synced or copied the signature file correctly the second time). So, at 31M, the boot.img passed.

I'm trying higher sizes now by increasing every 10M. Just on the NVMe, this is the results I'm seeing, but I'll try the SDCard as well to see if there's any issue with the nvme on my end.
```
dd if=/dev/urandom of=boot.img bs=1024 count=31000 # Passed
dd if=/dev/urandom of=boot.img bs=1024 count=40000 # Passed
dd if=/dev/urandom of=boot.img bs=1024 count=50000 # Failed with Bad signature boot.img
```

> It might be worth updating the bootloader and replacing the mass-storage-gadget64/boot.img file to see if you can still reproduce the issue.
Will update the bootloader as well. Last I tried `mass-storage-gadget64/boot.img`, it did load. The default image in the repo is about 30MB, but I don't believe I tried replacing it.

SyedFaisalAli, 2024-10-23:
It turns out, the boot partition flashed to the nvme from our Yocto WIC image was not set up to format with FAT32, defaulting to FAT16 instead. This was the major difference after comparing with the Raspberry Pi Imager, which flashed its boot partition on the nvme with FAT32, and allowed loading of a larger boot.img without issue.

Oddly, this only mattered for the nvme as we were able to flash our WIC image onto an SDCard with FAT16 formatted boot and load the boot.img. We were also able to load a larger boot.img from USB on our default and the latest eeprom.

In any case, I appreciate the guidance! Feel free to close this ticket if needed.

timg236, 2025-05-15:
I think we reached the end of this investigation. If anyone is able to create (and share) a FAT image containing a boot.img which doesn't work then please create a fresh issue attaching the image.

closed 2025-05-15 by timg236