raspberrypi/rpi-eeprom #629: Rpi 5 NVMe boots only if USB-MSD is there

doubleuerner, 2024-11-07:
### Describe the bug

I am struggling with booting from an NVMe disk (Toshiba KBG40ZNS256G) and after a couple of mistakes and spending fun but also confusing hours i think i discoverd a bug or a symptom caused by doing something wrong or missing something , or it is a bug for real. Who knows? I'll try to explain:

I found out that simply using dd an image to a NVMen0 (4096 sector size) with bs=515(default) gives a weird situation. But the SD card copier tool helped to overcome.

Some info about my setup:

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

picture:

I've cloned USB-MSD into NVMe disk using the "SD card Copier" tool. 
I changed the boot order,

BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
NET_INSTALL_AT_POWER_ON=1

During power up SD card is not present, USB thumbdrive neither.

I would expect to boot from NVMe, since i changed the boot order:
.

What i get is a pi that doesnt boot and in the diagnotic screen:

picture: 

nvme: error 10 

But now the suprise:

As soon i connect a USB-MSD, it boots from NVMe, it's so fast you could even not see any 'bios' screen or diagnostic screen.

I am happy my goal has been reached, but only to trick the pi with a USB-MSD first. after start up it is not mounted and i can just remove it.

root@rpi5-usb:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1  58.6G  0 disk
├─sda1        8:1    1   512M  0 part
└─sda2        8:2    1  58.1G  0 part
nvme0n1     259:0    0 238.5G  0 disk
├─nvme0n1p1 259:1    0     4G  0 part /boot/firmware
└─nvme0n1p2 259:2    0 234.4G  0 part /

It's not the case a config.txt or cmdline.txt is read from USB-MSD, because it requires then to do something with UUID or correct devices (eg sda / sdb) 

The last thing i want now is to get rid of the USB-MSD to be injected as a prerequisit to boot the NVMe ... or its a bug may be?
 

### Steps to reproduce the behaviour

NVMe disk partitioned and having OS installed.

power up without SD ; but with USB-MSD : it boot from NVMe disk

power up without SD ; and NO USB-MSD : nvme error 10, boot fails and others (SD, USB ) are try out.
[image]
[image]

### Device (s)

Raspberry Pi 5

### Bootloader configuration.
[... 742 lines cut ...]
  [4:4] : 0     Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0     Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0x1   Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 256,060,514,304
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : 01000000000000008ce38e040011a073
eui64   : 8ce38e040011a073
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x3 Degraded
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0x1 Better (in use)
root@rpi5-usb:~#

peterharperuk, 2024-11-07:
My guess is that there's something weird with your boot partition on the NVME drive. Why is it 4G in size? Are the cmdline.txt files the same on the USB disk and the NVME disk? I boot my Pi 5 from NVME all the time, so you've probably done something wrong.

doubleuerner, 2024-11-07:
well, if i just download the img.xz and unpack it and use dd to write it to NVMe disk, then i get this 4G boot partition. I discovered that dd used 512 and the nvme 4096 blocksize. So then i looked with fdisk into the img-file and see a partition of 512M, and that makes sense because the bs = x8, so 4GB. so i did all by hand using parted/fdisk and mounted and copied all files. Then still i doesnt work, and used the rpi-imager: there the 4GB boot partition was there , no succes. Then used sd card copier, and eventually no succes, until i also used the SUB-MSB stick. And yes the content on NVMe is exact the same as USB. so also cmdline.txt.

peterharperuk, 2024-11-07:
We don't support 4096 byte block size in the bootloader. That's the problem.

doubleuerner, 2024-11-07:
Aha, okay then I go immidatly change the sector size to 512 and try to dd the img to nvmen1. Thanks for the hint! I'll let you know the result!

https://man.archlinux.org/man/nvme-format.1.en

doubleuerner, 2024-11-07:
It works! 

Thanks , i learned a lot having tried 'everything' and taking all kinds of directions, but finally it works and i understand why.

peterharperuk, 2024-11-07:
One day we might fix this. It would be interesting to know if performance is noticeably better when the linux partition uses a 4096 byte block size.

peterharperuk, 2024-11-07:
Actually, reopen. We should probably throw a more obvious error to stop people wasting their time with this in future?

doubleuerner, 2024-11-07:
Well, i think this struggle appearing to be sector size other than 512 is very common in NVMe is just Google on this a few minutes ago. The NVMe card i have was from an old windows machine. I think , me as an hobbyist as many others re-use stuff and try to make it work, i don't know if you buy a new NVMe the default is 512 of something else. I thing even the rpi-imager uses dd as underlying method and i know this turns into weird partition table and filesystem not being mount-able.

doubleuerner, 2024-11-07:
The M.2 HAT+ has documentation explaining it all, but not this exception of 4096 sector size able to boot.

peterharperuk, 2024-11-07:
Yeah, even I missed "nvme: error 10" in your screen grab which means NMVE_ERROR_UNSUPPORTED_BLOCK_SIZE

peterharperuk, 2024-11-07:
I'm just suggesting making the error more obvious for now.

doubleuerner, 2024-11-07:
> Yeah, even I missed "nvme: error 10" in your screen grab which means NMVE_ERROR_UNSUPPORTED_BLOCK_SIZE

google on nvme: error 10 yesterday didn't show up something, also i tried to see if the source code is somewhere. But is this something that can be found by me for instance, or is it something in the code nobody except the developers can find?

closed 2025-04-07 by timg236

pelwell, 2025-04-10:
If your sector size is 4096 bytes then unfortunately the firmware won't be able to boot from it - trying again and again won't make any difference. As @timg236 said above:
> The feature is on the todo list (but on the back burner)

Which is why the issue was closed as "Not planned".

gshpychka, 2025-04-10:
> > The feature is on the todo list (but on the back burner)
> 
> Which is why the issue was closed as "Not planned".

But "on the todo list" and "not planned" are very different. What does "issue understood" mean (if you know)?

timg236, 2025-04-10:
Support for 4K native sectors is acknowledged as a useful but unresourced and unplanned enhancement. If you want to be alerted to any progress in this area please watch this issue https://github.com/raspberrypi/rpi-eeprom/issues/577

Skydea777, 2025-04-10:
No that is the thing the Sector Size is 512:

`sudo lshw -class disk -class storage
  *-nvme
       description: NVMe device
       product: E2M2 64GB
       vendor: O2 Micro, Inc.
       physical id: 0
       bus info: pci@0001:01:00.0
       logical name: /dev/nvme0
       version: 10100080
       serial: MJU412D00558B
       width: 64 bits
       clock: 33MHz
       capabilities: nvme pm msi pciexpress msix nvm_express bus_master cap_list
       configuration: driver=nvme latency=0 nqn=nqn.2014.08.org.nvmexpress:12170002MJU412D00558B state=live
       resources: irq:38 memory:1b00000000-1b00003fff
     *-namespace:0
          description: NVMe disk
          physical id: 0
          logical name: hwmon1
     *-namespace:1
          description: NVMe disk
          physical id: 2
          logical name: /dev/ng0n1
     *-namespace:2
          description: NVMe disk
          physical id: 1
          bus info: nvme@0:1
          logical name: /dev/nvme0n1
          size: 57GiB (61GB)
          capabilities: partitioned partitioned:dos
          configuration: logicalsectorsize=512 sectorsize=512 signature=ca0cc567 wwid=eui.01000000000000000000000000000000
`

And:
`pi@raspberrypi5:~ $ sudo nvme list /dev/nvme0
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            MJU412D00558B        E2M2 64GB                                1          61,87  GB /  61,87  GB    512   B +  0 B   10100080`

What Infos can I get for you? Because I know how .... tiring it is to try to troubleshoot something like this, as I said.. 30+ hours.

doubleuerner, 2025-04-10:
@Skydea777  Have you used nvmetool : https://man.archlinux.org/man/nvme-format.1.en
?
In this you can change on hardware level to 512 bytes per sector

Skydea777, 2025-04-10:
Yes I did, it has the correct Hardware Level.

I finally found something....
But now, I take deep breath, because I searched the Chinese Internet after this damned NVME Drive. (foresee e2m2 64gb)

And you know what?
It is the Standard 64GB Beginner NVME Drive that gets shipped with it.

This ... drive... is an eMMC.
IN GODS YEAR OF 2025 SOME HEATHENS STILL PRODUCE eMMC DRIVES!

https://global.microless.com/product/foresee-e2m2-64gb-m-2-pcie-gen-2-internal-ssd-up-to-410-mb-s-sequential-read-250-mb-s-sequential-write-speeds-1-5-m-hours-mtbf-compatible-with-steamdeck-surface-laptops-fe2h0m064g-b5x10/

30+ Hours for nothing!
I just ordered a new nvme drive, this is ridiculous.

Thank You guys, now I know, even if the Sector/Block Size is correct.
Some Bastard can still sell an NVME Drive that is in reality an eMMC Drive.
And it will present you the same Error.

You guys cannot imagine what kind of rage courses through me.

derlaft, 2025-05-05:
>  Some Bastard can still sell an NVME Drive that is in reality an eMMC Drive.

> Steamdeck 64GB NVME Drive

Steamdeck was always clearly advertised by valve to have eMMC storage. 

> This ... drive... is an eMMC.

Not sure what's the exact problem here. Shouldn't nvme-attached eMMC still work?

timg236, 2025-05-06:
Locked issue - the Raspberry Pi forums are the best place to have ad-hoc discussions about interop issues.