raspberrypi/rpi-eeprom #632: Cannot intersept the boot menu when NVME is set to boot before USB

solskogen, 2024-11-19:
### Describe the bug

With the eeprom from Tue Nov 12, I cannot get to the boot menu if NVME is set to boot before USB.

### Steps to reproduce the behaviour

Holding space with this won't get me to the boot menu
BOOT_ORDER=0xf461

But with this, I can.
BOOT_ORDER=0xf641

In this case both the NVME and USB contains a Raspberry Pi OS system.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```
# rpi-eeprom-config
[all]
BOOT_UART=1
BOOT_ORDER=0xf461
POWER_OFF_ON_HALT=1
PSU_MAX_CURRENT=5000
PCIE_PROBE=1
```

### System

N/A

peterharperuk, 2024-11-19:
This seems to work for me. Do you have your keyboard plugged directly into the Pi?

solskogen, 2024-11-19:
No, I don't. It's via the monitors hub. Directly seems to work fine. But it should work with the hub, correct?

pelwell, 2024-11-19:
(It's not about whether or not it should work, it's about understanding how your system differs)

peterharperuk, 2024-11-19:
If you add "NET_INSTALL_AT_POWER_ON=1" to your eeprom config it might start working. I am working on speeding up keyboard enumeration but it's a tradeoff between fast boot time and waiting for USB to find keyboards.

lurch, 2024-11-19:
@solskogen I guess you could also try increasing https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#NET_INSTALL_KEYBOARD_WAIT ?

solskogen, 2024-11-19:
@peterharperuk Not did not have any effect, but
@lurch setting NET_INSTALL_KEYBOARD_WAIT to 1200 worked! (1100 was not enough, however)

lurch, 2024-11-19:
Just in case it's not obvious what is going on here:
* The Pi firstly has to detect and enumerate all the USB devices connected directly to it (which takes a certain amount of time)
  * Then if any of those devices are a USB hub:
  * It has to detect and enumerate all the devices connected to the hub
    * Then if any of _those_ devices are a hub.... etc.

This is why it takes longer for your USB keyboard to be detected when it's plugged into your monitor, than when the keyboard is plugged directly into your Raspberry Pi.

solskogen, 2024-11-19:
Still a bit strange that this is an issue when it's set to boot from NVME first, but not when set to (try) booting from USB before NVME.

elgerg, 2024-11-19:
> Still a bit strange that this is an issue when it's set to boot from NVME first, but not when set to (try) booting from USB before NVME.

It kind of does. It would have to enumerate all USB devices before failing to find a drive then moving on to NVME.

lurch, 2024-11-19:
It'll also be doing USB-enumeration when doing boot-from-USB, and obviously it detects the keyboard at the same time as it's looking for a bootable USB drive.

closed 2024-11-20 by solskogen