raspberrypi/rpi-eeprom #636: Recover broken SPI rom (Rpi5)

nbuchwitz, 2024-11-26:
### Describe the bug

I ran an apt upgrade on a RPi5. After power down the device won't boot and presents an LED error code as soon as I connect the power supply: SPI EEPROM error - Pi 4 and Pi 5

How can I recover the broken device?

### Steps to reproduce the behaviour

- sudo apt update
- sudo apt upgrade (devices shuts down, might be unrelated to the actual update)

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Cannot provide this, as the device won't boot anymore

pelwell, 2024-11-26:
This: https://github.com/raspberrypi/rpi-eeprom#reset-to-factory-defaults ?

peterharperuk, 2024-11-26:
RPI imager has a menu "misc utility images" which has bootloader images which you can flash to a blank SD card to recover the bootloader.

nbuchwitz, 2024-11-26:
Thanks, already tried that but the device keeps blinking SOS. I will try some other sdcards later. First I will have to deal with the joy of pcie switch erratas ...

pelwell, 2024-11-26:
usbboot/rpiboot should also work: `rpiboot -d recovery5`, I think.

timg236, 2024-11-26:
> Thanks, already tried that but the device keeps blinking SOS. I will try some other sdcards later. First I will have to deal with the joy of pcie switch erratas ...

Please could you try  the SD-card result again and then power up the Pi without an SD-card or other boot-devices? It would be good to confirm whether the SD card update is failing or if it's immediately failing as a result of something on the SD-card.

Also, what size of SDRAM is it?

timg236, 2024-11-26:
Boot UART output from the SD card rescue would also be really helpful if you could provide that.

nbuchwitz, 2024-11-26:
Unfortunately I already fixed it with another sd card and a durability test case is running. After this is over (probably end of this week), I can test again with the broken SD and provide uart logs. 

btw: `rpiboot -d recovery5` did not work, nothing was detected. Anything like the nrpiboot pin on cm?

timg236, 2024-11-26:
On a Pi5 if you press and hold the power button before connecting the USB-C cable it will go into rpiboot mode.

The recovery.bin does a sha256 check of pieeprom.sig before flashing it, but if something was broken in the low-speed memory setup then that might cause this. However, APT update uses flashrom so I can't see how the initial problem occured.

timg236, 2024-12-16:
If this ever happens again it would be useful to grab the systemd journal from rpi-eeprom-update from the SD card in-case it was an update that triggered this. "sudo journalctl -u rpi-eeprom-update"

I've never seen flashrom fail, but as a belts and braces change I've update the script to be be more paranoid. It now explicitly calls "flashrom verify" even though default flashrom is to verify. It does this in a retry look and lowers the SPI clock frequency.
https://github.com/raspberrypi/rpi-eeprom/commit/af0426b473b3b117819435c76ac713ee64095f89

nbuchwitz, 2024-12-16:
I will keep it in mind and probably will try to reproduce the issue with the sd during the holidays.

timg236, 2025-04-07:
Closing as stale issue - the write / verify retry loops in rpi-eeprom-update are have been in APT for some months now and we've not had other reports.
If anyone can reliably reproduce a problem like this then please raise a new Issue so that can try to replicate the problem here.

closed 2025-04-07 by timg236