raspberrypi/rpi-eeprom #643: BOOT_UART default 1 breaks hats using gpio 14 and 15

digitalLumberjack, 2024-12-14:
### Describe the bug

Hello :)

With the commit https://github.com/raspberrypi/rpi-eeprom/commit/d57c084c9fd2cb7fc08f4b1ab2387df695067371#diff-637f7b97bba458badb691a1557c3d4648686292e948dbe3e8360564378b653efR83

The bootloader enable by default BOOT_UART=1 and break support for hats using gpio 14 and 15

### Steps to reproduce the behaviour

Update bootloader, reboot with a hat using gpio 14 and 15, see the error.

When setting BOOT_UART=0, it works again

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf41

pelwell, 2024-12-14:
We are aware, and a fix is in progress. In the meantime, `enable_uart=0` in config.txt should work around the problem.

lurch, 2024-12-14:
> With the commit [d57c084](https://github.com/raspberrypi/rpi-eeprom/commit/d57c084c9fd2cb7fc08f4b1ab2387df695067371)
> 
> The bootloader enable by default BOOT_UART=1 and break support for hats using gpio 14 and 15

FYI I suspect it was actually https://github.com/raspberrypi/rpi-eeprom/commit/fe7bfc720165464d9dfe2f85fe090ca22a625bd7 that changed the behaviour here. (This is just for information; I'm sure that Phil's fix will cure this issue)

digitalLumberjack, 2024-12-14:
Thank you, sorry if the feedback was a little bit to early.

lurch, 2024-12-14:
No need to apologise, there's no way for you to know that we were already aware of the issue! Thank you for your report.
We spotted this issue ourselves only a couple of days ago :slightly_smiling_face:

timg236, 2024-12-18:
I think the apt update will be wk2 now

timg236, 2024-12-18:
@nbuchwitz We managed to get the update out today - last one of the year I hope :)

nbuchwitz, 2024-12-19:
Thanks. I guess this can be closed then?

closed 2024-12-19 by timg236