raspberrypi/rpi-eeprom #646: Raspberry PI 5 failed to boot with an nvme hard drive

Artells, 2025-01-03:
### Describe the bug

Raspberry PI 5 is installed on an nvme hard disk and only ubuntu 24.04 series of systems can boot normally, but the compatibility of the new system is not good. After I tried ubuntu 20.04 and 22.04, I couldn't boot with nvme hard disk. How can I solve this problem?

### Steps to reproduce the behaviour

Create ubuntu 22.04 server lts firmware in windows using the official tool, then enter with the tf card installed in the official raspberry pi os system, change the eeprom Settings to boot with nvme. Start after the Settings are complete.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[image]

peterharperuk, 2025-01-03:
What version eeprom are you using? Any logs? At which point does it fail to boot? You haven't included enough detail with this report to get much help.

pelwell, 2025-01-03:
> ubuntu 24.04 series of systems can boot normally, but the compatibility of the new system is not good

What do you mean by this?

Artells, 2025-01-03:
> > Ubuntu 24.04 系列系统可以正常启动，但新系统的兼容性不佳
> 
> 这是什么意思？

PHP8.0 does not work properly after installation.

pelwell, 2025-01-03:
> PHP8.0 does not work properly after installation.

I can't help thinking that should be an easier problem to solve.

nbuchwitz, 2025-01-03:
Most definitely yes.
I can recommend @oerdnj s  repositories. He is the php debian php maintainer and packages even more versions in his sury.org repositories. You can use them with Ubuntu or even better raspberry pi os: https://deb.sury.org/

Bonus point when usinh rpi os: firmware, eeprom and kernel directly from the vendor ;)

timg236, 2025-01-06:
This issue does not include enough information to identify the issue or investigate further - please see the issue template which prompts for details of the NVMe hardware and bug.
For general advice on interop please use the Raspberry Pi forums.

timg236, 2025-04-07:
Stale issue - closing

closed 2025-04-07 by timg236