raspberrypi/rpi-eeprom #647: `ioctl_set_msg failed:-1` from `rpi-eeprom-config` after reboot

srcshelton, 2025-01-05:
### Describe the bug

I used `EDITOR=vim rpi-eeprom-config -e` to create a ~2248-byte firmware configuration file on a series of CM4 devices with the latest firmware (`1733575168`/Sat 7th Dec) - this change was accepted and an update written to apply on reboot.

After rebooting, however, I can't recall the current configuration:

```
sudo rpi-eeprom-config
ioctl_set_msg failed:-1
bootloader_config
```

So there appear to be two issues here:

* `rpi-eeprom-config` has allowed a configuration to be deployed which it can't later read-back;

* If there is an error, then `rpi-eeprom-config -e` will open an editor with the error message, rather than erroring-out - potentially allowing a valid configuration to be overwritten with the error message if the user isn't careful to erase the content and save before exiting their editor (which is likely non-obvious to most users, even those making low-level changes such as firmware configuration edits).

### Steps to reproduce the behaviour

Upload the following firmware configuration file with `EDITOR=vim rpi-eeprom-config -e`:

```
[all]
#BOOT_UART=0
BOOT_UART=1

# N.B. Not relevant on [pi5], [rpi500], [cm5].
# Set to 2 to use an external GPIO power button.
#
#WAKE_ON_GPIO=1
WAKE_ON_GPIO=0

# N.B. Defaults to 1 on [pi400], [pi500], [cm5].
#
#POWER_OFF_ON_HALT=0
POWER_OFF_ON_HALT=1

# Boot Order Codes, from [1].
#
# Try SD first (1), followed by NVMe PCIe ([cm4] & [pi5] only), USB
# mass-storage, then loop.
#
#  0x0 - Deprecated
#  0x1 - SD card (or eMMC)
#  0x2 - Network
#  0x3 - RPiBOOT ([cm4])
#  0x4 - USB-MSD
#  0x5 - BCM-USB-MSD (USB2.0 boot from USB-C socket or CM4IO USB-A), not [pi5]
#  0x6 - NVME ([cm4], [pi5])
#  0x7 - HTTP
#  0xe - STOP (display error pattern, reboot needed to continue)
#  0xf - RESTART
#
# [1]: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
#
#BOOT_ORDER=0xf41

# Set to -1 for infinite retries.
#
#MAX_RESTARTS=-1
#SD_BOOT_MAX_RETRIES=0
#NET_BOOT_MAX_RETRIES=0
#DHCP_TIMEOUT=45000
#DHCP_REQ_TIMEOUT=4000
#TFTP_FILE_TIMEOUT=30000
#TFTP_IP=192.168.1.99
#TFTP_PREFIX=0
#TFTP_PREFIX_STR=
#PXE_OPTION43=Raspberry Pi Boot
[... 220 lines cut ...]
  2.67 GPT partition: 11 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000201800..0002217ff
  2.74 Matched GUID bootable-part-idx 0 want partition 0
  2.79 type: 32 lba: 2103296 'BSD  4.4' ' BOOT_0     ' clusters 129024 1)
  2.86 rsc 32 fat-sectors 1008 root dir cluster 2 sectors 0 entries 0
  2.92 FAT32 clusters 129024
  2.98 Read config.txt bytes     7373 hnd 0x1ea53
  3.03 Read overclock.txt bytes     2296 hnd 0x1664f
  3.08 Read legacy.txt bytes     3563 hnd 0x16648
  3.11 pieeprom.sig
  3.12 hash: 0cef128231dd69ae90ae97f3fddd1215b94333bf17bf1ff0dcaae2e03351c901
  3.18 ts: 1735278939  
  3.06 SELF-UPDATE timestamp current 1735278939 new 1735278939 skip
  3.12 [sdcard] recover4.elf not found
  3.15 [sdcard] recvery.elf not found
  3.64 Read start4cd.elf bytes   812636 hnd 0x16014
  3.67 Read fixup4cd.dat bytes     3230 hnd 0x1f5b7
  3.70 0x00d03140 0x0000000 0x00001fff
  3.74 MEM GPU: 16 ARM: 998 TOTAL: 1014
  3.00 Firmware: 2ae30f53898ae2f1ba77ff570a92991bedfb0398 Nov 26 2024 12:52:41
  3.17 Starting start4cd.elf @ 0xff000200 partition 0
  3.19 PCI0 reset
  3.30 +

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
```

timg236, 2025-01-06:
What does "vcgencmd bootloader_config" report  in this configuration?

That's a rather larger EEPROM config but should be legal

srcshelton, 2025-01-06:
> What does "vcgencmd bootloader_config" report in this configuration?

Same output:

```
$ sudo vcgencmd bootloader_config
ioctl_set_msg failed:-1
bootloader_config
```

The configuration does appear to have applied, as `/proc/cmdline` contains:

```
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 numa=fake=2 system_heap.max_order=0 smsc95xx.macaddr=E4:5F:01:A8:3B:24 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=tty0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 root=PARTUUID=xxxx rootfstype=ext4 fsck.repair=yes rootwait cpufreq.default_governor=performance cgroup_enable=memory cgroup_memory=1 zswap.enabled=1 zswap.compressor=lz4 zswap.max_pool_percent=20 zswap.zpool=z3fold
```

… which includes `numa_policy=interleave […] numa=fake=2` (which doesn't appear in `cmdline.txt`), and I believe that is only inserted if the EEPROM configuration contains `SDRAM_BANKLOW=3` (unless that's now default?) which would suggest that the configuration can be applied but is not readable 😯

pelwell, 2025-01-06:
Error -1 is -EPERM. Running under strace, i.e. `sudo strace rpi-eeprom-config`, while very verbose, should enable you to work out which system call is failing.

srcshelton, 2025-01-06:
Additionally, almost the same configuration file (with the `WAKE_ON_GPIO` and `POWER_OFF_ON_HALT` overrides removed) works as expected on an RPi5 and RPi500 (albeit both running Debian).

srcshelton, 2025-01-06:
```
$ sudo strace -vvffs 128 -- vcgencmd bootloader_config 2>&1 | grep -FC 5 -e EINVAL -e ioctl_set_msg
mprotect(0x558e35f000, 4096, PROT_READ) = 0
mprotect(0x7fa8802000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7fa87f9000, 14850)             = 0
openat(AT_FDCWD, "/dev/vcio", O_RDONLY) = 3
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0x64, 0, 0x8), 0x7fdf236108) = -1 EINVAL (Invalid argument)
fstat(1, {st_dev=makedev(0, 0xd), st_ino=275688, st_mode=S_IFIFO|0600, st_nlink=1, st_uid=1000, st_gid=100, st_blksize=4096, st_blocks=0, st_size=0, st_atime=1736178913 /* 2025-01-06T15:55:13.114061971+0000 */, st_atime_nsec=114061971, st_mtime=1736178913 /* 2025-01-06T15:55:13.114061971+0000 */, st_mtime_nsec=114061971, st_ctime=1736178913 /* 2025-01-06T15:55:13.114061971+0000 */, st_ctime_nsec=114061971}) = 0
getrandom("\xc8\xab\x1e\xa5\x25\xf3\x39\xb1", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x559ff75000
brk(0x559ff96000)                       = 0x559ff96000
close(3)                                = 0
write(1, "ioctl_set_msg failed:-1\nbootloader_config\n", 42ioctl_set_msg failed:-1
bootloader_config
) = 42
exit_group(0)                           = ?
+++ exited with 0 +++
```

… so it seems to be the VCIO driver which is unhappy, even though other `vcgencmd` invocations succeed.

And also note that the only system change between this command working and now when it doesn't was uploading the new configuration and then rebooting… although that could be due to a formerly loaded module now not being automatically loaded?

Update: I've loaded every module for the current kernel, and I'm still getting the same error.

timg236, 2025-01-06:
Does it still fail if you make the configuration file smaller e.g. delete the comments to get it under 2K ?

srcshelton, 2025-01-06:
Hmm - it seems not: I've reduced the length to 1677 characters and I'm still seeing the same behaviour.  I guess that either means that this occurs well below 2k in size, that once the problem is triggered it keeps happening, or that it's related to the latest firmware or kernel releases (as both were updated between when this last worked as anticipated and now).

Is there any way (… short of recompiling the kernel with an updated driver with debug enabled?) to get more information about what's causing the ioctl syscall on the `vcio` device to return -1?

pelwell, 2025-01-06:
I think the limit is probably in vcgencmd in our utils repo. You can try increasing this constant: https://github.com/raspberrypi/utils/blob/master/vcgencmd/vcgencmd.c#L44

srcshelton, 2025-01-06:
> I think the limit is probably in vcgencmd in our utils repo. You can try increasing this constant: https://github.com/raspberrypi/utils/blob/master/vcgencmd/vcgencmd.c#L44

Happy to give that a try!  Are you able to tell me the physical limit/size of the EEPROM area reserved for bootloader configuration that sets an upper-limit for this constant?

pelwell, 2025-01-06:
For now, just make it 10 times larger to see if it fixes the problem.

srcshelton, 2025-01-06:
That's got it - I increased `MAX_STRING` to `4096` (and moved from commit https://github.com/raspberrypi/utils/commit/aaeac09691f37f1d33c8c5c61c1de91fc7bc64d9 to commit https://github.com/raspberrypi/utils/commit/33aa4da93d8da90d5075e8cd48e72cc90048b9f3), and things now work as expected.

This post - https://forums.raspberrypi.com/viewtopic.php?t=243549 seems to suggest that RPi4/CM4 models have 512KiB EEPROM capacity… should this therefore be the limit in https://github.com/raspberrypi/utils/blob/master/vcgencmd/vcgencmd.c#L44?

pelwell, 2025-01-06:
Were the legitimate, useful content to expand to the point that an existing EEPROM configuration no longer fitted, the tools would fail and users would get upset. It's much better to set a sensible limit that we will always support and that cannot be exceeded - 4kB feels about right, but I wouldn't object to 2kB.

timg236, 2025-01-06:
4K is a safe bet, it certainly won't exceed 8KB because there isn't enough free space in the Pi4/CM4 SPI flash to have a larger config file!

timg236, 2026-11-04:
vcgencmd limit was updated a while ago, closing as fixed.

closed 2026-11-04 by timg236