raspberrypi/rpi-eeprom #652: Raspberry Pi 5 with 16GB RAM does not boot after rpi-eeprom-config -e or a firmware update.

cmd0s, 2025-01-13:
### Describe the bug

Raspberry Pi 5 with 16 GB RAM does not boot after rpi-eeprom-config -e or a firmware update.
I don’t change anything after running the rpi-eeprom-config -e command, yet after a restart, the device doesn’t boot. The red LED stays lit. Pressing the power button does nothing. Disconnecting and reconnecting the power supply doesn’t help either.

After reflashing the SD card, the Raspberry Pi works as it did before.
I am using a clean Ubuntu 24.04.1 LTS 64-bit image flashed with Raspberry Pi Imager.

I also tried updating the firmware to pieeprom-2025-01-13.bin, but the result was the same.
The same operations on a Raspberry Pi 5 with 8 GB RAM work as expected.

### Steps to reproduce the behaviour

HW: Raspnerry Pi 5 16 GB

1. Using Raspberry Pi Imager flash microSD card with Ubuntu Server 24.04.1 LTS 64 Bit
2. Run Rpi with this microSD card.
3. Login to ssh
4. sudo rpi-eeprom-config -e
5. ctrl-x
6. sudo reboot

Now Rpi 5 16 GB will not boot (in my case)

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461

### System

$ vcgencmd bootloader_version
2024/09/23 14:02:56
version 2682625908f5585e5f4832bb82e74e7d757ec48f (release)
timestamp 1727096576
update-time 0
capabilities 0x0000007f

$ vcgencmd version
2024/09/23 14:02:56
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)

$ uname -a
Linux ubuclean 6.8.0-1010-raspi #11-Ubuntu SMP PREEMPT_DYNAMIC Thu Aug  8 23:22:41 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

timg236, 2025-01-13:
Please can you clarify this statement
"After reflashing the SD card, the Raspberry Pi works as it did before."

This shouldn't change the bootloader firmware so it's hard to see how this could change bootloader behavior.

If you can get the system back to a working state then please can you try to repeat the update and capture the output of rpi-eeprom-update. apt-update / apt full-upgrade should always be run before rpi-eeprom-update.

timg236, 2025-01-13:
It looks like the Ubuntu rpi-eeprom package is very old and recovery.bin does not support D0 / 16GB

You should be able to update rpi-eeprom from RPi OS which also uses flashrom to do an immediate update.

cmd0s, 2025-01-13:
> Please can you clarify this statement "After reflashing the SD card, the Raspberry Pi works as it did before."

When I execute the rpi-eeprom-config -e command or perform a firmware update, after running sudo reboot, my Raspberry Pi 5 with 16 GB RAM no longer boots. The red LED stays lit. The power button does nothing, and disconnecting and reconnecting the power supply doesn’t help either.

However, if I remove the SD card, insert it into another computer, and reflash it with Ubuntu Server 24.04.1 LTS using Raspberry Pi Imager, the Raspberry Pi 5 with 16 GB RAM boots again—of course, with a clean system. The bootloader seems to remain unchanged (the old version). So it seems that something in the firmware update process itself didn’t work correctly.

timg236, 2025-01-13:
Thanks. That would be explained by the recovery.bin being too old. 
The bootrom will load it, reject it and stop, so if you deleted that file it would work.

Workarounds.
1) Perform the update on RPi OS 

2) Use the latest software from Github
git clone https://github.com/raspberrypi/rpi-eeprom
cd rpi-eeprom
./test/install -b

cmd0s, 2025-01-13:
I can confirm that after
```sh
git clone https://github.com/raspberrypi/rpi-eeprom
cd rpi-eeprom
sudo ./test/install -b
```
FW update and config update working OK.

Thank you for your quick and accurate assistance!

timg236, 2025-01-13:
I can confirm that the version of rpi-eeprom in the default Ubuntu OS image is too old. Unfortunately, the latest APT package in Ubuntu is too old.
It looks like a newer release is in the works but is not on the Ubuntu APT servers yet (or not 24.04)
https://launchpad.net/ubuntu/+source/rpi-eeprom

closed 2025-01-14 by cmd0s