raspberrypi/rpi-eeprom #654: Firmware fails to boot a GPT SD-card using autoboot.txt on first partrition on Pi 4-B

gportay, 2025-01-14:
### Describe the bug

I faces a boot failure on a Raspberry Pi 4 (64-bit), with the following "settings":
 - SD-card, partitioned with a GPT partition-table (the `sfdisk --dump` is attached below),
 - Using `autoboot.txt` on partition 1 (not using the "tryboot reboot flag", its content is attached below),
 - The bootloader version is the latest as of today [3858f977ab6d689a226ad24d26749266762b7160](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/release-notes.md#2024-12-07-enable-banklow-and-so-numa-by-default-latest)

The bootloader reads the file `autoboot.txt` in the first partition of type `EBD0A0A2-B9E5-4433-87C0-68B6B72699C7`,

The bootloader reads the files (`config.txt`, `start4.elf`, and `fixup4.dat`) in the second partition (`boot_partition=2`), it starts the "firmware?" file `fixup4.dat` and hangs with the ACT LED blinking 7 times in loop.

IIUC, the ACT LED tells me the firmware does not find the kernel.

I am probably mixing firmware and bootloader terms. Could you clarify to me what file is the firmware and what file is the bootlader please?

Important: I am using Buildroot.

Note: The system boots correctly if using MBR SD-Card. The system boots correctly if using GPT SD-Card without `autoboot.txt` (i.e. booting `config.txt` on partition 1; in short, `sudo dd if=/dev/sdb2 of=/dev/sdb1`).

I appears the firmware cannot reads things from the SD-card if using `autoboot.txt`; see the error below.

```
MESS:00:00:04.389523:0: arasan: arasan_emmc_open
MESS:00:00:04.391180:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.511412:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.524282:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.558531:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.675970:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.688851:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.722748:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.736193:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.846761:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:04.859192:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.872056:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.902556:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_WRITE_PROTECTED before we're ready
MESS:00:00:04.915102:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.928296:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
```

### Steps to reproduce the behaviour

Clone the firmware repository and enters the sources:

```
git clone https://github.com/raspberrypi/firmware.git
cd firmware
git checkout 9e65e647c2d8cb1453875c509bf6a730288b7491
```

Make the partition-table to the SD-card...

```
cat - <<EOF | sudo sfdisk /dev/sdb
label: gpt
label-id: A64317D1-55CE-4B3D-A5EF-DD3C2CE247AA
device: /dev/sdb
unit: sectors
first-lba: 34
last-lba: 1245222
sector-size: 512

start=          34, size=       65536, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=BC675DA2-9FF2-4164-B9AD-780AA37B64DC, name="autoboot"
start=       65570, size=       65536, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=DB963A05-9914-44DB-97DA-DDAF089AFCA4, name="boot-a"
start=      131106, size=       65536, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=487F95B5-E299-40D4-A662-9A61E7D396ED, name="boot-b"
start=      196642, size=      524288, type=B921B045-1DF0-41C3-AF44-4C6F280D3FAE, uuid=2D9C70FC-863F-48C0-8AB9-6E6E72F20F7C, name="rootfs-a", attrs="GUID:63"
start=      720930, size=      524288, type=B921B045-1DF0-41C3-AF44-4C6F280D3FAE, uuid=859D3786-44EC-48F9-BB1F-E74AE90629BA, name="rootfs-b", attrs="GUID:63"
EOF
```

... and make the filesystems: 
[... 280 lines cut ...]
MESS:00:00:08.605206:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.614645:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.627864:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.635528:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:08.642074:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.749769:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.759090:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:08.771962:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e8007 emmc: 100000000 actual: 390625 div: 0x00000080 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:08.802445:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.811871:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0107 emmc: 100000000 actual: 50000000 div: 0x00000001 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:08.825072:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.832743:0: No compatible kernel found
```

### USB boot

N/A.

### NVMe boot

N/A.

### Network (TFTP boot)

N/A.

gportay, 2025-01-14:
Also, could you point me to the documentation which GPT UUID-Type should I use for the VFAT partition, the one for `autoboot.txt`, and the ones for `config.txt` (if differents)? Thanks.

timg236, 2025-01-14:
For autoboot.txt we recommend using MBR partitions

timg236, 2025-01-14:
The primary partition numbers are 1-based, so autoboot.txt would be partition 1, FAT A partition 2, FAT B partition 3

Partition 0 means "default" which in most cases will end up mapping to 1

Pi4/Pi5 and older accept the basic-data or EFI GUIDs
https://en.wikipedia.org/wiki/Microsoft_basic_data_partition

gportay, 2025-01-16:
@timg236, I still think there is something weird after the bootloader tries to load the files from the partition located by the `boot_partition` (if the partition table is GPT).

I wonder if `start4.elf` fails to read things from the eMMC bus.

yeryomin, 2025-01-16:
I have exactly same issue!
GPT partitions work just fine without autoboot part.
Using CM4 with IO board here.

timg236, 2025-01-16:
I don't think non-zero partitions (reboot or autoboot.txt) have ever worked with GPT on Pi4 ... I took a tour of the interactions between the legacy file-systems and based something together (attached) ... seems to work but have almost certainly broken something else.

[pi4-gpt-a-b.zip](https://github.com/user-attachments/files/18442314/pi4-gpt-a-b.zip)

MBR or GPT works with contiguous partition layouts like this. Don't expect boot to work from hybrid GPT EFI parts or non-contiguous boot partitions.
```
[all]
boot_partition=2
[gpio2=0]
boot_partition=3

```

timg236, 2025-01-17:
I forgot to say that the attached .zip contains a fix candidate. If you are able to verify that this works for you then it will be easier to get this into the release pipeline.

gportay, 2025-01-17:
Thanks, it works!

```
  0.61 RPi: BOOTSYS release VERSION:3858f977 DATE: 2024/12/07 TIME: 12:39:28
  0.65 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1733575168 serial 2ac43544 boardrev d03115 stc 565596
  0.81 PM_RSTS: 0x00001000
  0.81 part 00000000 reset_info 00000000
  0.84 uSD voltage 3.3V
  0.10 Initialising SDRAM rank 2 total-size: 64 Gbit 3200 (0x14 0x00)
  0.13 DDR 3200 1 0 64 152 BL:3
  2.11 OTP boardrev d03115 bootrom 8b0 8b0
  2.13 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.20 VC-JTAG unlocked

  2.98 RPi: BOOTLOADER release VERSION:3858f977 DATE: 2024/12/07 TIME: 12:39:28
  2.02 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1733575168 serial 2ac43544 boardrev d03115 stc 2502366
  2.37 PCI0 init
  2.37 PCI0 reset
  2.12 PCIe scan 00001106:00003483
  2.17 XHCI-STOP
  2.17 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  2.22 USBSTS 11
  2.24 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  2.29 xHC0 ports 5 slots 32 intrs 4
  2.39 Boot mode: SD (01) order f4
  2.39 USB2[1] 400202e1 connected
  2.57 USB2 root HUB port 1 init
  2.59 DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
  2.62 HUB init [01:00] 2.16 000000:01
  3.71 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  3.82 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  3.90 OCR c0ff8000 [127]
CID: 00744a45555344202010e51017b600dc
CSD: 400e00325b5900003bad7f800a400000
  3.98 SD: bus-width: 4 spec: 2 SCR: 0x02358000 0x00000000
  3.06 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  3.15 MBR: 0x00000001, 1245255 type: 0xee
  3.17 MBR: 0x00000000,       0 type: 0x00
  3.21 MBR: 0x00000000,       0 type: 0x00
  3.25 MBR: 0x00000000,       0 type: 0x00
  3.29 Trying partition: 0
  3.31 GPT 6e2d55525b4fc24b820820d789c096f9 000000001 000130047 num-partitions 128 entry-size 128 pentry 2 first 0
  3.41 GPT partition: 0 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000000022..000010021
  3.48 Matched GUID bootable-part-idx 0 want partition 0
  3.54 type: 16 lba: 34 'mkfs.fat' '  V       ^ ' clusters 16343 (4)
  3.59 rsc 4 fat-sectors 64 root dir cluster 1 sectors 32 entries 512
  3.66 FAT16 clusters 16343
  3.70 Read autoboot.txt bytes       64 hnd 0x2
  3.72 Select partition rsts 0 C(boot_partition) 2 EEPROM config 0 result 2
  3.79 Trying partition: 2
  3.82 GPT 6e2d55525b4fc24b820820d789c096f9 000000001 000130047 num-partitions 128 entry-size 128 pentry 2 first 0
  3.92 GPT partition: 0 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000000022..000010021
  3.99 Matched GUID bootable-part-idx 0 want partition 1
  4.04 GPT partition: 1 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000010022..000020021
  4.11 Matched GUID bootable-part-idx 1 want partition 1
  4.17 type: 16 lba: 65570 'mkfs.fat' '  V       ^ ' clusters 16343 (4)
  4.22 rsc 4 fat-sectors 64 root dir cluster 1 sectors 32 entries 512
  4.29 FAT16 clusters 16343
  4.33 Read config.txt bytes      195 hnd 0x1f
  4.35 [sdcard] pieeprom.upd not found
  4.39 [sdcard] recover4.elf not found
  4.42 [sdcard] recovery.elf not found
  4.72 Read start4.elf bytes  2262656 hnd 0x23
  4.75 Read fixup4.dat bytes     5463 hnd 0x20
  4.77 0x00d03115 0x00000000 0x00001fff
  4.81 MEM GPU: 76 ARM: 948 TOTAL: 1024
  4.50 Firmware: b25f3428a3370a5685ee026a7d0d6557438f078d Jan 16 2025 15:54:38
  4.83 Starting start4.elf @ 0xfec00200 partition 2
  4.85 XHCI-STOP
  4.87 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
[... 596 lines cut ...]
         Starting D-Bus System Message Bus...
         Starting Dropbear SSH daemon...
[  OK  ] Started Getty on tty1.
[  OK  ] Started RAUC Good-marking Service.
[  OK  ] Started Serial Getty on ttyAMA0.
[  OK  ] Reached target Login Prompts.
         Starting OpenSSH server daemon...
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Started Dropbear SSH daemon.
         Starting RAUC Update Service...
[  OK  ] Started OpenSSH server daemon.
[  OK  ] Started RAUC Update Service.
[  OK  ] Reached target Multi-User System.
[    8.642661] Bluetooth: hci0: command 0x0c14 tx timeout

Welcome to Buildroot
buildroot login: [   10.466853] IPv6: ADDRCONF(NETDEV_CHANGE): end0: link becomes ready
[   10.473528] bcmgenet fd580000.ethernet end0: Link is Up - 1Gbps/Full - flow control rx/tx
[   16.866661] Bluetooth: hci0: BCM: Reading local name failed (-110)

buildroot login: root
# md5sum /boot/firmware/fixup4.dat /boot/firmware/start4.elf 
ee0b18c353fbcc679f7d16d6bf0fe23c  /boot/firmware/fixup4.dat
8faecb9899bfa7d822980a034f5ce0c4  /boot/firmware/start4.elf
```

timg236, 2025-01-17:
Thanks for the test report!

SD_GPT_XBOOTLDR seems like it might be a useful addition on Pi5, although bootrom will not recognize this partition type UUID for loading recovery.bin from the SD card. On BCM2711 the split SPI flash bootloader and start4.elf make such a change more difficult e.g. it might load start4.elf and then fail unless both are up to date.

Hybrid should work so long as the GPT is in primary partition 4 i.e. so the bootrom and bootcode.bin can't see it. There is some native support for GPT in Pi3B (maybe just plus) in the bootrom. 

To see if the bootrom is able to load a partition enable UART debug in bootcode.bin - if you get UART output then it's fixable in software.
sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin

yeryomin, 2025-01-17:
@timg236 I can also confirm that your fixed version works as expected for me! thanks!

gportay, 2025-01-20:
@timg236, OOS, since the bootloader propagates the index of the nth FAT partition used to boot the system via the fdt, is it possible to get the device used to boot the system? i.e. the SD-card, the eMMC, the USB or even the network? "Similar" to the EFI variable LoaderDevicePartUUID (`/sys/firmware/efi/efivars/LoaderDevicePartUUID-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f`). The idea is to mount the appropriate FAT filesystem based on `/chosen/bootloader/partition` if there is multiple "bootable" plugged to the Raspberry Pi.

timg236, 2025-01-20:
See /proc/device-tree/chosen/bootloader/partition 

https://www.raspberrypi.com/documentation/computers/configuration.html#common-bootloader-properties-chosenbootloader

although the normal approach would be to put the partition id in the appropriate cmdline.txt

gportay, 2025-01-20:
> See /proc/device-tree/chosen/bootloader/partition
>
> https://www.raspberrypi.com/documentation/computers/configuration.html#common-bootloader-properties-chosenbootloader
> 

Do you mean I should use `/proc/device-tree/chosen/bootloader/boot-order` to know which [BOOT_ORDER](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER).

I would like to distinguish if I booted from the SD card or from the eMMC. And sadly, I cannot using the the `boot-mode`

> although the normal approach would be to put the partition id in the appropriate cmdline.txt

To be honest, I would prefer to have that information from the bootloader itself since it is the only reliable way to have it.

I guess this is not the right place to discuss any longer on that topic. And I am sorry for that. However, is there a better place for asking such feature?

timg236, 2025-01-20:
The partition number, boot-order and selected boot mode are made available via device-tree which uniquely identifies the boot media.
I don't foresee adding UUIDs or other data at the moment

pelwell, 2025-01-20:
> I would prefer to have that information from the bootloader itself since it is the only reliable way to have it.

I don't understand your logic - since cmdline.txt is one of several files loaded by the bootloader from the boot volume, it is arguable that its content is the most reliable indication of where it was loaded from.

lurch, 2025-01-20:
> Do you mean I should use `/proc/device-tree/chosen/bootloader/boot-order` to know which [BOOT_ORDER](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER).

@gportay I think that you might have misinterpreted the documentation - the `/proc/device-tree/chosen/bootloader/boot-mode` doesn't give you the `BOOT_ORDER` (which you could just read from the EEPROM config anyway), it tells you which _single_ boot-mode was used. The documentation for `/proc/device-tree/chosen/bootloader/boot-mode` simply cross-links to the `BOOT_ORDER` documentation to help you correlate the boot-mode number back into a human-readable string 🙂

nbuchwitz, 2025-01-20:
imho the value of `/proc/device-tree/chosen/bootloader/boot-order` is wrong:

```
$ sudo rpi-eeprom-config | grep BOOT_ORDER
BOOT_ORDER=0xf416
$ hexdump /proc/device-tree/chosen/bootloader/boot-mode
0000000 0000 0600                              
0000004
```

Shouldn't this be `0000 0006` ?

pelwell, 2025-01-20:
This is a quirk of hexdump defaulting to 16-bit words. I use `od -Ax -tx1 /proc/device-tree/chosen/bootloader/boot-mode`.

nbuchwitz, 2025-01-20:
Ah, thanks, forget my comment then.

gportay, 2025-01-21:
I am sorry, my asking is poluting the issue with unrelated purpose.

> @gportay I think that you might have misinterpreted the documentation - the /proc/device-tree/chosen/bootloader/boot-mode doesn't give you the BOOT_ORDER (which you could just read from the EEPROM [...]

I think I have understood the documentation. I get `0x00000001` telling me I booted from either SD-Card or eMMC, and I cannot make the distinction. I should have labeled the link `BOOT_MODE` in place of `BOOT_ORDER` (because it refers to the BOOT_ORDER values) and that bring some confusion.

> The partition number, boot-order and selected boot mode are made available via device-tree which uniquely identifies the boot media.

True, at the exception it is not possible to make the distinction between SD-Card or eMMC at it has been said, if both are made available at the same time, I am right?

> I don't foresee adding UUIDs or other data at the moment

I have mentioned the EFI variable (using UUID) to justify my request. And I do not think the UUID is the way to go as UUID are not necessarily uniques (at least for filesystems) especially with A/B redundancy filesystems or installer that copies itself to another device.

> I don't understand your logic - since cmdline.txt is one of several files loaded by the bootloader from the boot volume, it is arguable that its content is the most reliable indication of where it was loaded from.

Well, it seems to, but it implies to mount the nth FAT filesystem of the SD-Card and eMMC to compare the `cmdline.txt` file to `/proc/cmdline`. But the bootloader may modify it in the meanwhile, right? This is what I think it is not the most reliable way to identify the correct nth filesystem if there are multiple boot devices (eMMC, SD-Card, USB...).

FWIW, I guess the boot partition by identifying the root device and I expect it is on the same device. I was wondering if there is something even more simple using the device-tree and the answer is no for now.

To be honest, I think I was looking for a node `/chosen/bootloader/device` that would help to identify the boot device on bus linux side. I am probably the only one on earth asking for that. Again, sorry for all this noise, and thanks for your answers.

lorenzo-arena, 2025-01-24:
> I forgot to say that the attached .zip contains a fix candidate. If you are able to verify that this works for you then it will be easier to get this into the release pipeline.

I am currently dealing with the exact same issue on a CM4 (GPT partitioning + autoboot.txt not loading after start4.elf) and can confirm that the fix candidate worked for me as well.

Muzad, 2025-01-27:
I had a CM4 with my customized YOCTO image, and every things was ok. but after buying a new CM4 few days ago and writing my YOCTO image, i got error:
```
Firmware: 564e5f9b852b23a330b1764bcf0b2d022a20afd0 Feb 25 2021 12:10:40
start4.elf: is not compatible
This board requires newer software
Get the latest software from https://www.raspberrypi.com/software/
ERROR: 44
```
and after updating my firmware recipe, i got the exact same issue on my new CM4 (GPT partitioning + autoboot.txt not loading)

> I forgot to say that the attached .zip contains a fix candidate. If you are able to verify that this works for you then it will be easier to get this into the release pipeline.

I can confirm that the fix candidate worked for me too, but where can get it? i mean which version/tag of firmware can i use to clone in my yocto recipe .bb file?

timg236, 2025-01-27:
The firmware is not released yet, I'll post the tag on the github.com/raspberrypi/firmware once it's available

Muzad, 2025-02-04:
> The firmware is not released yet, I'll post the tag on the github.com/raspberrypi/firmware once it's available

After checking more, i found that although your fix candidate firmware resolved the boot problem, but it's always loading the device tree from first partition and not my autoboot.txt partition!

I mean 'reboot' and 'reboot 1' always load the same dtb. while it was completely okay in the old `Firmware: 564e5f9b852b23a330b1764bcf0b2d022a20afd0 Feb 25 2021 12:10:40`

timg236, 2025-02-18:
The firmware should now be in rpi-update
https://github.com/raspberrypi/firmware/commit/9995946dce284505d3dd46c7563d74a691f38a70

Muzad, 2025-02-18:
[The firmware](https://github.com/raspberrypi/firmware/commit/9995946dce284505d3dd46c7563d74a691f38a70) worked and resolved boot problem but it doesn't load right device tree still. (always the first partition .dtb is loaded)

timg236, 2025-02-18:
Fix merged.

closed 2025-02-18 by timg236