raspberrypi/rpi-eeprom #655: Harmless error updating rpi-eeprom

gitbls, 2025-01-18:
### Describe the bug

Have not seen these on prior updates. They seem harmless, as the eeprom does get updated.
```
apt upgrade --yes
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  rpi-eeprom
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 2,550 kB of archives.
After this operation, 4,196 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 rpi-eeprom all 26.9-1 [2,550 kB]
Fetched 2,550 kB in 0s (86.7 MB/s)   
apt-listchanges: Reading changelogs...
(Reading database ... 104805 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_26.9-1_all.deb ...
Unpacking rpi-eeprom (26.9-1) over (26.8-1) ...
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2712/latest': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2712/default': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2712': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2711/latest': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2711/default': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi/bootloader-2711': Directory not empty
dpkg: warning: unable to delete old directory '/lib/firmware/raspberrypi': Directory not empty
Setting up rpi-eeprom (26.9-1) ...
Processing triggers for man-db (2.11.2-2) ...
```
See also https://forums.raspberrypi.com/viewtopic.php?t=382649 for further comments, especially Murph9000's. 

### Steps to reproduce the behaviour

Update a Pi (even happens on my Pi0w!) that does not have the latest rpi-eeprom installed.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Doesn't matter. Appears to happen on all Pi models

lurch, 2025-01-18:
If you run `apt changelog rpi-eeprom` on a Raspberry Pi, it includes:
```
rpi-eeprom (26.9-1) bookworm; urgency=medium

  [ Tim Gover ]
  * pieeprom-2025-01-13: 2712: Improved SDRAM refresh timings for Pi5 -
    16GB (latest)
  * pieeprom-2025-01-14: 2712: Add set_reboot_order API (latest)

  [ Serge Schneider ]
  * Add salsa-ci.yml

  [ andrum993 ]
  * 2712 release-notes.md: correct name of new config parameter

 -- Tim Gover <tim.gover@raspberrypi.com>  Fri, 17 Jan 2025 16:05:43 +0000

rpi-eeprom (26.8-2) bookworm; urgency=medium

  * Move files from /lib to /usr/lib
  * Bump standards version to 4.6.2
  * Bump compat version to 12
  * Update lintian-overrides

 -- Serge Schneider <serge@raspberrypi.com>  Mon, 13 Jan 2025 14:56:01 +0000
```
so I guess it's the "Move files from /lib to /usr/lib" change that is provoking the above-mentioned harmless warning from `apt`.

@XECDesign / @timg236 If the `.deb` package has been updated to install the EEPROM images into `/usr/lib/firmware/` instead of `/lib/firmware/`, should a similar change be made to e.g. https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L17-L18 ? (just in case there are any systems where `/lib/` _isn't_ a symlink to `/usr/lib/`)
Also, the `dpkg -L rpi-eeprom` output includes `/lib/systemd/system/rpi-eeprom-update.service` - should that also be moved to `/usr/lib/`?

timg236, 2025-01-19:
It's harmless noise from 'merge-usr', possibly indicating that rpi-update was run previously on that device.

closed 2025-01-19 by timg236

gitbls, 2025-01-19:
> It's harmless noise from 'merge-usr', possibly indicating that rpi-update was run previously on that device.

I agree that it's harmless but want to note that I saw this on systems that had definitely NOT had rpi-update run on them.

timg236, 2025-01-19:
> > It's harmless noise from 'merge-usr', possibly indicating that rpi-update was run previously on that device.
> 
> I agree that it's harmless but want to note that I saw this on systems that had definitely NOT had rpi-update run on them.

Yep. Thanks for the detailed report