raspberrypi/rpi-eeprom #660: Raspberry pi 5 active cooler fan full speed all time

wkg21, 2025-02-05:
### Describe the bug

I updated my raspberry pi 5 eeprom to:

root@pineberry:/boot/firmware/overlays# rpi-eeprom-update -a
BOOTLOADER: up to date
CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
Use raspi-config to change the release.

and after update fan connected to built-in 4-pin JST-SH PWM fan connector. started working on full speed despite low temperature:

# vcgencmd measure_temp
temp=26.8'C

and is not obeying settings in /boot/firmware/config.txt

dtparam=fan_temp0=45000
dtparam=fan_temp0_hyst=5000
dtparam=fan_temp0_speed=75

dtparam=fan_temp1=50000
dtparam=fan_temp1_hyst=5000
dtparam=fan_temp1_speed=128

dtparam=fan_temp2=55000
dtparam=fan_temp2_hyst=9000
dtparam=fan_temp2_speed=192

dtparam=fan_temp3=60000
dtparam=fan_temp3_hyst=9000
dtparam=fan_temp3_speed=255

found solution this need to be added to config.txt

dtparam=cooling_fan=on

so behavior changed with a new firmware when you compare what is in README.

less /boot/overlays/README
cooling_fan Enables the Pi 5 cooling fan (enabled automatically by the firmware)

so looks like that not enabled automatically now.

### Steps to reproduce the behaviour

1. have a fan connected to built-in 4-pin JST-SH PWM fan connector
2. update epprom to CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)  I was on september 2024 before

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

# rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

### System

# cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

# vcgencmd bootloader_version
2025/01/22 00:16:51
version a77530632346c8ac1d579fa3c369f9980401e821 (release)
timestamp 1737505011
update-time 1738681359
capabilities 0x0000007f

# vcgencmd version
2025/01/22 00:16:51 
Copyright (c) 2012 Broadcom
version a7753063 (release) (embedded)

# uname -a
Linux pineberry 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

pelwell, 2025-02-05:
Using that EEPROMversion the cooling fan is working as expected for me, with or without your dtparams, and without explicitly loading the cooling-fan overlay.

1. Where are you reading the temperature? I recommend `cat /sys/class/hwmon/hwmon0/temp1_input`.
2. If I unplug the fan, reboot the Pi, then plug it back in again, I get the behaviour you do. Running `sudo dtparam cooling_fan` fixes it. I guess that works for you as well?
3. How are you booting the Pi - SD card, or something else?

During reboot, do you see the fan twitch at all before it starts running at full speed? The firmware is supposed to let the fan spin a bit so that activity on the FAN_TACH line indicates the presence of a fan. Newer versions of the EEPROM then turn off the fan until the kernel starts, whereas it used to be left on.

wkg21, 2025-02-05:
Ad.1     i'm using "vcgencmd measure_temp" to get temperature
Ad.2     yes i unplugged fan and plugged back and issiue persisted until i added : dtparam cooling_fan and after that fan control started working.
Ad.3    using nvme

without dtparam cooling_fan when rebooting fan was at full speed, stopped for a while and again went full speed. 
now with "dtparam cooling_fan on" fan is not running during boot and after. it only starts when cpu goest 45 degress

pelwell, 2025-02-05:
When the fan is spinning, what does `cat /sys/class/hwmon/*/fan1_input` show?

pelwell, 2025-02-05:
And no, the README doesn't need updating - that is how it works for me, and it is how it should also work for you.

wkg21, 2025-02-05:
!. fan runs at full speed.

grep fan /boot/firmware/config.txt
# dtparam=cooling_fan=on

cat /sys/class/hwmon/*/fan1_input 
cat: '/sys/class/hwmon/*/fan1_input': No such file or directory

2. fan runs only when temp higher than my 45 degrees

root@pineberry:~# grep cooling /boot/firmware/config.txt 
dtparam=cooling_fan=on

root@pineberry:~# cat /sys/class/hwmon/*/fan1_input 
0

root@pineberry:~# vcgencmd measure_temp ; cat /sys/class/hwmon/*/fan1_input
temp=45.5'C
1789

wkg21, 2025-02-05:
> And no, the README doesn't need updating - that is how it works for me, and it is how it should also work for you.

unfortunately it doesn't for me. BTW  mine is fresh install with "Raspberry Pi Imager" and "Raspberry Pi OS Lite 64bit"

ivi-lab, 2025-02-10:
I have the exact same issue after updating to the latest version (1737505011) from an old version (1731427844).

I've tested this with a clean Pi OS install with no edits (e.g config.txt). Here are some closer observations:

At startup the fan now pulses twice at 100%. No real issue, just an observation from the previous one pulse it used to do.

Only after a **soft reboot** does the fan continuously run at 100%. If a **soft/hard shutdown** is performed and then powered up, the fan speed behaves fine. Also to note there is no fan pulse after a soft reboot, it just continuously runs at 100%.

ivi-lab, 2025-02-10:
@pelwell Exactly the same as his initial report although he was previously on a different eeprom version. In both cases the fan speed runs at 100% even when the temp is low. This only started after the latest eeprom update.

@wkg21 I'm curious what M.2 HAT are you using? 

Maybe this has something to do with additional hardware (a HAT). I have a PoE M.2 HAT sitting on mine. My test results were all while connected to PoE. There's no difference whether booting from a M.2 or SD card. What I haven't tried is removing the HAT. I'll report back soon.

ivi-lab, 2025-02-10:
So I removed my HAT and used an official Pi 5 power supply. It made no difference, the fan speed issue is still the same. 

I also did the test on a second RPi 5 8GB with the same results.

wkg21, 2025-02-11:
I have removed nvme and did fresh install on sdcard and have exactly the same behavior. fan speed 100% all the time without dtparam=cooling_fan=on and normal behavior after adding i that setting to config.txt.

for nvme i'm using pineboard bottom version.

ivi-lab, 2025-02-11:
@wkg21 How does the fan behave when you shut it down (not reboot) and turn it back on?

wkg21, 2025-02-11:
when i poweroff without cooling_fan option in config. rpi shuts down as expected and when powering it on by pressing power button,  fan is off for ~8s and then it goes full speed.

KelvinHu, 2025-08-12:
This is probably not related, but are third-party active coolers be auto-detected by the firmware? I have a Waveshare Active Cooler B for the RPi 5 and it necessitates `dtparam=cooling_fan=on` when plugged into the fan header.

pelwell, 2025-08-12:
@KelvinHu The auto-detection has changed/improved in more recent firmwares. What is your EEPROM version (`vcgencmd bootloader_version`)?

KelvinHu, 2025-08-12:
> [@KelvinHu](https://github.com/KelvinHu) The auto-detection has changed/improved in more recent firmwares. What is your EEPROM version (`vcgencmd bootloader_version`)?

```
2025/07/17 17:25:12
version a668b6e6edce3274de221324b93cb8741e4a7f7c (release)
timestamp 1752769512
update-time 1754979517
capabilities 0x0000007f
```

Here is my `/boot/firmware/config.txt` as well if it's relevant:

```
[pi5]
# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,cma-512
# max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run as fast as firmware / board allows
arm_boost=1

# Set PCIe slot to gen three
dtparam=pciex1_gen=3

# Explicity enable active cooler fan header
dtparam=cooling_fan=on

# Fan Threshold 0 - 40 degrees
dtparam=fan_temp0=40000
dtparam=fan_temp0_hyst=35000
dtparam=fan_temp0_speed=63

# Fan Threshold 1 - 50 degrees
dtparam=fan_temp1=50000
dtparam=fan_temp1_hyst=45000
dtparam=fan_temp1_speed=127

# Fan Threshold 2 - 60 degrees
dtparam=fan_temp2=60000
dtparam=fan_temp2_hyst=55000
dtparam=fan_temp2_speed=191

# Fan Threshold 3 - 70 degrees
dtparam=fan_temp3=70000
dtparam=fan_temp3_hyst=65000
dtparam=fan_temp3_speed=254

# Setup ethernet over USB-C port
dtoverlay=dwc2
```