raspberrypi/rpi-eeprom #661: Pi 4 mod. B / BCM4345/6 / not set regulatory country code on latest bootloader/GPU firmware

rseleven, 2025-02-08:
### Describe the bug

[Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/):

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.5
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Revision	: d03115
Serial		:
Model		: Raspberry Pi 4 Model B Rev 1.5
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

...

EEPROM
------

BOOTLOADER: up to date
   CURRENT: Сб 07 дек 2024 12:39:28 UTC (1733575168)
    LATEST: Сб 07 дек 2024 12:39:28 UTC (1733575168)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2711/latest)
            Use raspi-config to change the release.

```

dmesg:
```
[    0.076283] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-11-26T12:52:28, variant start
[    0.080287] raspberrypi-firmware soc:firmware: Firmware hash is 2ae30f53898ae2f1ba77ff570a92991bedfb0398
...
[   18.738682] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[   18.748360] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[   46.086090] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[ 1898.255613] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
[ 1902.355723] brcmfmac: brcmf_set_channel: set chanspec 0x100c fail, reason -52
[ 1902.356167] brcmfmac: brcmf_set_channel: set chanspec 0x100d fail, reason -52
[ 1902.356576] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[ 1902.467742] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[ 1902.575619] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
...
[ 1954.246037] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting

```

When purchasing the board, the bootloader firmware specified version 2023. With this version, it was possible to set the country code.

After updating the bootloader firmware to version 2024-12-07, setting the country code stopped working.

I also opened the topic here: [https://forums.raspberrypi.com/](https://forums.raspberrypi.com/viewtopic.php?p=2291411)

### Steps to reproduce the behaviour

`https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#bootloader_update_stable`

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
```

### Bootloader logs

```
[    0.076283] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-11-26T12:52:28, variant start
[    0.080287] raspberrypi-firmware soc:firmware: Firmware hash is 2ae30f53898ae2f1ba77ff570a92991bedfb0398
...
[   18.738682] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[   18.748360] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[   46.086090] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[ 1898.255613] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
[ 1902.355723] brcmfmac: brcmf_set_channel: set chanspec 0x100c fail, reason -52
[ 1902.356167] brcmfmac: brcmf_set_channel: set chanspec 0x100d fail, reason -52
[ 1902.356576] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[ 1902.467742] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[ 1902.575619] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
...
[ 1954.246037] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting

```

timg236, 2025-02-10:
The bootloader firmware does not get involved with WiFi setup so closing as can't reproduce. This will be due to something else on the OS image so I recommend installing a fresh RPi OS image.

closed 2025-02-10 by timg236

rseleven, 2025-02-10:
@timg236, have you tried to reproduce the problem?

lurch, 2025-02-10:
@rseleven If you're convinced that this is a bootloader firmware problem, you can try using `rpi-eeprom-update -f` to downgrade to an older firmware and see if that fixes the problem.

I agree with Tim that it's more likely to be an OS-side problem (either in the kernel or driver) rather than a firmware problem, so if you're still experiencing the problem after installing a fresh copy of Raspberry Pi OS, please open a new issue at https://github.com/raspberrypi/bookworm-feedback/issues with detailed reproduction steps, and I'll try to investigate the issue further.

rseleven, 2025-02-12:
Thanks for the detailed answer. I'm sorry, I was mistaken about the connection between bootloader/GPU firmware and regulatory country code.