raspberrypi/rpi-eeprom #668: [CM4] No longer boots after updating to 1739293213 ("Tue 11 Feb 17:00:13 UTC 2025")

srcshelton, 2025-02-17:
### Describe the bug

After updating to firmware 1739293213 ("Tue 11 Feb 17:00:13 UTC 2025"), one of my CM4 units is no longer booting (from eMMC).  It can still be restarted in MSD mode and its filesystems appear to be intact.

I've held-off from updating the other three identical CM4 units in the cluster for fear of them also being affected, but do have these additional nodes which could be used to test a solution if necessary...

### Steps to reproduce the behaviour

1. Reboot successfully
2. Install firmware 1739293213
3. Reboot (and subsequently power-cycle) and the boot process appears to never complete:

```
TIMESTAMP=1739293213 serial f80ba322 boardrev d03140 stc 530933
  0.40 PM_RSTS 00001000
  0.42 POWER_OFF_ON_HALT: 0 WAIT_FOR_PWER_BUTTON 0 power-on-reset 1
  0.64 part 00000000 reset_info 00000000
  0.65 uSD voltage 3.3V
  0.90 Initialising SDRAM rank 2 total-size: 64 Gbit 3200 (0x14 0x00)
  0.94 DDR 3200 1 0 64 152 BL:3
  2.38 OTP boardrev d03140 bootrom 48b0 48b0
  2.39 Customer key hash 000000000000000000000000000000000000000000000000000000000000000
  2.47 VC-JTAG unlocked

  2.24 RPi: BOOTLOADER release VERSION:75c1e570 DATE: 2025/02/11 TIME: 17:00:13
  2.29 BOOTMODE: 0x06 partition 0 build-ts BULD_TIMESTAMP=1739293213 serial f80ba322 boardrev d03140 stc 2529023
  2.64 Boot mode: SD (01) order f64
  2.79 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.89SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.98 EMMC
  2.98 SD retry 1 oc 0
  2.23 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.34 OCR c0ff8080 [0]
CID: 00150100424a5444345203708e4e9788
  2.54 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 25000000 Hz actual: 25000000 HZ div: 8 (4) status: 0x1fff0000 delay: 4
  2.64 S HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  2.73 MBR: 0x0000001,    2047 type: 0xee
  2.76 MBR: 0x00000800,    4096 type: 0x04
  2.79 MBR: 0x00201800,  131072 type: 0x0b
  2.83 MBR: 0x1f1f800,  131072 type: 0x0b
  2.87 Trying partition: 0
  2.90 GPT e60773e0a240de478c1d95d7d0aa35a3 000000001 003a3dfff num-paritions 128 entry-size 128 pentry 2 first 0
  2.00 GPT partition: 0 b921b045-1df0-41c3-af44-4c6f280d3fae 000001800..0002017ff
 2.07 GPT partition: 1 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000201800..0002217ff
  2.14 Matched GUID bootable-part-idx 0 want patition 0
  2.19 type: 32 lba: 2103296 'BSD  4.4' ' BOOT_0     ' clusters 129024 (1)
  2.26 rsc 32 fat-sectors 1008 root dir clster 2 sectors 0 entries 0
  2.32 FAT32 clusters 129024
  2.40 [sdcard] autoboot.txt not found
  2.41 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 1
  2.48 Tryig partition: 1
  2.50 GPT e60773e0a240de478c1d95d7d0aa35a3 000000001 003a3dfff num-partitions 128 entry-size 128 pentry 2 first0
  2.60 GPT partition: 0 b921b045-1df0-41c3-af44-4c6f280d3fae 000001800..0002017ff
  2.67 GPT partition: 1 ebd0a0a2-b9e5-4433-87c0-68b6b72699c7 000201800..0002217ff
  2.74 Matched GUID bootable-part-idx 0 want partition 0
  2.80 type: 32 lba: 2103296 'SD  4.4' ' BOOT_0     ' clusters 129024 (1)
  2.86 rsc 32 fat-sectors 1008 root dir cluster 2 sectors 0 entries 0
  2.93 FAT32clusters 129024
  2.99 Read config.txt bytes     7379 hnd 0x12e9d
  3.04 Read overclock.txt bytes     2296 hnd 0x1d839
  3.10 Read legacy.txt bytes     3563 hnd 0x1d832
  3.12 pieeprom.sig
  3.13 hash: 45e09714d5379b2479ddf6f7e8d7a5f59756fac5e5174ac548421adaaeef9fc
  3.20 ts: 1739790528
  3.08 SELF-UPDATE timestamp current 1739790528 new 1739790528 skip
  3.14 [sdcard] recover4.elf not found
  3.17 [sdcard] recvery.elf not found
  3.45 Read start4.elf bytes  2262304 hnd 0x1f220
  3.48 Read fixup4.dat bytes     5459 hnd 0x1ef58
  3.51 0x00d03140 0x00000000 0x00001fff
[... 184 lines cut ...]
[cm4]
otg_mode=1
[all]
uart_2ndstage=1
include overclock.txt
include legacy.txt
```

Where all directives in `overclock.txt` and `legacy.txt` are commented.

### Bootloader logs

(See above)

### USB boot

n/a

### NVMe boot

n/a

### Network (TFTP boot)

n/a

timg236, 2025-02-17:
Re: Nit-pick: e missing from compatibl(e).
It looks like characters are being lost on your UART :"e.g. recvery.elf not found"

It looks like the bootloader successfully loaded start.elf from partition1 but start.elf cannot read from the disk. It's GPT partition layout which isn't well supported.

What's the partition layout?

srcshelton, 2025-02-17:
> Both are present in the source - you are losing characters somewhere.

Fair - the BMC UART implementation has always been a bit dodgy (to say the least…)

srcshelton, 2025-02-17:
> What's the partition layout?

```
$ fdisk -l /dev/sdb
Disk /dev/sdb: 29.12 GiB, 31268536320 bytes, 61071360 sectors
Disk model: 0001            
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E07307E6-40A2-47DE-8C1D-95D7D0AA35A3

Device         Start      End  Sectors  Size Type
/dev/sdb1       6144  2103295  2097152    1G Linux root (ARM-64)
/dev/sdb2    2103296  2234367   131072   64M Microsoft basic data
/dev/sdb3    2234368 14817279 12582912    6G Linux variable data
/dev/sdb4   14817280 23205887  8388608    4G Linux /usr (ARM-64)
/dev/sdb5   23205888 27400191  4194304    2G Linux home
/dev/sdb6   27400192 29497343  2097152    1G Linux filesystem
/dev/sdb7   29497344 30537727  1040384  508M Linux swap
/dev/sdb64  30537728 32634879  2097152    1G Linux root (ARM-64)
/dev/sdb65  32634880 32765951   131072   64M Microsoft basic data
/dev/sdb66  32765952 45348863 12582912    6G Linux variable data
/dev/sdb67  45348864 53737471  8388608    4G Linux /usr (ARM-64)
/dev/sdb68  53737472 57931775  4194304    2G Linux home
/dev/sdb69  57931776 60028927  2097152    1G Linux filesystem
/dev/sdb70  60028928 61069311  1040384  508M Linux swap
/dev/sdb128     2048     6143     4096    2M Microsoft basic data

Partition table entries are not in disk order.
```

This has always worked to date, and with firmware releases up to and including 1733575168 (the three other nodes and several other RPi devices have a similar setup, differing only in size).

ragazenta, 2025-02-18:
I also got this problem on Pi 4B. I use GPT partition via `systemd-repart`.

I see that content of rpi-eeprom, bootloader-2711/default was updated recently, I thought it will be frozen at `2024-04-15`, which worked well. In the mean time I think it's safe to downgrade to `2024-04-15` for Pi 4B.

| Bootloader Version | Firmware Version | Works |
|---|---|---|
| 1739293213 (2025/02/11 17:00:13) |  b25f3428a3370a5685ee026a7d0d6557438f078d (Jan 16 2025 15:54:38 ) **(pre-release)** | Yes :white_check_mark: |
| 1739293213 (2025/02/11 17:00:13) |  2ae30f53898ae2f1ba77ff570a92991bedfb0398 (Nov 26 2024 12:52:28) | No :x: |
| 1713186734 (2024/04/15 14:12:14) |  2ae30f53898ae2f1ba77ff570a92991bedfb0398 (Nov 26 2024 12:52:28) | Yes :white_check_mark: |

timg236, 2025-02-18:
I believe the issue is that the bootloader now correctly passes the partition number (resolves 0 to actual number) and passes it to start.elf. Start.elf has never previously correctly handled non-default partition numbers with GPT partitions.

N.B. In any A/B boot-system I would recommend putting all the FAT/EFI partitions together at the start of the GPT or in the MBR world put A/B boot in the first primary partitions and put rootfs as the last or in extended.

timg236, 2025-02-18:
The start.elf updates should now be in rpi-update
https://github.com/raspberrypi/firmware/commit/9995946dce284505d3dd46c7563d74a691f38a70

srcshelton, 2025-02-18:
I see the recent update's ChangeLog states:

> Walk the partition table if the requested partition is not bootable and PARTITION_WALK is set in the bootloader config

… would `PARTITION_WALK` need to be set to resolve this issue, or should things just-work?

stapelberg, 2025-02-18:
I also ran into this issue with https://gokrazy.org/: Updating my Raspberry Pi 4’s EEPROM to https://github.com/raspberrypi/rpi-eeprom/commit/bad328a72ed87edcf6ab821b5c4d31456cb2c2ea resulted in a no-longer booting system.

Updating the firmware to https://github.com/raspberrypi/firmware/commit/9995946dce284505d3dd46c7563d74a691f38a70 does indeed fix the issue for me.

Just in case it helps further narrow down/confirm the problem, my partition layout is:

```
% sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 14,84 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 60C24CC1-F3F9-427A-8199-73B10A8B0000

Device           Start      End  Sectors  Size Type
/dev/mmcblk0p1    8192   212991   204800  100M EFI System
/dev/mmcblk0p2  212992  1236991  1024000  500M Linux root (ARM-64)
/dev/mmcblk0p3 1236992  2260991  1024000  500M Linux filesystem
/dev/mmcblk0p4 2260992 31116253 28855262 13,8G Linux filesystem
```

(The code which generates the MBR+GPT partition table is at https://github.com/gokrazy/tools/blob/8bcde52fb89a27075b0262fe617a96b5d8a90636/packer/packer.go#L433)

timg236, 2025-02-21:
To know for sure I would need to see the UART log from the old bootloader but it does appear that the answer it is to update the RPi 4 firmware first before updating the bootloader. Hopefully, the start4.elf firmware will be in an APT release before too long.
The issues is that older start4.elf does not properly handle the partition number parameter with GPT partitions.

sairon, 2025-03-05:
It appears the root cause for https://github.com/home-assistant/operating-system/issues/3911 is the same. But unlike in #654, the regression appears in the latest 2025-02-11 only, 2024-12-07 is fine (reported breaking in #654). Both with the testing firmware from #654 and with the latest one from the `master` branch of the `firmware` repo, the problem goes away. HAOS also uses hybrid MBR/GPT layout (which obviously suffers from the same flaw with the old `start4.elf`). I'm attaching relevant booloader UART logs below, just to confirm it's indeed the known problem.

At least I know where is the issue coming from and how to work it around, but for proper downstream fix I need to wait for the fix to be merged in the `stable` branch of the firmware repo.

---

<details>
  <summary>2025/02/11 EEPROM with 1.20250127 firmware</summary>

```
  0.75 RPi: BOOTSYS release VERSION:75c1e570 DATE: 2025/02/11 TIME: 17:00:13
  0.79 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1739293213 serial 4a1313db boardrev c03115 stc 579272
  0.89 PM_RSTS 00001000
  0.91 POWER_OFF_ON_HALT: 0 WAIT_FOR_POWER_BUTTON 0 power-on-reset 1
  0.02 part 00000000 reset_info 00000000
  0.03 uSD voltage 3.3V
  0.29 Initialising SDRAM rank 2 total-size: 32 Gbit 3200 (0x15 0x00)
  0.32 DDR 3200 1 0 32 152 BL:3
  2.96 OTP boardrev c03115 bootrom 8b0 8b0
  2.98 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.05 VC-JTAG unlocked

  2.84 RPi: BOOTLOADER release VERSION:75c1e570 DATE: 2025/02/11 TIME: 17:00:13
  2.88 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1739293213 serial 4a1313db boardrev c03115 stc 2488534
  2.23 PCI0 init
  2.24 PCI0 reset
  2.98 PCIe scan 00001106:00003483
  2.03 XHCI-STOP
  2.03 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  2.08 USBSTS 11
  2.10 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  2.15 xHC0 ports 5 slots 32 intrs 4
  2.25 Boot mode: SD (01) order f4
  2.25 USB2[1] 400202e1 connected
  2.43 USB2 root HUB port 1 init
  2.45 DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
  2.48 HUB init [01:00] 2.16 000000:01
  3.62 HUB [01:00] 2.16 000000:01 init port 2 speed 3
  3.64 DEV [02:01] 2.00 000002:01 class 9 VID 1a40 PID 0101
  3.69 HUB init [02:01] 2.00 000002:01
  3.74 HUB [02:01] 2.00 000002:01 init port 1 speed 1
  3.83 DEV [03:02] 1.16 000012:01 class 0 VID 1a86 PID e329
  3.88 HID [03:02] 1.16 000012:01 register HID
  3.57 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  3.68 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  3.38 OCR c0ff8000 [207]
CID: 001b534d4a42315135306aa963c60176
CSD: 400e00325b590000eebf7f800a400000
  3.46 SD: bus-width: 4 spec: 2 SCR: 0x02058083 0x00000000
  3.54 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  3.63 MBR: 0x00000800,   65536 type: 0x0c
  3.65 MBR: 0x00000001,      33 type: 0xee
  3.69 MBR: 0x00000000,       0 type: 0x00
  3.73 MBR: 0x00000000,       0 type: 0x00
  3.76 Trying partition: 0
  3.79 type: 16 lba: 2048 'mkfs.fat' '  V       ^ ' clusters 16343 (4)
  3.85 rsc 4 fat-sectors 64 root dir cluster 1 sectors 32 entries 512
  3.92 FAT16 clusters 16343
  3.96 [sdcard] autoboot.txt not found
  3.97 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 1
  4.04 Trying partition: 1
  4.07 type: 16 lba: 2048 'mkfs.fat' '  V       ^ ' clusters 16343 (4)
  4.13 rsc 4 fat-sectors 64 root dir cluster 1 sectors 32 entries 512
  4.19 FAT16 clusters 16343
  4.24 Read config.txt bytes     2174 hnd 0x2e24
  4.27 [sdcard] pieeprom.upd not found
  4.30 [sdcard] recover4.elf not found
  4.33 [sdcard] recovery.elf not found
  4.59 Read start4.elf bytes  2260128 hnd 0xb6f
  4.63 Read fixup4.dat bytes     5456 hnd 0x5e
[... 379 lines cut ...]
MESS:00:00:05.096847:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:05.104242:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:05.107484:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xdb90
MESS:00:00:05.127860:0: brfs: File read: 56208 bytes
MESS:00:00:05.141214:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:05.169536:0: brfs: File read: 5451 bytes
MESS:00:00:05.172733:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.177259:0: dtparam: audio=on
MESS:00:00:05.185498:0: brfs: File read: 2175 bytes
MESS:00:00:05.194920:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
MESS:00:00:05.217470:0: Loaded overlay 'vc4-fkms-v3d-pi4'
MESS:00:00:05.292359:0: brfs: File read: 1559 bytes
MESS:00:00:05.295137:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:05.299629:0: Read command line from file 'cmdline.txt':
MESS:00:00:05.305459:0: 'dwc_otg.lpm_enable=0 console=tty0 usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u,152d:0562:u,125f:a88a:u,152d:a583:u,152d:a578:u'
MESS:00:00:05.667342:0: brfs: File read: 209 bytes
MESS:00:00:05.780250:0: brfs: File read: /mfs/sd/u-boot.bin
MESS:00:00:05.782779:0: Loaded 'u-boot.bin' to 0x200000 size 0x88eb8
MESS:00:00:05.789816:0: Kernel relocated to 0x80000
MESS:00:00:05.793445:0: Device tree loaded to 0x2eff1e00 (size 0xe1f2)
MESS:00:00:05.804619:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:05.810863:0: uart: Baud rate change done...
MESS:00:00:05.812887:0:
```
</details>

timg236, 2025-03-05:
Thanks for the test report. I'm closing since the OP reported that the reported issue has been resolved albeit in rpi-update. I think it's likely that an APT release will be made soon since the current start.elf firmware has been in rpi-update for a little while now with no obvious blockers for an APT update

Regarding the last post
  4.37 Firmware: 2ae30f53898ae2f1ba77ff570a92991bedfb0398 Nov 26 2024 12:52:28
This firmware is too old to understand the "partition 1" parameter
  4.70 Starting start4.elf @ 0xfec00200 partition 1

I'll comment on this issue as soon as the APT release is made to notify anyone watching this.

closed 2025-03-05 by timg236