raspberrypi/rpi-eeprom #670: [CM5 Lite] SDCard is not detected on bootloader phase.

qkdxorjs1002, 2025-02-21:
### Describe the bug

If turn off the device completely after a normal boot, the bootloader will not recognize the SDCard on the next boot.

The following error appears and does not boot.
`failed to open device 'sdcard' (cmd 371a0010 status 1fff0001)`

Also, when the bootloader entry `RESTART` and `BOOT_ORDER` repeatly, it was very rare to boot.
Based on this, I tried to insert the SDCard when the bootloader is in BOOT_ORDER, it will be surely booted.
with this eeprom config
```
[all]
POWER_OFF_ON_HALT=1
WAKE_ON_GPIO=0

BOOT_ORDER=0xf1
SD_BOOT_MAX_RETRIES=2
```

Other users are also experiencing, and it seems to be more frequent in a specific model.

My SDCard is Samsung Pro Plus 1TB, and it was used normally with CM4 Lite.

### Steps to reproduce the behaviour

1. Prepare CM5 Lite with CM4 compatible board
2. Insert a normally operating SDCard.
3. Turn on the device.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=1
WAKE_ON_GPIO=0
```

### System

```sh
❯ cat /etc/rpi-issue
Raspberry Pi reference 2025-01-23
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, , stage2
```

```sh
❯ vcgencmd bootloader_version
2025/02/12 10:51:52
version f788aab60121515fe59c37c899186831e6c19585 (release)
timestamp 1739357512
update-time 1740162663
capabilities 0x0000007f
```

```sh
❯ vcgencmd version
2025/02/12 10:51:52
Copyright (c) 2012 Broadcom
version f788aab6 (release) (embedded)
```

```sh
❯ uname -a
Linux uconsole 6.6.78-v8-16k+ #4 SMP PREEMPT Tue Feb 18 08:49:38 EST 2025 aarch64 GNU/Linux
```

### Bootloader logs

[image]

qkdxorjs1002, 2025-02-24:
```
❯ sudo vclog -m
040745.307: Initial voltage 800000 temp 57064
040945.404: avs_2712: AVS pred 8558 855800 temp 57614
040949.012: vpred 855 mV +0
040957.665: FB framebuffer_swap 1
040959.534: Select resolution HDMI0/2 hotplug 1 max_mode 2
040966.588: BSC_A no ACK
040970.512: BSC_A no ACK
040974.275: HDMI0: Unable to read EDID block 0
040980.125: Select resolution HDMI1/2 hotplug 0 max_mode 2
040997.336: dtb_file 'bcm2712-rpi-cm5l-cm5io.dtb'
041102.117: Loaded overlay 'bcm2712d0'
041188.272: dtparam: audio=on
041196.855: Unknown dtparam 'audio' - ignored
041199.815: dterror: overlay 'audremap' is not supported on the 'bcm2712' platform
041220.089: dterror: can't find symbol 'audio_pins'
041223.527: dterror: Failed to resolve overlay 'audremap'
041246.609: Loaded overlay 'dwc2'
041248.472: dtparam: dr_mode=host
041253.356: dtparam: ant2=true
041326.001: dterror: can't find symbol 'genet'
041329.006: dterror: Failed to resolve overlay 'clockworkpi-uconsole'
041402.892: Loaded overlay 'vc4-kms-v3d-pi5'
041405.717: dtparam: cma-384=true
041410.564: dtparam: pciex1=off
041424.905: dtparam: sd_debug=on
041431.070: Unknown dtparam 'sd_debug' - ignored
046402.032: BSC_A no ACK
046403.264: BSC_A no ACK
046407.189: BSC_A no ACK
046431.121: BSC_B no ACK
046432.348: BSC_B no ACK
046436.273: BSC_B no ACK
046460.047: Starting OS 46460 ms
046465.652: 00000040: -> 00000480
046467.503: 00000030: -> 00100080
046472.215: 00000034: -> 00100080
046476.929: 00000038: -> 00100080
046481.641: 0000003c: -> 00100080
046591.204: sdram: sdram refresh 2081->4162 (2)
106544.865: initial_turbo of 60 deactivated
```

qkdxorjs1002, 2025-02-25:
failed to open device 'sdcard' (cmd 61a0010 status 1fff0000)

timg236, 2025-02-25:
Sounds like an electrical issue. You could try adding SD_QUIRKS=1 in the latest bootloader config to disable high speed modes

closed 2025-02-25 by qkdxorjs1002