raspberrypi/rpi-eeprom #673: Raspberry Pi 5 serial console device changes with pieeprom-2025-02-12.bin

davidemyers, 2025-03-01:
### Describe the bug

After updating a Raspberry Pi 5 to `pieeprom-2025-02-12.bin` from this repository the serial console changes from `ttyAMA10` to `ttyAMA0`. This causes havoc if something is already using `ttyAMA0`. Falling back to `pieeprom-2025-01-22.bin` restores the previous behavior.

I've seen this on two separate Pi 5's both running Ubuntu Server 24.04 with kernel 6.8.0-1019-raspi. `cmdline.txt` contains the default `console=serial0,115200`. One of these systems is a time server with a Waveshare GNSS HAT using `ttyAMA0` via GPIO 14 & 15, and this system becomes unusable on `pieeprom-2025-02-12.bin` with many errors filling the log.

The Pi 5 time server has `dtparam=uart0` in `config.txt` while the other one does not. Both have `dtoverlay=disable-bt`. Both boot from NVMe.

Here's what's in the logs after booting from the Pi 5 that's not the time server:

Wrong console with `pieeprom-2025-02-12.bin`:
```
Feb 28 17:40:12 pi kernel: Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=2C:CF:67:7E:E0:48 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 multipath=off dwc_otg.lpm_enable=0 console=tty1 root=ZFS=rpool/ROOT/ubuntu rootwait init_on_alloc=0
Feb 28 17:40:12 pi kernel: 1f00030000.serial: ttyAMA0 at MMIO 0x1f00030000 (irq = 134, base_baud = 0) is a PL011 AXI
Feb 28 17:40:12 pi kernel: printk: legacy console [ttyAMA0] enabled
Feb 28 17:40:12 pi systemd[1]: Expecting device dev-ttyAMA0.device - /dev/ttyAMA0...
Feb 28 17:40:12 pi systemd[1]: Found device dev-ttyAMA0.device - /dev/ttyAMA0.
Feb 28 17:40:25 pi systemd[1]: Started serial-getty@ttyAMA0.service - Serial Getty on ttyAMA0.
```

Correct console with `pieeprom-2025-01-22.bin`:
```
Feb 28 18:48:51 pi kernel: Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=2C:CF:67:7E:E0:48 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 multipath=off dwc_otg.lpm_enable=0 console=tty1 root=ZFS=rpool/ROOT/ubuntu rootwait init_on_alloc=0
Feb 28 18:48:51 pi kernel: 107d001000.serial: ttyAMA10 at MMIO 0x107d001000 (irq = 15, base_baud = 0) is a PL011 rev2
Feb 28 18:48:51 pi kernel: printk: legacy console [ttyAMA10] enabled
Feb 28 18:48:51 pi systemd[1]: Expecting device dev-ttyAMA10.device - /dev/ttyAMA10...
Feb 28 18:48:52 pi systemd[1]: Found device dev-ttyAMA10.device - /dev/ttyAMA10.
Feb 28 18:49:10 pi systemd[1]: Started serial-getty@ttyAMA10.service - Serial Getty on ttyAMA10.
```

This is an example of some of the log messages that appear on the system where the GNSS HAT is trying to use `ttyAMA0`:
```
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: sysrq: HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) show-blocked-tasks(w) dump-ftrace-buffer(z) 
Feb 28 17:18:23 chronos kernel: raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110
Feb 28 17:18:23 chronos kernel: raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
```

There's also what looks like a kernel "oops" message which I can post if necessary, as well as error messages from `gpsd`, upset that it can't talk to the HAT.

### Steps to reproduce the behaviour

Install `pieeprom-2025-02-12.bin`.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Pi 5 that's not a time server:
```
[all]
BOOT_UART=1
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
BOOT_ORDER=0xf461
PCIE_PROBE=1
```
The time server is the same but with `BOOT_ORDER=0xf416`.

pelwell, 2025-03-01:
You must have enable_uart=1 in config.txt. Try removing it

davidemyers, 2025-03-01:
You are correct, the default Ubuntu Server `config.txt` contains `enable_uart=1` in an `[all]` section. Commenting it out caused the console to be initialized on `ttyAMA10` with `pieeprom-2025-02-12.bin` as expected. It also works to leave it uncommented and add `enable_uart=0` at the bottom of the file where I have my other customization.

For some reason the `enable_uart=1` had no effect on the Pi 5 until this latest EEPROM release. I think it made sense for the Pi 4 but could cause people problems on the Pi 5.

timg236, 2025-03-06:
Rather than putting enable uart in the [all] section I would use the [pi4] section for all BCM2711 variants.

Working as expected so closing this issue.

closed 2025-03-06 by timg236

davidemyers, 2025-03-06:
I've opened bug [#2101104](https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/2101104) with Ubuntu.

Thanks.