raspberrypi/rpi-eeprom #675: Experimental 2712 DDR init improvements

timg236, 2025-03-06:
### Describe the bug

Placeholder issue to share experimental firmware that improves DDR init on some marginal boards that encounter intermittent 8-flashes (SDRAM error) 

### Steps to reproduce the behaviour

Flash the attached rpi-eeprom-recovery.zip to a SPARE SD card using the RPi Imager.

Boot the Pi5 with this bootloader recovery image - wait for green HDMI screen / steady flashing pattern.

 If this works then attempt to boot the regular OS image

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Any

timg236, 2025-03-06:
[rpi-eeprom-recovery-sdram-init-2025-03-06.zip](https://github.com/user-attachments/files/19111749/rpi-eeprom-recovery-sdram-init-2025-03-06.zip)

timg236, 2025-03-08:
Added release candidate - simplified version of original patch

[rpi-eeprom-recovery-sdram-init-2025-03-08.zip](https://github.com/user-attachments/files/19145172/rpi-eeprom-recovery-sdram-init-2025-03-08.zip)

eXtorian, 2025-03-10:
> Added release candidate - simplified version of original patch
> 
> [rpi-eeprom-recovery-sdram-init-2025-03-08.zip](https://github.com/user-attachments/files/19145172/rpi-eeprom-recovery-sdram-init-2025-03-08.zip)

Been having this issue for a week or two (8 fast flashing green lights on power on) on a Pi 5. I want to say it first appeared after doing a "sudo rpi-update", but it's hard to tell as it was a brand new Pi 5.

Rebooting by "sudo shutdown now -r" would cause the issue to appear maybe 50% of the time. Power cycling one or more times would eventually result in an almost normal boot (it would boot into a red/grey screen for a few seconds before proceeding normally). It would also occur via "sudo shutdown now -h" and power cycling.

Anyway, I installed the simplified version of the patch, and so far I've rebooted the Pi 5 about 20 times perfectly normally.

Thank you for your work on this.

holzgeist, 2025-03-10:
Same here! I had the same issues, this patch (resp the new version that was released 2h ago) works fine

timg236, 2025-03-10:
Thanks for the test reports. The firmware update is in rpi-update. It should get into apt soon barring any reports of regressions

timg236, 2025-05-15:
These fixes are in the default and RPi Imager bootloader releases. Closing as completed.