raspberrypi/rpi-eeprom #684: RPi4 update from 2024-10-21 to 2025-02-11 broke Samsung EVO 64GB SD card compatibility

bmork, 2025-03-22:
### Describe the bug

I have been booting my 8GB RPi4 from a Samsung EVO 64GB SD card for a long time without any issues.  But after running a self-upgrade from pieeprom-2024-10-21.bin  to pieeprom-2025-02-11.bin the boot stopped at the "Rainbow screen".

Note that the pieeprom.upd and pieeprom.sig files were read from this card by the previous eeprom version immediately prior to this failure. The eeprom updates was successfully applied and then it failed to boot with then new eeprom version.

Attempting to reboot the system multiple times by cutting power made no difference.  It consistently stopped at the rainbow without proceeding to boot debug output or continuing with the other configured boot methods.

Removing the SD card allowed the system to boot successfully from a USB  MSD.  There was no indication of any errors, and `vcgencmd bootloader_version` and `bootloader_config` showed the expected output.

I moved the SD card to my laptop to confirm that it was dead, faling or read-only.  But to my surprise, the card was functioning perfectly.  No read errors, write was enabled, and no other problems detected.  I used the laptop to create a USB MSD copy of the SD card and tested that in the RPI4.  It booted and ran without any issues.

So there were apparently no problems with the SD card.

I tried it a few more times in the RPI4, with the same result as before:  Boot hanging at the rainbow.

Then I used the USB MSD to self downgrade the eeprom to the previous version I had been running (2024-10-21).  Tried the SD card again, and it booted without issues!

Looks like some change between the 2024-10-21 and 2025-02-11 stable releases broke compatibility with this specific SD card.

I have not tested the 2024-12-07 release or any beta/latest releases between these two stable versions.

### Steps to reproduce the behaviour

- use a Samsung EVO 64GB SD card as boot device
- update eeprom to the 2025-02-11 stable release

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
[all]
BOOT_UART=1
BOOT_ORDER=0xf241
NET_INSTALL_ENABLED=0
```
The config was the same for both eeprom versions mentioned in this bug report

### System

```
root@rattata:/tmp# vcgencmd bootloader_version
2025/02/11 17:00:13
version 75c1e570e99e14684e79e6ab8f5355d831440c71 (release)
timestamp 1739293213
update-time 1739909555
capabilities 0x0000007f
```

```
root@rattata:/tmp# vcgencmd version
Dec  8 2023 12:36:15 
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start)
```

Not running Raspbian, but can't see how OS and kernel is relevant given that the system fails before attempting to load the kernel or firmware

timg236, 2025-03-22:
You haven't specified what OS you are running or the partition layout and that is likely to be an important factor.

The rainbow screen is generated by start4.elf rather that the SPI flash bootloader so from the description the SPI flash stage has completed and start4.elf is running.

Please can you try running updating the GPU firmware on the OS image via 'rpi-update'

bmork, 2025-03-22:
I am running a Debian stable, with Debian's kernel.  Currently
```
root@rattata:~# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@rattata:~# uname -a
Linux rattata 6.1.0-32-arm64 #1 SMP Debian 6.1.129-1 (2025-03-06) aarch64 GNU/Linux
```

But I am using an older (before the dma-controller split between 2835 and 2711, which is incompatible with Debian stable kernels)  device tree blob from https://github.com/raspberrypi/firmware.git , along with some config settings to maximalize Debian/mainline kernel compatibility:
```
root@rattata:~# cat /boot/firmware/config.txt 
arm_64bit=1
enable_gic=1
disable_commandline_tags=1
disable_overscan=1
arm_boost=1

# avoid turning on TV on reboot
hdmi_ignore_cec_init=1

# use downstream .dtb with upstream kernel on Pi 4 (vc4-kms-v3d-pi4 and dwc2)
dtoverlay=upstream-pi4

# set compatible matching mainline mailbox driver
dtoverlay=vchiq

# automatic bluetooth driver attach
dtparam=krnbt=on

# turn on miniuart (UART1 - ttyS0)
enable_uart=1

# enable uart output from firmware
uart_2ndstage=1

# directly loading kernel
upstream_kernel=1
kernel=vmlinuz
initramfs initrd.img
```

The "vchiq" overlay is a hack to be able to run vcgencmd etc on a mainline kernel.  It siimply adds a compat string matching the vchiq staging driver provided by Debian:
```
/dts-v1/;
/plugin/;

/{
        compatible = "brcm,bcm2711";

        fragment@0 {
                target = <&vchiq>;
                __overlay__ {
                        compatible = "brcm,bcm2711-vchiq", "brcm,bcm2835-vchiq";
                };
        };
};
```

I can try to reproduce the issue without this if you believe it makes a difference, but then I won't be able to use vcgencmd to verify bootloader and firmware versions or config.

> Please can you try running updating the GPU firmware on the OS image via 'rpi-update'

Will test again with the latest firmware, but hope I can take it from https://github.com/raspberrypi/firmware.git ?  Not exactly sure what  'rpi-update' does.

bmork, 2025-03-22:
> Will test again with the latest firmware

Success!  Thanks! 

Didn't realize that the rainbow screen is generated by the firmware, since all Google hits indicated SD card issues.  Which in my simple mind would have prevented the firmware from loading.

Now running fine with
```
root@rattata:/tmp# vcgencmd bootloader_version
2025/02/11 17:00:13
version 75c1e570e99e14684e79e6ab8f5355d831440c71 (release)
timestamp 1739293213
update-time 1742645086
capabilities 0x0000007f
root@rattata:/tmp# vcgencmd version
Mar 19 2025 18:24:21 
Copyright (c) 2012 Broadcom
version ca6e8171a80ea46924ffaa629250bfb482f3a02c (clean) (release) (start)
```

closed 2025-03-22 by bmork

bmork, 2025-03-22:
> Out of interest, is this a GPT / Hybrid GPT setup?

Yes, it is.  It's GPT with a "protective MBR".

```
root@rattata:/tmp# fdisk -lu
Disk /dev/mmcblk0: 59.63 GiB, 64021856256 bytes, 125042688 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: D7B73D21-56D7-CC41-8542-8E534D657F1D

Device           Start      End  Sectors  Size Type
/dev/mmcblk0p1    2048  1050623  1048576  512M EFI System
/dev/mmcblk0p2 1050624 34605056 33554433   16G Linux filesystem
root@rattata:/tmp# file -s /dev/mmcblk0
/dev/mmcblk0: DOS/MBR boot sector; partition 1 : ID=0xee, start-CHS (0x0,0,2), end-CHS (0x3ff,255,63), startsector 1, 125042687 sectors, extended partition table (last)
```

What confused me a bit here was that the same combination of new eeprom and old firmware worked with a USB MSD configured the same way:
```
root@miraculix:/tmp# fdisk -lu /dev/sda
Disk /dev/sda: 28.67 GiB, 30784094208 bytes, 60125184 sectors
Disk model:  SanDisk 3.2Gen1
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: EFD09ABE-A727-1742-BE98-1115C858E288

Device       Start      End  Sectors  Size Type
/dev/sda1     2048  1050623  1048576  512M EFI System
/dev/sda2  1050624 60123135 59072512 28.2G Linux filesystem
root@miraculix:/tmp# file -s /dev/sda
/dev/sda: DOS/MBR boot sector; partition 1 : ID=0xee, start-CHS (0x0,0,2), end-CHS (0x3ff,255,63), startsector 1, 60125183 sectors, extended partition table (last)
```

I guess there is some important difference I don't see here...

timg236, 2025-03-24:
IIRC the legacy file-system for SD re-probes the partition tables but USB-MSD hands over a 'live' USB stack and file-system. One of a number of things that were removed / re-written on Pi5 :)