raspberrypi/rpi-eeprom #686: Raspberry Pi 5 secure boot firmware failure to find boot.sig log message is misleading

bradfa, 2025-03-27:
### Describe the bug

For Raspberry Pi 5 EEPROM firmware version 2025.03.10, after configuring secure boot by programming the OTP public key hash and properly inserting the public key into the EEPROM, when booting an SD card which does not have a `boot.sig` file but *DOES* have a `boot.img` file, then the firmware will complain that it cannot find the `boot.img` file, like:

```
  5.16 secure-boot
  5.18 Loading boot.img ...
  5.21 [sdcard] boot.img not found
  5.24 Error 6 loading boot.img
```

This appears to just be a logging issue to me, where the `Loading boot.img ...` is printed always as a marker to indicate that both `boot.img` and `boot.sig` are being read.  But the error issued indicates that `boot.img` is missing when in fact `boot.sig` is the file which could not be properly read.

### Steps to reproduce the behaviour

1. Flash EEPROM 2025.03.10 with secure boot enabled (my test also has the OTP secure boot hash set but unsure if this matters) and boot a secure-boot enabled SD card.  Observe that the firmware logs show the expected loading of `boot.img` and `boot.sig` like:

```
  5.21 secure-boot
  5.23 Loading boot.img ...
  5.26 boot.sig
  5.27 hash: bec15c9668e5d52dd0eb8afe47bea3350b006cf892f3a9591258c70a68c1654e
  5.34 ts: 1733479805
  5.36 rsa2048: 7806f4cdcafd9dd0737737176b14e585df82e75341c5ceb445e57482740de5faf4902b8ea7ab939a2a10ca760d3e4a70c868273927b38f7638392e3258608995dfa748cc44ec46a4dfe25c1e5ff299d45544340aceaa50d8d7aff5d3797a08b0797f5c13ddb04bd40aa36d29f256d04a9051ce5cb438209c50399cbe4522b508a8c65f55ad2be8efef523794f885ad0e36c66e3c2d43e43cf5b7a75c30fb21f12de072957fcaddd62ecd4f95f3d82604a78416bf2c98074bfe34a440002dadceeb38663d59288de605d4cb0707d4e589405493a6b8d20749180d88dad571c406afcc263b2cca4247d4c58473ed430f9c6fca77adbc93e6f3e80fea52b2be89dd
  8.96 Verifying
 15.96 RSA verify
 15.08 rsa-verify pass (0x0)
```

2. Delete the `boot.sig` file.
3. Reboot.
4. Observe that now the firmware complains that `boot.img` could not be read:

```
  5.34 secure-boot
  5.36 Loading boot.img ...
  5.39 [sdcard] boot.img not found
  5.42 Error 6 loading boot.img
```

### Device (s)

Other

### Bootloader configuration.

`rpi-eeprom-config` does not appear to function correctly when secure boot is enabled, but I believe this was the config I used to build the firmware:

```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf461
ENABLE_SELF_UPDATE=0
```

timg236, 2025-03-27:
Yes, I can see that could be confusing. In signed boot mode it loads pairs of files .sig / .img and the .img isn't loaded if the .sig isn't found. It might be possible to change the text to say
Unable to load signed file (boot.img + boot.sig)

KOLACH1, 2025-08-11:
Hello! I have the same problem (but I've only just started learning this platform and don't understand what you mean by “secure boot”). I only have Raspberry Imager and that's it.