raspberrypi/rpi-eeprom #687: Netboot is not retrying when TFTP download is timed out

mast, 2025-03-31:
### Describe the bug

From time to time we observe situation when rpi5 can't boot via network. After showing boot sequence on HDMI we see black screen and boot sequence never restarts. We found that it's probably due to timeout on TFTP download or when download is corrupted for some reason.
Expectation is: bootloader should detect that netboot is failed and restart boot sequence (after reaching boot mode 0xf).

### Steps to reproduce the behaviour

In our production environment issue is rarely reproducible, mainly when there is some network issues happening or TFTP is not replying fast enough under load.
But we are able to easily reproduce the situation by setting TFTP_FILE_TIMEOUT to extremely small value (5 seconds).
We will be setting TFTP_FILE_TIMEOUT to something big as a workaround, but I believe download still can come corrupted and then boot sequence restart should be proper solution.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf261

### Bootloader logs

This boot log is generated using following config (allowing problem to be easily reproduced):

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf261
TFTP_FILE_TIMEOUT=5000  # this is added 

  1.43 RPi: BOOTSYS release VERSION:897f94ed DATE: 2025/03/27 TIME: 00:16:44
  1.47 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1743034604 serial bd989ae1 boardrev d04170 stc 1647194
  1.57 AON_RESET: 00000003 PM_RSTS 00001000
  1.61 POWER_OFF_ON_HALT: 0 WAIT_FOR_POWER_BUTTON 0 power-on-reset 1
  1.70 RP1_BOOT chip ID: 0x20001927
  1.75 PCIEx1: PWR 0 DET_WAKE 1
  1.77 part 00000000 reset_info 00000000
  1.79 PMIC reset-event 00000000 rtc 00000001 alarm 00000000 enabled 0
  1.85 uSD voltage 3.3V
  1.04 Initialising SDRAM rank 2 total-size: 64 Gbit 4267 (0x14 0x00)
  1.07 DDR 4267 1 0 64 152 BL:1
  3.31 OTP boardrev d04170 bootrom a a
  3.32 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  3.39 VC-JTAG unlocked
  3.62 RP1_BOOT chip ID: 0x20001927

  4.97 RP1_BOOT chip ID: 0x20001927
  4.98 RP1_BOOT: fw size 46888
  4.36 PCI2 init
  4.36 PCI2 reset
  4.80 PCIe scan 00001de4:00000001
  4.81 RP1_CHIP_INFO 20001927

  4.83 RPi: BOOTLOADER release VERSION:897f94ed DATE: 2025/03/27 TIME: 00:16:44
  4.91 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1743034604 serial bd989ae1 boardrev d04170 stc 4991056
  5.01 PCIEx1: PWR 1 DET_WAKE 1
  5.27 usb_pd_init status 3
  5.27 USB_PD CONFIG 0 41
  5.42 XHCI-STOP
  5.42 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  5.47 USBSTS 1
  5.49 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  5.54 xHC0 ports 3 slots 64 intrs 4
  5.66 XHCI-STOP
  5.66 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  5.71 USBSTS 1
  5.73 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  5.78 xHC1 ports 3 slots 64 intrs 4
[... 248 lines cut ...]
 25.11 MESS:00:00:25.411697:0: Loaded overlay 'dwc2'
 25.13 MESS:00:00:25.413562:0: dtparam: dr_mode=peripheral
 25.12 TFTP_GET: 08:00:27:00:d4:68 10.14.0.3 bd989ae1//cmdline.txt

 25.15 RX: 57 IP: 0 IPV4: 51 MAC: 51 UDP: 46 UDP RECV: 39 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 25.25 TFTP: complete 80
 25.26 RX: 59 IP: 0 IPV4: 53 MAC: 53 UDP: 48 UDP RECV: 41 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 25.33 Read bd989ae1//cmdline.txt bytes       80 hnd 0x0
 25.44 MESS:00:00:25.744402:0: RPM 6999, max RPM 6999
 25.28 TFTP_GET: 08:00:27:00:d4:68 10.14.0.3 bd989ae1/armstub8-2712.bin

 25.31 RX: 59 IP: 0 IPV4: 53 MAC: 53 UDP: 48 UDP RECV: 41 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 25.41 TFTP 1: file /tftp/bd989ae1/armstub8-2712.bin not found
 25.45 RX: 60 IP: 0 IPV4: 54 MAC: 54 UDP: 49 UDP RECV: 42 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 25.53 BMD "bd989ae1/armstub8-2712.bin" not found
 25.57 fs_open: 'bd989ae1/armstub8-2712.bin'
 25.61 Loading 'kernel_2712.img' to 0x00000000 offset 0x200000
 25.67 TFTP_GET: 08:00:27:00:d4:68 10.14.0.3 bd989ae1/kernel_2712.img

 25.73 RX: 60 IP: 0 IPV4: 54 MAC: 54 UDP: 49 UDP RECV: 42 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 30.67 RX: 199 IP: 0 IPV4: 193 MAC: 193 UDP: 188 UDP RECV: 180 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 30.72 Read bd989ae1/kernel_2712.img failed
 30.76  kernel_2712.img

and that's it, it just hangs here forever.

mast, 2025-03-31:
Additionally in this situation (when kernel is failed to load and no restart is happening), CPU fan is off and pi can get extremely hot.

timg236, 2025-04-02:
On Pi4 and Pi5 there is a "point of no return" where it's not possible to restart the boot-order sequence.

On Pi4 this is when start4.elf is loaded

On Pi5 - this is after config.txt / 'os_check' device-tree validation.

This is because it's not possible to unpick all the state after some parts of the rtos are running. 

On Pi5 the solution is likely to be a boot watchdog timer+ policy to resume the boot-order at the next element or completely reset on time out. It's a fairly invasive change and some thought will need to go into how to handle slow boot modes i.e. network regarding the heartbeat timer.

timg236, 2025-08-06:
The watchdog changes are merged so closing this issue.

timg236, 2025-10-21:
Now pressing the close button :)

closed 2025-10-21 by timg236