raspberrypi/rpi-eeprom #688: program_pubkey=1 seems to do nothing ? Can't enable secure boot

pseregiet, 2025-04-02:
### Describe the bug

Hello. I used to have a working secure boot working on RPI5. I created a signed pieeprom.upd + sig, put a recovery.bin on the sdcard. That programmed the new EEPROM image and enabled secure boot. Hence forth only a signed boot.img was loaded.

boot.img contains config.txt with program_pubkey=1. That should write the key to the OTP. Recently however I noticed that my Customer key hash is still 0000000000000000000000000000000000000000000000000000000000000000. And sure enough, the public recovery.bin could still run and write an EEPROM image that disabled secure boot.

I was really confused because I have one rpi5 that has the Customer key written and I did it the same way, I never tried any other boot other than from SD card. On that RPI I can only run my own counter signed recovery.bin and I cannot disable secure boot anymore.

I looked at my history and found some logs I haven't seen in a long long while... I wonder why
```
SIG pieeprom.sig <hash> 1724505847
Reading EEPROM: 2097152 bytes 0xc1160000
2483ms
Bootloader EEPROM is up to date
RSA verify
rsa-verify pass (0x0)
Public key hash <hash>
OTP updated for key <hash>
rename recovery.bin to RECOVERY.000
'RECOVERY.000' already exists
rename recovery.bin to RECOVERY.001
```

Did I have some build of recovery.bin that prints on UART and now I have one that doesn't ? strange.

In any case, I though I'll just use latest code, i've pulled the entire usbboot repo and noticed updates to the rpi-eeprm and secure boot related tools. They mention, that the SECURE_BOOT=1 flag was removed from eeprom boot.conf file.

With the latest eeprom firmware and the signing script i got a new pieeprom.upd, i've programmed it and now the secure boot doesn't work at all. Seems like I have  a completely stock rpi that can boot any unsigned boot.img. Any idea what is going on ? Is enabling secure boot from the pi itself not supported anymore ? Was I using some beta release that could do it ? That's how it looks like to me.

### Steps to reproduce the behaviour

Sign an EEPROM image like this Readme says https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery5/README.md
Write EEPROM using recovery.bin
Run boot.img with config.txt that contains program_pubkey=1
Secure boot should be enabled, key written to OTP. But it never happens.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

The "old" eeprom config that enables secure boot but doesn't seem to write the key to OTP
```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=1

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB (RP1), NVMe PCIe, then network
BOOT_ORDER=0xf6241

# Disable self-update mode - to prevent automatic updates of unsigned bootloader images
ENABLE_SELF_UPDATE=0

# Select signed-boot mode in the EEPROM. This can be used to during development
# to test the signed boot image. Once secure boot is enabled via OTP this setting
# has no effect i.e. it is always 1.
SIGNED_BOOT=1
```

The new config, default from https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery5/boot.conf
```
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=1

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB (RP1), NVMe PCIe, then network
BOOT_ORDER=0xf2461
[... 269 lines cut ...]
  7.04 Read kernel_2712.img bytes   587256 hnd 0x10c
  7.06 XHCI-STOP
  7.08 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  7.14 USBSTS 0
  7.17 XHCI-STOP
  7.17 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  7.23 USBSTS 18
  7.46 PCI1 reset
  7.56 PCI2 reset
  7.66 set_reboot_order 0
  7.66 set_reboot_arg1 0
  7.68 USB-OTG disconnect
  7.10 MESS:00:00:07.310151:0: Starting OS 7310 ms
  7.15 MESS:00:00:07.315677:0: 00000040: -> 00000480
  7.17 MESS:00:00:07.317527:0: 00000030: -> 00100080
  7.22 MESS:00:00:07.322239:0: 00000034: -> 00100080
  7.26 MESS:00:00:07.326952:0: 00000038: -> 00100080
  7.31 MESS:00:00:07.331666:0: 0000003c: -> 00100080

NOTICE:  BL31: v2.6(release):v2.6-240-gfc45bc492
NOTICE:  BL31: Built : 12:55:13, Dec  4 2024

U-Boot 2024.04 (Apr 02 2024 - 10:58:58 +0000)

```

timg236, 2025-04-02:
Unless you really need to write your own provisioning tool (which is not recommended, the bootrom signing process is quite complex) please use the secure-boot-provisioner.  
https://github.com/raspberrypi/rpi-sb-provisioner

"Run boot.img with config.txt that contains program_pubkey=1"

That will do nothing. The key is only read from `config.txt` in `secure-boot-recovery5`  when `config.txt` is read over `rpiboot`
https://github.com/raspberrypi/usbboot/tree/master/secure-boot-recovery5

Please edit the `config.txt` file in `secure-boot-recovery5`, run `../tools/update-pieeprom.sh -f -k "${KEY_FILE}"` and capture the UART output of the Pi5 whilst rpiboot is running. It seems to work for us with `secure-boot-provisioner` which wraps these tools.

N.B I don't recommend attempting to use U-boot with secure-boot unless you have already designed how u-boot will verify kernel and all of its dependencies against the hardware root of trust.

pseregiet, 2025-04-02:
> Unless you really need to write your own provisioning tool (which is not recommended, the bootrom signing process is quite complex) please use the secure-boot-provisioner. [...]

I have seen rpi-sb-provisioner but it seems to require an extra hardware and do an usbboot. Can't Rpi5 provision itself ? Like I said, I have one rpi5 that has secure boot enabled with the customer key written to OTP and I never used usbboot on it. It was all done like I described from the sdcard. I'd show you the boot logs from it but I bricked it while doing something unrelated... Maybe I'll make a separate issue about it. But even in its current state it will only run the counter signed recovery.bin (it will flash green led on it and renmate it to RECOVERY.000, the public unsigned recovery.bin doesn't do it). So to me it proves that this rpi5 has the secure boot written to OTP.

As for U-Boot, with my current knowledge I'd not use it indeed. I only use it to select which kernel image and which root partition to use for a A/B boot. Since then I switched to using autoboot.txt but I'm still using the old boot.img containing U-Boot.

timg236, 2025-04-02:
If you think the key customer signing key is already written then start the process from scratch using rpiboot but specify the -r flag

`../tools/update-pieeprom.sh -fr -k "${KEY_FILE}"`
https://github.com/raspberrypi/usbboot/tree/master/secure-boot-recovery5#sign-the-eeprom-and-the-second-stage-bootloader/

If that works you should be able to sign the mass-storage-gadget64 image and prove to yourself that verified boot is enabled and then sign your custom boot.img / OS image.

The bootloader prints the hash of the key digest on the UART at power on so you can also use this to check if the key is programmed.

timg236, 2025-04-02:
It's not possible. External hardware is required to avoid accidental programming of OTP - if there is a path, it's a bug.

pseregiet, 2025-04-02:
> https://www.raspberrypi.com/documentation/computers/config_txt.html#secure-boot-configuration-properties also says: "The following config.txt properties are used to program the secure-boot OTP settings. These changes are irreversible and can only be programmed via RPIBOOT when flashing the bootloader EEPROM image. This ensures that secure-boot cannot be set remotely or by accidentally inserting a stale SD card image."

Yes I remember seeing that but then I managed to make it work anyway and I assumed this statement is wrong. I'm not sure now how I managed to do it. To give you context, I was looking for a way to enable secure boot without any external hardware, because it was not essential to us. I just wanted to enable it if it was possible and I made it work on one of my pies. Convinced I found a way to do it I used this combination (pieeprom.upd+sig+recovery.bin+boot.img+sig) on all rpi5s that we use. Months later, last night, I noticed that the customer hash in the bootrom log is still 000000.

When I first started in august a secure boot was considered beta for raspberry pi5. Is it really possible that I found a bug and programmed secure boot on the pi itself and that should not be possible ?

lurch, 2025-04-02:
AFAIK the only "external hardware" necessary is a standard USB cable between a USB-A port on your PC and the Pi5's USB-C power-input port, so maybe you _did_ use RPIBOOT but you'd forgotten doing so?

https://www.raspberrypi.com/documentation/computers/config_txt.html#program_rpiboot_gpio
"On BCM2712, you can alternatively force RPIBOOT mode by holding down the power button while simultaneously connecting a USB-C power supply." (with your PC acting as the "power supply" in this case)

pseregiet, 2025-04-02:
> AFAIK the only "external hardware" necessary is a standard USB cable between a USB-A port on your PC and the Pi5's USB-C power-input port, so maybe you _did_ use RPIBOOT but you'd forgotten doing [...]

I guess it is possible but I don't remember ever doing that. I was looking for a way to do it on the pi itself, connecting a different computer to the pis at the factory was not an option so I was not considering that. Well anyway, if its not supported then its not supported. Whether I somehow managed to do it on an old bootrom is not relevant (though I am curious...)

May I ask, why is not possible to program from the PI ? We can program the general purpose OTP (I store a LUKS key there) on the pi, why not the secure boot key and enable bit ?

lurch, 2025-04-02:
> why not the secure boot key and enable bit

I guess the most obvious answer, is that if you accidentally enable secure boot with the "wrong" key, then you've effectively just turned your Pi into a large paperweight.

timg236, 2025-05-20:
We discussed this again and came to the same conclusion that secure-boot should not be enabled via 'config.txt alone. The rpiboot / sb-provisioning flow wraps much of the complexity here and goes to some lengths to avoid bricking boards or losing key material.

closed 2025-05-20 by timg236

pseregiet, 2025-05-20:
That's a shame. I understand you wanting to prevent accidental enabling of secure boot but I think it is valid and sensible to have this as an option. I think it would be really nice to have a RPI5 fully provision itself with just the right SD card image without needing any extra hardware to connect. 

Couldn't this really be added ? Let's say an extra config.txt entry, some print and a delay on the uart to give user a 60 seconds to react and turn off the pi if they don't want to proceed or something.

bfreis, 2025-05-31:
+1. That's a shame.

Enabling secure boot is the _only_ part of my entire provisioning system that will require physical interaction with my Pis. And it's a hassle getting them out of their enclosure so I can access stuff.

Honestly, I find "wrapping the complexity" a very poor excuse. It is complex, and people who can't deal with the complexity shouldn't be doing this anyways. And if they do, and they mess up, they will be able to mess up anyways. This is only hurting those who know what they're doing and want convenience, while not being effective at preventing people from messing up.