raspberrypi/rpi-eeprom #690: Cooling_fan not recognized after bootloader/firmware update to 2025/03/19 13:41:26 version cec1d3ae (release) (embedded)

Floyd2000, 2025-04-03:
### Describe the bug

Issue can be reproduced on 2 Raspberry Pi 5 with fresh image of Raspberry Pi OS Lite 64bit Bookworm on an SDCARD.
With the bootloader/firmware version from 2025/03/10, the active cooler (original Raspberry Pi cooler) is reliably recognized as can be seen in "cat /sys/devices/platform/cooling_fan/hwmon/hwmon1/fan1_input".
With the updated version the cooling_fan folder won't get created as the system does not seem to recognize the cooler.
A workaround seems to be manually adding the "dtparam=cooling_fan" in /boot/firmware/config.txt.

The kernel version of the fresh image is 6.1.51 and gets updated to 6.12.20 by apt upgrade. However, kernel version does not seem to make a difference.

What could be reason?

### Steps to reproduce the behaviour

1. sudo raspi-config
2. 6 Advanced Options >> A5 Bootloader Version >> Switch between "E1 Latest" = 2025/03/19 and "E2 Default" = 2025/03/10
3. Reboot and check the existence of the folder /sys/devices/platform/cooling_fan/

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

rpi-eeprom-config 
   [all]
   BOOT_UART=1
   BOOT_ORDER=0xf461
   NET_INSTALL_AT_POWER_ON=1

rpi-eeprom-update 
   BOOTLOADER: up to date
   CURRENT: Wed 19 Mar 13:41:26 UTC 2025 (1742391686)
   LATEST: Wed 19 Mar 13:41:26 UTC 2025 (1742391686)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release

### System

cat /etc/rpi-issue 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

vcgencmd bootloader_version 
2025/03/19 13:41:26
version cec1d3ae40f4a1cb24fe3c42d60153968695385b (release)
timestamp 1742391686
update-time 1743694228
capabilities 0x0000007f

vcgencmd version 
2025/03/19 13:41:26 
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)

uname -a
Linux <hostname> 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

timg236, 2025-04-03:
Please could you run "sudo vclog -m" on failing configuration - without the dtoverlay. You should see something like

```
009281.343: RPM 4470, max RPM 4470
010866.465: Starting OS 10866 ms
```

Floyd2000, 2025-04-04:
@timg236 Thanks for your reply.

This is what I gathered after a couple of reboots:
```
1. 006820.403: RPM 0, max RPM 307
   008711.802: Starting OS 8711 ms

2. 006823.022: RPM 0, max RPM 307
   008714.423: Starting OS 8714 ms

3. 006826.794: RPM 0, max RPM 307
   008718.202: Starting OS 8718 ms
```

This is what I gathered after Hard Power Ons (not soft power off, but hard power off):
```
1. 013046.819: RPM 0, max RPM 307
   015510.476: Starting OS 15510 ms

2. 013051.217: RPM 0, max RPM 306
   015516.007: Starting OS 15515 ms
```

pelwell, 2025-04-04:
Interestingly, my Pi cooling fan shows very non-zero RPM values, even on a reboot:
```
007242.082: RPM 5569, max RPM 5569
```
What values do you get with no fan attached?

Floyd2000, 2025-04-04:
Without fan attached I get:
`013047.247: RPM 0, max RPM 0`

pelwell, 2025-04-04:
Thanks. The official fan seems to bounce the TACH line when it is first powered, before the fan actually spins.

If you aren't relying on USB for your storage, there's a little test you can run to see how the fan reacts to being power-cycled:
```
$ pinctrl USB_VBUS_EN op dl; pinctrl FAN_PWM op dl; sleep 3; pinctrl poll USB_VBUS_EN,FAN_TACH | head -15 & sleep 1; pinctrl USB_VBUS_EN op dh; sleep 3; pinctrl USB_VBUS_EN a2; pinctrl FAN_PWM a0
```
This will turn off the USB bus power for a few seconds, but then show how the fan responds. I get:
```
29: hi // FAN_TACH/GPIO29
42: lo // USB_VBUS_EN/GPIO42
+1000609us
42: hi // USB_VBUS_EN/GPIO42
+36389us
29: lo // FAN_TACH/GPIO29
+106428us
29: hi // FAN_TACH/GPIO29
+14875us
29: lo // FAN_TACH/GPIO29
+10807us
29: hi // FAN_TACH/GPIO29
+9065us
29: lo // FAN_TACH/GPIO29
+7873us
```
This shows an initial delay of 36ms before the first tach change, another 106ms until the fan starts to move, then decreasing intervals as it accelerates.

What results do you get?

oozebot, 2025-04-07:
so I just stumbled upon this after days of fighting with two machines that were recently upgraded. Luckily we didn't update our whole fleet. Is there any chance this could have bricked our RPi5s? We are using the official active coolers. I've done about everything (re-imaged SSD, replaced SSD, replaced M.2 hat, installed an external cooling fan) and nothing works! The RPi 5 will boot but is stupid slow and locks up within minutes. Quite frustrating..

oozebot, 2025-04-07:
Yeah, sorry - I'm probably more frustrated than necessary as we were testing a new build and initially thought it was our code.. but now, the initial boot (RPi OS lite, 64-bit - completely fresh install), even the login prompt is very slow. It has progressively gotten worse (I was leaving them on as I didn't realize the issue) to the point I can't even do a successful apt update before it basically just stops functioning. No logging, just.. stops.

oozebot, 2025-04-07:
The particular RPi5 I'm working with and referencing is now toast. It's a third RPi used for testing and I think it sat in a thermal throttling state overnight several days. Switching the M.2 hat to a fresh-out-of-the-box RPi5 is blazing fast. The other two machines still seem to operate upon reboot, but now I'm suspect of them.

And yes, I did try switching back to the previous bootloader, but I believe it was too late. Guess I'll try to warranty these three.

oozebot, 2025-04-07:
final update - it appears this has affected the PCIe interface on the RPi5s as it boots/operates from an SD without issue. Perhaps they are more susceptible to thermal overload. Unfortunately, that was only confirmed after ruining a second, brand new SSD. ugh..

pelwell, 2025-04-07:
How is the Pi getting so hot? Is it the workload you are running, or a case keeping it toasty?

timg236, 2025-04-08:
If an auto-detect is not working then add the dtoverlay. It doesn't sound like a warranty replacement issue.

oozebot, 2025-04-11:
Final final update: No fan negatively affected the M.2 hats - not the RPi or the SSD. The hat sat pretty close to the cooler and must have overheated. Replacing the M.2 hat with a new design (went with a WaveShare Active Cooler / M.2 combo) has stabilized our systems and no further crashing has been observed.

oozebot, 2025-04-14:
@lurch - it was a 3rd party HAT that came out prior to the official HAT.. what really threw us for a loop was how it affected multiple at the same time. Needless-to-say, we've dumped all those old HATs prior to updating the rest of our machines.

lurch, 2025-04-14:
> it was a 3rd party HAT that came out prior to the official HAT

Our official M.2 HAT went through a **lot** of interoperability testing before release 🙂

mikeg-dal, 2025-05-28:
Just found this while trying to troubleshoot my PI5 with WaveShare POE+ and NVME HAT.  Fan at max speed no matter what I do.

013577.265: RPM 9677, max RPM 9677

raspberrypi5:/sys/devices/platform/cooling_fan/hwmon/hwmon3 $ cat fan1_input 
12728

aspberrypi5:~ $ rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Thu  8 May 14:13:17 UTC 2025 (1746713597)
    LATEST: Thu  8 May 14:13:17 UTC 2025 (1746713597)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

Booting of NVME drive and PoE+ powering the unit.

timg236, 2025-05-28:
> Just found this while trying to troubleshoot my PI5 with WaveShare POE+ and NVME HAT. Fan at max speed no matter what I do.
>
> 013577.265: RPM 9677, max RPM 9677 [...]

This issue is about failing to detect a fan. The logs show the fan being detected so is a different issue, probably a problem with the pwm pin.

mikeg-dal, 2025-05-28:
> a

Thanks for the info, Guess I will keep troubleshooting the issue.

Floyd2000, 2025-05-29:
Issue closed because fan was recognized again with bootloader versions after 2025/03/19. Thanks for your quick support! :-)

closed 2025-05-29 by Floyd2000

timg236, 2025-05-29:
> > a
> 
> Thanks for the info, Guess I will keep troubleshooting the issue.

I suggest inspecting the physical connection, IIRC if it's shorted to ground then it will run full speed. Try driving it high via pinctrl (see previous comments on this issue)