raspberrypi/rpi-eeprom #692: USB speed on latest firmware

bigalnz, 2025-04-06:
### Describe the bug

I had a python project which ran fine on BCM2712 C0 hardware - it involves streaming a lot of data across the USB3 interface from a Airspy HF SDR.

I try this same code on a BCM2712 D0 with latest firmware and it drops a lot of data. Querying if it is possible there is a USB3 speed issue in this FW?

I tried to insert the old SD card from the C0 hardware but of course that didnt boot. I also tried to boot a new SD card with fresh install, which did boot and then chroot to the old file system and run a upgrade, but the old file system fails to boot when I do this. I get a message:

brcm-pcie 1000120000.pcie: Unable to find MSI PCI address
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_cmd"
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_dat0"
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_dat1"
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_dat2"
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_dat3"

then a flashing cursor.

Not sure if the potentially slow USB is a eeprom issue or something else, but raising it as a possibility.

### Steps to reproduce the behaviour

As above.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

BOOTLOADER: up to date
   CURRENT: Mon 10 Mar 2025 17:10:37 UTC (1741626637)
    LATEST: Mon 10 Mar 2025 17:10:37 UTC (1741626637)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

### System

al@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
al@raspberrypi:~ $ vcgencmd bootloader_version
2025/03/10 17:10:37
version 2bb2ae640058a2f3aa8dcbdc2da33302e509668d (release)
timestamp 1741626637
update-time 1743889345
capabilities 0x0000007f

al@raspberrypi:~ $ vcgencmd version
2025/03/10 17:10:37
Copyright (c) 2012 Broadcom
version 2bb2ae64 (release) (embedded)
al@raspberrypi:~ $ uname -a
Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
al@raspberrypi:~ $

bigalnz, 2025-04-07:
What do you mean reproduce on Rpi OS? I have installed a fresh copy of Rpi OS (Bookworm) on a new SD and installed the program from GitHub and that's when I get the dropped data.

I tried rolling back firmware to the Janurary 2025 release but that made no difference.

timg236, 2025-04-07:
The following messages shouldn't appear in any bookworm image since Pi5 2GB D0 was release (mid 2024).

brcm-pcie 1000120000.pcie: Unable to find MSI PCI address
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_cmd"
pinctrl-bcm2712 107d504100.pinctrl: could not map pin config for "emmc_dat0"

After this is resolved then you'd need to describe exactly what data you are capturing and how to the dropped data can be reproduced to determine whether it's an issue with the 6.6 vs  6.12 kernel

bigalnz, 2025-04-07:
Sorry for the confusion - those messages are only present when I tried to boot the OLD SD card which I had tried (and failed)  to upgrade to a current release.

The data I am capturing is a stream of data coming from a AirSpyHF+ SDR (Software Defined Radio) connected to USB3 ports on the Rpi. 

The AirSpyHF+ lib is a c library (and my project is Python based) so I use ctypes in Python to read in the data:
https://github.com/airspy/airspyhf

This all was working beautifully on the original Rpi5 c0 - but not the replacement Rpi d0. 

I can link you to my project github here : https://github.com/bigalnz/test_fft

Not sure if it can be reproduced without the AirSpy Hardware?

bigalnz, 2025-04-08:
I guess the question still remains - is it possible to manually upgrade the old SD card to boot in the new hardware?

timg236, 2025-04-08:
Yes, to update a RPi OS image run

```
sudo apt update
sudo apt full-upgrade

```

bigalnz, 2025-04-08:
sudo apt update
sudo apt full-upgrade

How are these commands run on a SD card that doesnt boot?

popcornmix, 2025-04-08:
> How are these commands run on a SD card that doesnt boot?

didn't you say:

> This all was working beautifully on the original Rpi5 c0 - but not the replacement Rpi d0.

Can you boot on the c0, run the update, and then switch the sdcard to the d0?

bigalnz, 2025-04-08:
> Ok, rather than attempting to migrate the current SD card about trying
> 
> Stock RPi OS 64-bit on an SD card (6.6 kernel) vs Stock RPi OS 64-bit on an SD card after apt update /upgrade (6.12 kernel)
> 
> That will help verify whether there is a performance difference related to the Linux kernel versions.
> 
> Also, do you see USB related errors in dmesg (e.g. dmesg | grep -i usb) when you encounter data-loss?

So if I download the image writing app, and create a new fresh SD card it will be Stock on 6.6? Then if I was to run a update.upgrade it should end up on 6.12?

popcornmix, 2025-04-08:
> So if I download the image writing app, and create a new fresh SD card it will be Stock on 6.6? Then if I was to run a update.upgrade it should end up on 6.12?

yes

bigalnz, 2025-04-08:
No difference in performance between kernel versions.

Nothing interesting on dmesg| grep -i usb

[    0.447887] usb usb3: SerialNumber: xhci-hcd.1
[    0.447947] hub 3-0:1.0: USB hub found
[    0.448038] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.12
[    0.448040] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.448042] usb usb4: Product: xHCI Host Controller
[    0.448043] usb usb4: Manufacturer: Linux 6.12.20+rpt-rpi-2712 xhci-hcd
[    0.448045] usb usb4: SerialNumber: xhci-hcd.1
[    0.448098] hub 4-0:1.0: USB hub found
[    0.693772] usb 3-1: new high-speed USB device number 2 using xhci-hcd
[    0.834027] usb 3-1: New USB device found, idVendor=03eb, idProduct=800c, bcdDevice= 1.00
[    0.834033] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    0.834035] usb 3-1: Product: AIRSPY HF+
[    0.834037] usb 3-1: Manufacturer: www.airspy.com
[    0.834038] usb 3-1: SerialNumber: AIRSPYHF SN:DC52978027444A4D
[    2.634083] usbcore: registered new interface driver brcmfmac
 

And just dmesg

[    2.599308] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    2.599351] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    2.599587] Bluetooth: HCI UART protocol Broadcom registered
[    2.629977] brcmfmac: F1 signature read @0x18000000=0x15264345
[    2.633791] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    2.634083] usbcore: registered new interface driver brcmfmac
[    2.646337] [drm] Initialized vc4 0.0.0 for axi:gpu on minor 1
[    2.723773] Console: switching to colour frame buffer device 240x67
[    2.741407] vc4-drm axi:gpu: [drm] fb0: vc4drmfb frame buffer device
[    2.846939] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    2.847211] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    2.954146] Bluetooth: hci0: BCM: chip id 107
[    2.954350] Bluetooth: hci0: BCM: features 0x2f
[    2.955386] Bluetooth: hci0: BCM4345C0
[    2.955388] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    2.963083] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    3.562138] Adding 524272k swap on /var/swap.  Priority:-2 extents:7 across:2228224k SS
[    3.634443] Bluetooth: hci0: BCM: features 0x2f
[    3.635809] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    3.635812] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    3.636122] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    3.656408] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.656415] Bluetooth: BNEP filters: protocol multicast
[    3.656421] Bluetooth: BNEP socket layer initialized
[    3.658236] Bluetooth: MGMT ver 1.23
[    3.669019] NET: Registered PF_ALG protocol family
[    4.356646] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    4.356656] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    4.360536] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    4.373079] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    6.571510] Bluetooth: RFCOMM TTY layer initialized
[    6.571522] Bluetooth: RFCOMM socket layer initialized
[    6.571531] Bluetooth: RFCOMM ver 1.11

One thing I did find interesting is I can adjust the sample rate of the radio for example 768000 to 456000 etc .... no matter how low I set the sample rate it always drops heaps of samples. I would expect if this was a bottle neck issue with the code, then the dropping would decrease with the lower samples but this was not the case.

bigalnz, 2025-04-08:
Is there anywhere I can buy specifically a RPi5 c0 ?

popcornmix, 2025-04-09:
> Is there anywhere I can buy specifically a RPi5 c0 ?

There's no specific product numbers for c1 and d0, so not directly.
You may find a seller that has older stock but you'd likely have to phone around.
Second hand may be a good option.

A D0 chip ends with D0T on the second line of the text on the main CPU. A C1 chip ends with C1T, so they can be determined without booting up.

timg236, 2025-04-09:
Does adding force_turbo=1 in config.txt make any difference to performance? The logs don't indicate any USB errors, but if the application is not keeping up / requesting bulk transfers fast enough that wouldn't be a USB error as such.

Also please can you try running 'vcgencmd get_throttled' after the first errors are encountered.

timg236, 2025-04-09:
We are looking at getting Airspy HF SDR as a test case.
Is this the same as your model https://moonrakeronline.com/airspy-hf-plus-discovery-high-peformance-sdr-receiver

bigalnz, 2025-04-09:
> We are looking at getting Airspy HF SDR as a test case. Is this the same as your model https://moonrakeronline.com/airspy-hf-plus-discovery-high-peformance-sdr-receiver

Yes thats it. If you want I can give you a ssh session on my Rpi

bigalnz, 2025-04-09:
> Does adding force_turbo=1 in config.txt make any difference to performance? The logs don't indicate any USB errors, but if the application is not keeping up / requesting bulk transfers fast enough [...]

I added the force_turbo=1 in.

Before I ran the code I checked get_throttled and got `throttled=0x0` - it was the same afterwards.

timg236, 2025-04-11:
Thanks. We've ordered one so that we can run it with the big USB analyzer attached.

Please can you add instructions to reproduce your setup on RPi OS

bigalnz, 2025-04-13:
to install I did 

`python -m venv kiwitracker
cd kiwitracker
pip install kiwitracker@git+https://github.com/bigalnz/test_fft.git
cd ../`

Get AirspyHF+ lib

`sudo apt-get install build-essential cmake libusb-1.0-0-dev pkg-config
wget https://github.com/airspy/airspyhf/archive/master.zip
unzip master.zip
cd airspyhf-matser
mkdir build
cd build
cmake ../ -DINSTALL_UDEV_RULES=ON
make
sudo make install
sudo ldconfig`

CD back into kiwitracker/test_ftt

Then try running kiwitracker with 👍 
`kiwitracker -s 768000 --center 160425000 --no-use-gps --loglevel info -f test_windows_bottom_mtn.fc32`

bigalnz, 2025-04-23:
I tried on a older RPi5 and still get dropped samples. I think it is a c0?

`(kiwitracker) pi@rpi:~/kiwitracker/test_fft $ uname -a
Linux rpi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
(kiwitracker) pi@rpi:~/kiwitracker/test_fft $ vcgencmd version
2024/09/23 14:02:56
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)
(kiwitracker) pi@rpi:~/kiwitracker/test_fft $ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
(kiwitracker) pi@rpi:~/kiwitracker/test_fft $
`

P33M, 2025-05-06:
Following your instructions results in the following when attempting to run kiwitracker:
```

timg236, 2025-05-14:
@bigalnz We now have an airspy, please could you check the software installation instructions for KiwiTracker ?

timg236, 2025-10-21:
Closing due to inactivity.

closed 2025-10-21 by timg236