raspberrypi/rpi-eeprom #699: RPI 5 bootloader affect Hifiberry card

JN19aban, 2025-04-26:
### Describe the bug

Hi,

I have an RPI5 which I was updated the bootloader to the latest `pieeprom-2025-04-07.bin` and after that the problems started with Hifiberry sound card.

Furthermore, after the update the `Hifiberry DAC2 PRO` was not discovered (which before the update was working normally) and I did reboot sometimes and then it recognized correct, but the audio performance was not good, very blurry sound and huge latency. After couple reboots and revert to the previous kernel `6.6.78` from `6.12.20` was still the same and I wondering if my sound card was broken and I did one more reboot. 

And then BOOM, no more discover of the DAC2 PRO, no led active, no anything, no matter how many reboot I have done......

Then I bought a new one the same model `(DAC2 PRO)` and after I install it to the RPI5 the same thing started to happen as the "doomed sound card"...... bad audio and not discover until do some reboots..... 

For not brake the new soundcard I removed from RPI5 and install it on the old RPI4 rev 1.4 I have.... and it was worked flawless, perfect audio, detect every time I do reboot....

So I was thinking my RPI5 is broken and did one last effort to downgrade the firmware to the default `pieeprom-2025-03-10.bin` and I install again the Hifiberry sound card to RPI5....

And after that the audio performance came back, it detect normally no matter how many reboot you do.

So I do not know if the old Hifiberry was the time to die or the firmware kill it but I just want to let you know just in case something has to do with the firmware.

Best regards,
Jim

### Steps to reproduce the behaviour

Just update to the latest bootloader.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
BOOT_ORDER=0xf461
NET_INSTALL_AT_POWER_ON=1

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

2025/03/10 17:10:37
version 2bb2ae640058a2f3aa8dcbdc2da33302e509668d (release)
timestamp 1741626637
update-time 1745630843
capabilities 0x0000007f

2025/03/10 17:10:37
Copyright (c) 2012 Broadcom
version 2bb2ae64 (release) (embedded)

6.6.78-v8-16k+ #1854 SMP PREEMPT Mon Feb 17 16:37:33 GMT 2025 aarch64 GNU/Linux

lurch, 2025-04-26:
Does your "old" DAC2 PRO HAT work on the Raspberry Pi 4 too, or is it definitely "dead"?

JN19aban, 2025-04-26:
Sadly it is "dead" not recognized any more on both RPI's (RPI5 and RPI4)........

pelwell, 2025-04-28:
Is this a problem you have heard about from any other users, @HiFiBerry?

JN19aban, 2025-04-28:
Hi,

I just upload log files of the "dead" and new `DAC2 PRO` just in case something is there...

Also to mention that the log files are with `6.6.78` kernel and stable bootloader  `pieeprom-2025-03-10.bin`.
Really I hope just to be a coincidence, but I am not risking to break and a new sound card for test again.

Best regards,
Jim

[dmesg bad DAC2 PRO.txt](https://github.com/user-attachments/files/19948338/dmesg.bad.DAC2.PRO.txt)
[i2cdetect-bad DAC2 PRO.txt](https://github.com/user-attachments/files/19948337/i2cdetect-bad.DAC2.PRO.txt)
[i2cdetect NEW DAC2 PRO .txt](https://github.com/user-attachments/files/19948339/i2cdetect.NEW.DAC2.PRO.txt)
[dmesg NEW DAC2 PRO.txt](https://github.com/user-attachments/files/19948340/dmesg.NEW.DAC2.PRO.txt)